raspberrypi/linux #2056: CONFIG_DMA_BCM2708 doesn't support module compilation

agherzan, 2017-06-05:
CONFIG_DMA_BCM2708 is defined as a tristate config in https://github.com/raspberrypi/linux/blob/204050d0eafb565b68abf512710036c10ef1bd23/drivers/dma/Kconfig .

The code for the above config includes https://github.com/raspberrypi/linux/blob/rpi-4.9.y/include/linux/platform_data/dma-bcm2708.h which defines (for example) bcm_dmaman_probe based on CONFIG_DMA_BCM2708 definition. In the case of CONFIG_DMA_BCM2708 set on module, CONFIG_DMA_BCM2708_MODULE ends up defined and not CONFIG_DMA_BCM2708 which breaks compilation while hitting a redefinition error.

One solution would be to restrict CONFIG_DMA_BCM2708 to bool. A second option would be to take into consideration CONFIG_DMA_BCM2708_MODULE too in dma-bcm2708.h . I can provide patch for either of these options.

popcornmix, 2017-06-05:
The amount of code in bcm2708-dmaengine.c is tiny, so no big loss if it is forced to be boolean.
But if fixing it up to build as a module is trivial then I'm fine with that.
@pelwell any preference?

agherzan, 2017-06-05:
I was able to compile it with a simple change in the dma-bcm2708.h which doesn't only checks for CONFIG_DMA_BCM2708 but for CONFIG_DMA_BCM2708_MODULE too. So both solutions seem trivial.

pelwell, 2017-06-05:
If it really is that simple then let's keep the option of a module build - submit a Pull Request.

Thanks in advance.

closed 2017-06-06 by pelwell