raspberrypi/linux #2059: i2c-rtc-overlay, __overrides__ in wrong order

oliskoli, 2017-06-06:
Current version activates mcp7940x when using overlay parameter ds3231.

diff --git a/arch/arm/boot/dts/overlays/i2c-rtc-overlay.dts b/arch/arm/boot/dts/overlays/i2c-rtc-overlay.dts
index b73f902289c8..1b2328ebc577 100644
--- a/arch/arm/boot/dts/overlays/i2c-rtc-overlay.dts
+++ b/arch/arm/boot/dts/overlays/i2c-rtc-overlay.dts
@@ -147,9 +147,9 @@
                abx80x = <0>,"+0";
                ds1307 = <0>,"+1";
                ds1339 = <0>,"+2";
-               ds3231 = <0>,"+3";
-               mcp7940x = <0>,"+4";
-               mcp7941x = <0>,"+5";
+               mcp7940x = <0>,"+3";
+               mcp7941x = <0>,"+4";
+               ds3231 = <0>,"+5";
                pcf2127 = <0>,"+6";
                pcf8523 = <0>,"+7";
                pcf8563 = <0>,"+8";

pelwell, 2017-06-07:
It's a fair cop - sorry about that. I'll add some automated testing for these container overlays.

pelwell, 2017-06-07:
The fix will be in the next firmware release, but in the meantime you can download an updated dtbo here: https://drive.google.com/open?id=0B_P-i4u-SLBXN240NEVJUmZXak0

oliskoli, 2017-06-07:
thanks for quick response

closed 2017-06-07 by oliskoli