raspberrypi/linux #2060: SPI: decouple spi speed from core clock ?

penfold42, 2017-06-07:
Is there a way to keep a consistent spi clock when the core clock changes ?

I guess the spi driver would need to adjust the divisor to account for the dynamic core clock

pelwell, 2017-06-07:
SPI is one of several devices that suffer from being tied to the VPU system (core) clock. UART1 is the worst because the clocking is implicit. The new ARM-side clock driver in 4.9 now reports the higher (turbo) clock frequency to its dependents, so SPI and I2C divisors are calculated such that the resulting device clock should never be higher than the expected value, but can be lower.

The firmware has to be in control of the core clock in case of over-temperature or under-voltage conditions - we don't want the VPU to wait until Linux says it is OK to change the clock, so some fluctuation is always possible. There is always the workaround for stopping the VPU clock from changing by setting core_freq=250 in config.txt. For most use cases you are unlikely to notice any adverse effects, and it is already implied by "enable_uart=1" on a Pi3 (unless the pi3-miniuart-bt or pi3-disable-bt overlay is loaded). 

Is your question resulting from unreliability, reduced throughput or something else?

penfold42, 2017-06-07:
Thanks for the detailed reply. In this case it's driving led strips by (ab)using the spi interface - there is no clock connected, just carefully timed bit streams.

I've been using force_turbo to peg it successfully but couldn't help wondering if the spi driver could automatically recalculate the divisor under a clean frequency change.
This would require we don't change core clock mid DMA transfer.

pelwell, 2017-06-07:
Any clock change would have to be very carefully timed, which is particularly difficult when the turbo is ultimately controlled by the VPU while the SPI driver runs on the ARM.

Setting core_freq to 250 will still allow the ARM to enter turbo mode but without a corresponding core clock change. This protects you from over temp/under volt throttling and is generally kinder on the CPU than force_turbo=1.

penfold42, 2017-06-07:
Setting core_freq seems to work for me with osmc now that it's 4.9 - it for some reason never worked a couple of years ago for me (but I wasn't very scientific back then !)

So, to summarise before I update some doco and recommendations to users of Hyperion and rpi_281x.

The leds have a reasonable range of tolerance for the bit rate.

1) use core_freq if you can - then the spi rate will be fixed

2) if you haven't/can't/won't:
Kernel >= 4.9 pick a spi bit rate at the higher end of the allowable led bit rate range
Kernel < 4.9 pick a spi rate at the lower end of the range

pelwell, 2017-06-07:
I can't readily explain the observed pre-4.9 behaviour - although 4.4 didn't use the ARM-side clock manager the end result ought to have been the same - but otherwise your advice looks good.

pelwell, 2017-06-07:
You can close this issue if you are happy it has been resolved.

P.S. I like your avatar - there was a time I would have decoded it without having to look anything up.

penfold42, 2017-06-07:
Thanks again for your clarifications

Re: the avatar - I really should have used position independent code but, oh well, picking the address to assemble it to is another part of the lame challenge !

closed 2017-06-07 by penfold42