raspberrypi/linux #2061: bcm2835-cpufreq: Clarify license

lategoodbye, 2017-06-07:
Stephen Warren raised [concerns](http://lists.infradead.org/pipermail/linux-rpi-kernel/2015-October/002445.html) about the license header of bcm2835-cpufreq during the last upstreaming approach. It would be nice to get this "fixed" here, so the upstream guys can concentrate on the technical part.

popcornmix, 2017-06-07:
The same licence clause seems to be used in other upstream code. e.g.
https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/net/ethernet/broadcom/bnx2x/bnx2x_sriov.c

lategoodbye, 2017-06-07:
The ethernet driver has been written and mainlined by a broadcom employee. IANAL, but as an external i'm not comfortable with such special clauses and i don't want ignore an experienced Linux maintainer.

popcornmix, 2017-06-07:
I've asked and we should be able to get permission from Broadcom to replace the licence with a generic GPL2 one. I'll let you know when we have an email confirming that.

popcornmix, 2017-06-21:
This has been agreed by Broadcom. Eben (using his Broadcom employee status) has signed off.
See: https://github.com/raspberrypi/linux/commit/031c46312c88c009b114cefcfeb51217b4ddc23c

lategoodbye, 2017-06-21:
Great. Thanks

closed 2017-06-21 by lategoodbye

yuergen, 2018-06-04:
Hello. The updated license is great! Is there currently an effort to upstream this? It would be pretty great to have dynamic cpu frequencies available for the mainline kernel and thus in fedora for example.