raspberrypi/linux #2063: Include pwm_bl.c/pwm_bl.ko in Raspberry Pi config

Gadgetoid, 2017-06-08:
Is it possible to include pwm_bl.ko by default in Raspbian?

Source: https://github.com/raspberrypi/linux/blob/204050d0eafb565b68abf512710036c10ef1bd23/drivers/video/backlight/pwm_bl.c

We've got a new product- Hyper Pixel- that can use this module to drive its dimable backlight. I've compiled it manually and tested it by loading with a device-tree blob.

Thank you!

pelwell, 2017-06-09:
At first glance I don't see any reason why we couldn't include this. How are you getting on with the overlay?

Gadgetoid, 2017-06-09:
I have finished a PWM backlight specific overlay which gets it running on the Pi 3 and Pi Zero W (the two platforms I've so far tested). I'm seriously out of my depth, though, so this may be much more complex than is necessary:

```
/dts-v1/;
/plugin/;

/ {
    compatible = "brcm,bcm2835", "bcrm,bcm2708";

    fragment@0 {
        target = <&gpio>;
        __overlay__ {
            pwm_pins: pwm_pins {
                brcm,pins = <18 19>;
                brcm,function = <2 2>;
            };
        };
    };

    fragment@1 {
        target = <&clk_pwm>;
        __overlay__ {
            // Rename the fixed "pwm" clock to avoid a clash
            clock-output-names = "fake_pwm";
        };
    };

    fragment@2 {
        target = <&pwm>;
        __overlay__ {
            #clock-cells = <1>;
            clocks = <&cprman 30>;  /* 30 should be BCM2835_CLOCK_PWM */
            assigned-clocks = <&cprman 30>;
            assigned-clock-rates = <10000000>;
            pinctrl-names = "default";
            pinctrl-0 = <&pwm_pins>;
            status = "okay";
        };
    };

    fragment@3 {
        target = <&cprman>;
        __overlay__  {
            status = "okay";
        };
    };

    fragment@4 {
        target-path = "/";
        __overlay__ {
            rpi_backlight: rpi_backlight {
                compatible = "pwm-backlight";
                brightness-levels = <0 6 8 12 16 24 32 40 48 64 96 128 160 192 224 255>;
                default-brightness-level = <8>;
                pwms = <&pwm 1 5000000 0>;
                //enable-gpios = <&gpio 19 1>;
               status = "okay";
            };
        };
    };

     __overrides__ {
        pin   = <&pwm_pins>,"brcm,pins:0";
        pin2  = <&pwm_pins>,"brcm,pins:4";
        func  = <&pwm_pins>,"brcm,function:0";
        func2 = <&pwm_pins>,"brcm,function:4";
    };
};
```

We'll most likely be putting together a pair of overlays to generally support HyperPixel- one `hyperpixel-pwm-backlight.dtbo` and on`hyperpixel-gpio-backlight.dtbo` but are currently using separate overlays for the backlight and display.

On a related note, with audio enabled on the Pi, but switched out to HDMI the PWM backlight module does not work at all. Attempting to power it on will clamp the pin low- turning off the backlight- and nothing seems to bring it back into any sane state. It appears that the audio driver, even when it's outputting to HDMI, plays havoc with the PWM registers.  I'll be damned if I can find the specific module source that would give me any insight into why, though.

pelwell, 2017-06-12:
It looks like you've written this overlay against the 4.4 tree - the clk_pwm node and label don't exist in the 4.9 tree, which will prevent the overlay from loading. To handle both cases you could change `target = <&pwm_clk>;` to `target-path = "/soc/clocks/clock@4";` - any missing nodes in the path will be created as necessary - but a 4.9-based Raspbian is imminent, so you could make it a requirement.

The Pi analogue audio output (i.e. the 3.5mm socket) is driven from the PWM peripheral, so it is no surprise that enabling it interferes with other PWM users. The "bcm2835" sound driver sends PCM audio to the VPU cores, which then route it to either HDMI or the PWM peripheral, which is why you wouldn't have found the driver module. They are mapped to GPIOs 40&41 or 40&45, depending on the Pi model. `raspi-gpio get` shows you the current pin functions.

Gadgetoid, 2017-06-26:
Ah, the VPU cores handling the audio output explains a lot about my various woes with trying to stop analog audio clobbering the PWM peripheral, even when it's not actively in use. Drat! And thank you.

Your comments regarding 4.4 vs 4.9 tree go somewhat over my head, but since I now have a Pi up and running with the mentioned imminent 4.9-based Raspbian I have something I can test against, get it up and running and report back. Thank you!

JamesH65, 2017-09-13:
@gadgetoid Anything further required on this issue or can it be closed.

Gadgetoid, 2017-09-20:
I think it's still valid- but it appears you're waiting for me to finish a working overlay before including both the module and the overlay?

pelwell, 2017-09-20:
Yes.

JamesH65, 2017-12-04:
@Gadgetoid Any progress on this?

Gadgetoid, 2017-12-10:
Sorry- not yet! It's a mixture of "I have no idea what I'm going" and "I have no time to find out." It's on my list.

Presumably the overlay would have to to be suitably generic, too?

closed 2019-09-18 by JamesH65