raspberrypi/linux #2066: Request: Enable gs_usb module (USB-CAN-Adapter)

HubertD, 2017-06-12:
The gs_usb kernel module (/drivers/net/can/usb/gs_usb.c) provides a SocketCAN interface for the "Geschwister Schneider USB/CAN devices" and also some other open hardware / open firmware USB-CAN adapters like the bytewerk.org candleLight and candleZero as well as the "cantact" and "canable" hardware when using the candleLight firmware.

As SocketCAN is included in the raspbian default kernel and the gs_usb module is enabled in popular desktop distributions (e.g. Ubuntu), I would love to see the module enabled in future raspbian default kernels.

This should be as easy as setting "CONFIG_CAN_GS_USB = m".

References:
http://schneidersoft.net/en//node/2
https://github.com/HubertD/candleLight
https://github.com/HubertD/candleZero
https://github.com/HubertD/candleLight_fw
http://linklayer.github.io/cantact/
http://canable.io/

popcornmix, 2017-06-12:
Looks okay to me.

popcornmix, 2017-06-12:
Will be in next rpi-update kernel. See: https://github.com/raspberrypi/linux/commit/737cc547080537b99f82291e8ed90075d34e2688

HubertD, 2017-06-13:
great! thank you very much!

closed 2017-06-13 by HubertD

popcornmix, 2017-06-13:
This is in latest rpi-update kernel.