raspberrypi/linux #2073: Current lite image is running kernel 4.4.50, but only has package for 4.9.24+ installed

knarrff, 2017-06-15:
The current (2017-04-10-raspbian-jessie-lite) lite image boots kernel 4.4.50, but has the package for 4.9.24+ installed. Besides being out of sync with the non-lite version, this leaves the kernel without matching kernel modules in /usr/lib.

Is there a reason, why the lite version was built using 4.4.50?

XECDesign, 2017-06-16:
What do you mean by 'has the package for 4.9.24+ installed'? I have just booted installed and booted raspbian lite and the content of /lib/modules/ matches the output of `uname -a`. Did you by any chance run `apt-get dist-upgrade` (which would upgrade to 4.9.24) without rebooting (which would leave you without modules for the currently running kernel)?

knarrff, 2017-06-16:
On Fri, Jun 16, 2017 at 02:42:52AM -0700, XECDesign wrote:
> What do you mean by 'has the package for 4.9.24+ installed'? I have
> just booted installed and booted raspbian lite and the content of
> /lib/modules/ matches the output of `uname -a`. Did you by any [...]

Hi

Thanks for a quick answer. I did not run dist-upgrade. I'll check later 
today from a fresh image. The only upgrade I run is a regular 'apt-get 
update; apt-get upgrade', as part of the actual installation. Maybe this 
upgraded to the new kernel and didn't update the actual boot image (I 
did reboot). This should be reproducible, as this is an automatic image 
installation. I'll get back to you soon.

thanks, Frank

XECDesign, 2017-06-16:
`apt-get upgrade` would have the same behaviour in this case. If you skip all of that, you should be able to verify that raspbian lite ships with an appropriate kernel and modules.

knarrff, 2017-06-16:
Indeed, it ships with matching kernel and modules. 'apt-get upgrade' installs the newer kernel, in the process removing the modules for the still-running kernel. For some reason (and contrary of what I though was what happened) I didn't reboot, leaving the system in that state.

In this sense, I am closing the report.

It is a little disconcerting that 'apt-get upgrade' can leave you in such a state. It makes 'apt-get upgrade' an unsafe operation, best avoided in an automated script without explicit check if a restart is needed. Is there a reason this isn't handled the 'Debian-way' by using a meta-package that always depends on the latest image? This way, 'apt-get upgrade' would install new kernels, and the system would boot into them when rebooted, but the old kernel packages will not be removed, so that there is no immediate need to reboot.

closed 2017-06-16 by knarrff