raspberrypi/linux #2075: 4.9.y: USB serial CH340 driver bug: data size=5

Aydan, 2017-06-16:
[this](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/drivers/usb/serial/ch341.c?id=7c61b0d5e82bfe29b8dda55745afbf65b6ccc901) change is "missing" from the 4.9 kernel and it makes my CH340 based USB serial adapter non-functional.
seems the data size is set to 5 bits, and it's not possible to change it.
I tried via pyserial and stty, set data size to 7 bits but still only get the 5 LSBs correctly.

My Kernel is
`Linux raspberrypi3 4.9.24-v7+ #993 SMP Wed Apr 26 18:01:23 BST 2017 armv7l GNU/Linux
`
on a raspberry pi 3
the adapter works in windows

pelwell, 2017-06-20:
The patch you link to is in 4.11 It backports trivially to 4.10, but because of several other changes there are some tricky conflicts when backporting to 4.9. By the time I'd finished patching it up it was identical to the 4.10.y version.

Please `sudo rpi-update` to `4.9.33-v7+`, copy [this module file](https://drive.google.com/file/d/0B_P-i4u-SLBXWE16MEZUV0FNLVk/view?usp=sharing) into `/lib/modules/4.9.33-v7+/kernel/drivers/usb/serial`, and reboot. If it works as you expect it to then we'll merge the backport.

Aydan, 2017-06-20:
Works for me :-), thanks for the quick fix.

pelwell, 2017-06-20:
Cool. I'll get this into the kernel tree, then a firmware build.

popcornmix, 2017-06-22:
Latest rpi-update kernel contains @pelwell's latest fix.

Aydan, 2017-06-23:
confirmed working on fresh raspbian lite install

closed 2017-06-23 by pelwell