raspberrypi/linux #2076: Pi Zero USB OTG Ethernet suspended during RTP stream of Pi Camera

marty-sullivan, 2017-06-18:
I have my Pi Zero set up to stream via raspicam + gstreamer RTP over USB OTG Ethernet to another machine. After the stream runs fine for a few minutes, the connection is suspended (RTP, SSH, everything) and the logs show this message:

dwc2 20980000.usb: dwc2_hsotg_ep_sethalt(ep cea91610 ep1in, 0)

Why is the driver suspending the connection?

lategoodbye, 2017-06-19:
Hi Marty,
could you please provide more information (kernel version, used gadget module, steps to reproduce).
Usually i would point you to linux-usb kernel mailing, but my recent dwc2 questions didn't get a reply.

Is it possible to reproduce (without raspicam) this with my latest [changes](http://lists.infradead.org/pipermail/linux-rpi-kernel/2017-May/006349.html) in [linux-next](https://git.kernel.org/pub/scm/linux/kernel/git/next/linux-next.git) (upstream)?

marty-sullivan, 2017-06-19:
I'm using the latest kernel available via rpi update, I believe 4.9.33+ if I'm remembering right. Although, the problem seems to also happen in 4.4.50+.

I am using the g_ether gadget module, and the USB host is a  15-inch Macbook Pro Late-2014.

I set up the module and the internet forwarding, etc. using the guide available at Adafruit: https://learn.adafruit.com/turning-your-raspberry-pi-zero-into-a-usb-gadget/ethernet-gadget

Can you provide some info on what this dwc2_hsotg_ep_sethalt means? I'm thinking it's actually the Macbook suspending the connection, but without knowing enough about what's happening, I don't know what end the problem is coming from. (Really hoping it's the Macbook haha)

lategoodbye, 2017-06-19:
I looked at the dwc2 gadget source code and the output is only a info that for a USB endpoint the halt should be cleared. So i can't see any connection to a suspending process.

I suggest to narrow down the issue by testing the following cases:
  - test the same scenario as above with a Windows or a Linux PC
  - instead of the complex scenario with raspicam and gstreamer simply make ping for a long duration

Additional questions:
  - How do get the message log after the network layer is broken?
  - Do you see any difference with ifconfig on the RPI Zero before and after the issue?
  - Does your Macbook runs from battery or AC?

marty-sullivan, 2017-06-19:
Thanks for the info. I will test with a Linux machine this evening. I have left my SSH connection open for long periods without issue, so I will try leaving something like iperf running for a long period. 

For your questions:

- I get the log message by unplugging the pi and rebooting.
- I can't see into the Pi after the issue without rebooting with my current setup because I am using Pi Zero 1.3, so no way to get in or connect a keyboard without disconnecting the USB OTG. I have ordered a few Pi Zero W, so if I continue seeing this error there, I will be able to log in via WiFi.
- Macbook is on AC power and set not to sleep. I am also live-viewing on the Macbook as the stream is broken.

lategoodbye, 2017-06-19:
iperf is a good idea. Using an additional WiFi connection could influence the test scenario.

So you don't have a USB to serial adapter in order to connect on GPIO 14/15?

marty-sullivan, 2017-06-19:
I don't have an adapter, but sounds like a good tool to have so I just ordered one. Thanks for the suggestion.

marty-sullivan, 2017-06-19:
I found some time to run iperf for an hour pushing UDP traffic over the usb ethernet. It ran to completion, so I'm thinking the problem is likely on the end of my streaming logic... Although, the problem doesn't seem to happen when I am using a WiFi dongle instead of USB.

I'll still test on Linux later to rule out the Macbook as the problem.

JamesH65, 2017-09-13:
@marty-sullivan Can this issue now be closed?

marty-sullivan, 2017-09-13:
I lost track of this with other projects, but I'm going to assume the issue was with the Macbook suspending the USB session or a similar problem. I didn't seem to see the problem elsewhere, although I wasn't able to test much further.

closed 2017-09-13 by marty-sullivan