raspberrypi/linux #2078: Make raspberry header of HDMI console customizable

MaxValue, 2017-06-20:
Currently, if you do not have the graphical user-interface enabled and attach a screen to the HDMI port, you can interact with your Pi.
You will also see that at the top there is a header consisting of 4 stylized Raspberries.

It would be really nice to being able to customize this header image. Currently this is not possible, because the image is compiled _into_ the kernel. [You would have to recompile the kernel to change it](https://www.raspberrypi.org/forums/viewtopic.php?f=31&t=11642).

As recompiling the linux kernel [is more or less a hairy task](http://www.arm9board.net/wiki/index.php?title=Create_your_own_boot_logo), it would be much more user friendly to just provide a path to a .ppm file inside the boot.txt file, reboot the Pi and see the new logo.

Is this possible to implement (technically)?

popcornmix, 2017-06-20:
Anything is possible, but it is not a trivial amount of work and will benefit very few people.
I don't imagine this is something that we will be doing.

If it is important to you then learning how to build the kernel is probably best.

MaxValue, 2017-06-20:
Ok. Thanks for the fast response!

closed 2017-06-20 by popcornmix