raspberrypi/linux #2080: Checkout kernel source

GigaSchmiede, 2017-06-22:
Hey guys,

i have to build the kernel with version 4.9.27 and rt-preempt.
When i clone the branch 4.9.y and checkout 4.9.27 there a files lost e.g. mkknlimg...

Why are the files in 4.9.y and not in 4.9.27 but in the past as 4.9.27 was 4.9.y they are?

pelwell, 2017-06-22:
You should be using the `rpi-4.9.y` branch from the `git@github.com:/raspberrypi/linux.git` repo, otherwise you won't be getting the Pi-specific extensions and enhancements.

Having said that, we no longer recommend using `mkknlimg` - just copy the output `zImage` to `kernel.img` or `kernel7.img`.

popcornmix, 2017-06-22:
Make sure you are using the 4.9.27 commit from raspberrypi/rpi-4.9.y tree not the v4.9.27 tag from the upstream kernel tree which won't have Pi specific commits in.
e.g. 
```
git checkout 89f3b8d5f264d5dab9818c6667c71e3cc55b13f5
```

GigaSchmiede, 2017-06-22:
Thanks 👍 
My mistake was the wrong tree like @popcornmix  said.

closed 2017-06-22 by GigaSchmiede