raspberrypi/linux #2081: USB keyboard on single-process mode

vanfanel, 2017-06-22:
Hi,

For a particular project, I need the Pi to run in single-process mode ("init=/bin/bash" in config,txt, for example, in this case it's our own program).
The thing is, UDEV isn't running obviously, so USB keyboard is not working in this mode.
In theory, it should be possible to have USB devices working in single-process mode with initramfs hooks, as initramfs is used for these early tasks that need to be done before giving the control to init.
However, this is not documented anywhere so I am totally lost here.
So, what initramfs hooks should I add, and where should I add them?
I know I should run update-initramfs after adding them, but that's all I can find.

Thanks

pelwell, 2017-06-22:
This is not the right place to ask for general advice like this - try the [Advanced Users forum](https://www.raspberrypi.org/forums/viewforum.php?f=29).

closed 2017-06-22 by pelwell

vanfanel, 2017-06-22:
Sorry about that, pelwell