raspberrypi/linux #2082: Multiple USB webcams stop working after a few minutes of use

tonystuart, 2017-06-22:
I have a Raspberry Pi model 3 with three USB webcams, a USB keyboard and a USB mouse. The 3 webcams are plugged directly into the Raspberry Pi. The USB keyboard and mouse are plugged in via a four port hub.

I am using `motion`to stream the webcam video. Everything works fine for anywhere from about 30 seconds to a few minutes. Then the webcams, keyboard and mouse stop working. The following message appears repeatedly in the dmesg output:

```
uvcvideo: Failed to set UVC probe control : -110 (exp. 26).
```

See below for the full dmesg and lsusb -v output .

I have tried a number of resolutions and frame rates, but the results are the same. The `motion` command is configured to use three threads, one per camera. The CPU utilization for `motion` is about 18%, mostly allocated to one of the four cores. 

I see that issue #618 is similar and I have tried the following things:

- Setting uvcvideo quirks to 0x100 in /etc/modprobe.d/uvcvideo.conf
- Adding dwc_otg.fiq_fsm_enable=0 to /boot/cmdline.txt and rebooting
- Replacing dwc_otg.fiq_fsm_enable=0 with dwc_otg.fiq_fsm_mask=0x3 and rebooting
- Enabling uvcvideo trace level 0x408

However, none of these resolve the issue. Any thoughts on what else to try?

Here is the lsusb -v output:

```
Bus 001 Device 008: ID 1908:2310 GEMBIRD 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 ?
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x1908 GEMBIRD
  idProduct          0x2310 
  bcdDevice            1.00
  iManufacturer           1 Generic
  iProduct                2 USB2.0 PC CAMERA
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength          449
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              256mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         2
      bFunctionClass         14 Video
      bFunctionSubClass       3 Video Interface Collection
      bFunctionProtocol       0 
      iFunction               2 USB2.0 PC CAMERA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass        14 Video
      bInterfaceSubClass      1 Video Control
      bInterfaceProtocol      0 
      iInterface              2 USB2.0 PC CAMERA
      VideoControl Interface Descriptor:
[... 1928 lines cut ...]
[ 1149.047185] smsc95xx 1-1.1:1.0 eth0: Failed to read MII_BMSR
[ 1249.329415] uvcvideo: Trying format 0x32315559 (YU12): 320x240.
[ 1249.329429] uvcvideo: Using default frame interval 33333.3 us (30.0 fps).
[ 1254.887194] uvcvideo: Failed to set UVC probe control : -110 (exp. 26).
[ 1254.887573] uvcvideo: Trying format 0x56595559 (YUYV): 320x240.
[ 1254.887585] uvcvideo: Using default frame interval 33333.3 us (30.0 fps).
[ 1259.927218] uvcvideo: Failed to set UVC probe control : -110 (exp. 26).
[ 1259.930885] uvcvideo: Trying format 0x32315559 (YU12): 320x240.
[ 1259.930898] uvcvideo: Using default frame interval 33333.3 us (30.0 fps).
[ 1264.967196] uvcvideo: Failed to set UVC probe control : -110 (exp. 26).
[ 1264.967604] uvcvideo: Trying format 0x56595559 (YUYV): 320x240.
[ 1264.967616] uvcvideo: Using default frame interval 33333.3 us (30.0 fps).
[ 1270.007163] uvcvideo: Failed to set UVC probe control : -110 (exp. 26).
[ 1270.009914] uvcvideo: Trying format 0x32315559 (YU12): 320x240.
[ 1270.009925] uvcvideo: Using default frame interval 33333.3 us (30.0 fps).
[ 1275.047175] uvcvideo: Failed to set UVC probe control : -110 (exp. 26).
[ 1275.047568] uvcvideo: Trying format 0x56595559 (YUYV): 320x240.
[ 1275.047578] uvcvideo: Using default frame interval 33333.3 us (30.0 fps).
[ 1280.087171] uvcvideo: Failed to set UVC probe control : -110 (exp. 26).
[ 1288.333077] uvcvideo: Trying format 0x32315559 (YU12): 320x240.
[ 1288.333092] uvcvideo: Using default frame interval 33333.3 us (30.0 fps).
[ 1293.927172] uvcvideo: Failed to set UVC probe control : -110 (exp. 26).
[ 1293.927557] uvcvideo: Trying format 0x56595559 (YUYV): 320x240.
[ 1293.927568] uvcvideo: Using default frame interval 33333.3 us (30.0 fps).
```

P33M, 2017-06-23:
Back-of-the-envelope calculation of bandwidth usage:

3072 bytes = 24576 bits per camera per microframe (negating bit-stuff overhead and protocol overhead of a few bytes)
24576 x 3 = 73728 bits per microframe
73728 x 8000 = 589x10^6 bits per second
USB2.0 has a 480mbit/s raw data rate. I suspect it works briefly because each camera doesn't always send the maximum amount of data per microframe.

The bug in this case is with the bandwidth reservation code - it should reject transfers that could exceed the maximum periodic bandwidth reservation for USB2.0.

JamesH65, 2017-09-13:
@p33m Is the bandwidth reservation code down to us? Or should this be closed?

towen, 2022-08-28:
There is a long-standing issue over on the Digilent forums regarding the Analog Discovery 2 not working correctly on raspberry pi 3. As another high-bandwidth USB2 device, I wonder if this could be the root cause?

https://forum.digilent.com/topic/1713-analog-discovery-2-vs-raspberry-pi-3