raspberrypi/linux #2083: Has ipvlan been removed?

alexellis, 2017-06-23:
Hi, I was doing some testing with folks in the Docker community and we found the `ipvlan` Kernel module   disappears when we do an `rpi-update`. Apparently Justin Cormack from Docker Inc raised an issue here last year asking for the module to be included.

Please could it be reintroduced to the Kernel? At the moment Docker Swarm will break if we do an `dpi-update`.

Thanks,

Alex

alexellis, 2017-06-23:
I can't find the original issue or PR - but I saw these which are related:

https://github.com/raspberrypi/linux/issues/1612 @grigorig
https://github.com/raspberrypi/linux/issues/1809 @DieterReuter

remonlam, 2017-06-24:
Apparently it is broken for some time.

Looking at the commit logs over at [rpi-firmware](https://github.com/Hexxeh/rpi-firmware/tree/master) it appears that kernel [4.4.50-v7+](https://github.com/Hexxeh/rpi-firmware/tree/52241088c1da59a359110d39c1875cda56496764/modules/4.4.50-v7%2B/kernel/drivers/net) is the last one that has support for IPVLAN, the next kernel version [4.9.11-v7+](https://github.com/Hexxeh/rpi-firmware/tree/3e1c3e9ba2b948e6023114b9633223363da78679/modules/4.9.11-v7%2B/kernel/drivers/net) does not have IPVLAN support enabled.

If I'm correct when running `rpi-update` by default, it's using the [next branch](https://github.com/Hexxeh/rpi-firmware/tree/next/modules/4.9.13-v7%2B/kernel/drivers/net), however this is also without IPVLAN support enabled.

JamesH65, 2017-09-13:
@popcornmix Was this omitted deliberately or can we re-add it?

wedi, 2018-03-28:
Are there any news about this?

popcornmix, 2018-03-29:
I assume this is ``CONFIG_IPVLAN``?
It depends on ``CONFIG_NET_L3_MASTER_DEV`` which isn't currently enabled.
That's a bool so does add some extra core code, but doesn't look too much.
@pelwell any objections?

pelwell, 2018-03-29:
No objections - go for it.

alexellis, 2018-03-29:
Thanks guys. Any idea why this took 9 months for an actionable response?

pelwell, 2018-03-29:
Because although this is probably the most important issue in the world to you, we have lots of other issues to look at, and occasionally one gets forgotten.

pelwell, 2018-03-29:
Why did it take a third party to remind us?

wedi, 2018-04-11:
Thank you very much for your quick response. Works like a charm. :+1:

closed 2018-04-11 by popcornmix