raspberrypi/linux #2084: Unable to lowering the raspberry pi Bluetooth power

fafa92, 2017-06-24:
I'm trying to turn my raspberry to an iBeacon but I cannot make it transmit with lower power, I've changed Tx power many times, when I use my BLE scanner I see that Tx power has changed but RSSI hasn't at all while with other beacon devices changing transmission power leads to a lower powered measure. Has anyone had the same problem? this is the command that I run.(I've changed C8 to 88, CE, E7 and etc.)
sudo hcitool -i hci0 cmd 0x08 0x0008 1E 02 01 1A 1A FF 4C 00 02 15 63 6F 3F 8F 64 91 4B EE 95 F7 D8 CC 64 A8 63 B5 00 00 00 00 C8

gilvader, 2017-08-03:
This is only changing the content of the advertising data, NOT the actual TX power but rather "Expected received Signal strength expected at 1m distance" as Apple has specified their manufacturer specified data.

JamesH65, 2017-09-13:
Has this been sufficiently answered? Can it be closed?

ping @fafa92 @gilvader

closed 2017-12-04 by JamesH65