raspberrypi/linux #2087: brcmfmac: brcmf_inetaddr_changed: fail to get arp ip table err:-23

wschlich, 2017-06-26:
I'm using an RPi 2 B with the official RPi wifi dongle and the latest Raspbian with Kernel 4.9.33-v7+.

I get this message:
[code]
brcmfmac: brcmf_inetaddr_changed: fail to get arp ip table err:-23
[/code]

First, wifi worked well for some time.

Then, after 2 days, the RPi had a ping response time of around 1000msec and SSH login was very sluggish.

A simple "ifdown wlan0; ifup wlan0" made everything fine again, but it also made the error message reappear.
[code]
root@raspberrypi:~# dmesg  | grep brcm
[    5.511009] usbcore: registered new interface driver brcmfmac
[    5.831855] brcmfmac: Firmware version = wl0: Apr  3 2014 04:43:32 version 6.10.198.66 (r467479) FWID 01-32bd010e
[    7.152127] brcmfmac: power management disabled
[   13.744410] brcmfmac: brcmf_inetaddr_changed: fail to get arp ip table err:-23
[236056.551568] brcmfmac: brcmf_inetaddr_changed: fail to get arp ip table err:-23
[236074.832915] brcmfmac: brcmf_inetaddr_changed: fail to get arp ip table err:-23
root@raspberrypi:~# 
[/code]

Any idea what this error message means and how the issue can be fixed?

Any idea why the wifi connection was experiencing these connectivity issues?

Cheers,
Wolfram

JamesH65, 2017-06-26:
Thanks for the report. A quick google on the error message shows this is not only seen on Pi, and some comments indicate this is a reception issue. Have you changed anything in your wifi network topology that could have caused a drop in signal strength?

wschlich, 2017-06-28:
@JamesH65 : Well, reception couldn't be better:
```
root@raspberrypi:~# ifdown wlan0; sleep 5; ifup wlan0
root@raspberrypi:~# dmesg | grep brcmfmac | tail -n 4
[190172.003117] brcmfmac: power management disabled
[190172.701886] brcmfmac: brcmf_inetaddr_changed: fail to get arp ip table err:-23
[190177.110271] brcmfmac: power management disabled
[190184.273144] brcmfmac: brcmf_inetaddr_changed: fail to get arp ip table err:-23
root@raspberrypi:~# iwconfig wlan0 | grep Quality
          Link Quality=70/70  Signal level=-31 dBm  
root@raspberrypi:~# 
```
So, even with PM disabled and full signal, the error message appears.

Johnny-quattro, 2017-07-11:
I can report the same problem:

Type: Pi 2, Revision: 01, Memory: 1024MB, Maker: Sony
  *--> Raspberry Pi 2 Model B Rev 1.1

C3PI5:/home/pi $ iwconfig wlan0 | grep Quality
          Link Quality=67/70  Signal level=-43 dBm

C3PI5:/home/pi $ dmesg -T | grep brcmfmac
[Tue Jul 11 01:03:23 2017] usbcore: registered new interface driver brcmfmac
[Tue Jul 11 01:03:23 2017] brcmfmac: Firmware version = wl0: Apr  3 2014 04:43:32 version 6.10.198.66 (r467479) FWID 01-32bd010e
[Tue Jul 11 01:03:26 2017] brcmfmac: power management disabled
[Tue Jul 11 01:03:37 2017] brcmfmac: brcmf_inetaddr_changed: fail to get arp ip table err:-23

wschlich, 2017-07-27:
@JamesH65 any news on this issue? :)

JamesH65, 2017-07-27:
I've not been looking at it I am afraid. I can only suggest trying some of the bleeding edge kernels (4.13) to if any of the recent changes in there have made a difference. You will need to build the kernel yourself though, which may not be something you want to do.

wschlich, 2017-07-27:
@JamesH65 Right :) I'm using Gentoo for over a decade, but I'm not into compiling stuff for and on my RasPis, tbh =) Any idea when a newer kernel will be released for raspbian?

wschlich, 2017-07-27:
@JamesH65 no, nothing unusual here.
Original RPi WiFi USB adapter directly plugged into an RPi 2 B with a sufficiently powerful power supply. 
WiFi connected to a TP Link Archer C7 running the very recent LEDE version 17.01.2 (OpenWrt fork).

wschlich, 2017-07-27:
@JamesH65 

I believe this is what fails:
https://github.com/torvalds/linux/blob/master/drivers/net/wireless/broadcom/brcm80211/brcmfmac/core.c#L769

```
	/* retrieve the table from firmware */
	ret = brcmf_fil_iovar_data_get(ifp, "arp_hostip", addr_table,
				       sizeof(addr_table));
	if (ret) {
		brcmf_err("fail to get arp ip table err:%d\n", ret);
		return NOTIFY_OK;
	}
```
Where does the brcmfmac firmware come from?

wschlich, 2017-07-27:
@JamesH65 The error still occurs with the latest rpi-update installed:

```
root@raspberrypi:~# dmesg | grep brcmfmac
[    5.788228] usbcore: registered new interface driver brcmfmac
[    6.056063] brcmfmac: Firmware version = wl0: Apr  3 2014 04:43:32 version 6.10.198.66 (r467479) FWID 01-32bd010e
[   19.192132] brcmfmac: power management disabled
[   19.307496] brcmfmac: power management disabled
[   30.124226] brcmfmac: brcmf_inetaddr_changed: fail to get arp ip table err:-23
root@raspberrypi:~# 
```

wschlich, 2017-07-27:
Hmm, the firmware for my BCM43143 seems to come from the firmware-brcm80211 package. Also seems that the file `/lib/firmware/brcm/brcmfmac43143.bin` hasn't changed since 06/2014:

https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/log/brcm/brcmfmac43143.bin

JamesH65, 2017-07-27:
brcmfmac is the Broadcom linux driver for the Wifi chip. We are currently talking to Cypress re: newer firmware, along with some bug fixes. Cypress bought the chip from Broadcom, so its all a bit complicated.
(The chip is a 43430 IIRC)

wschlich, 2017-07-27:
@JamesH65 My device has the USB IDs `0a5c:bd1e`. According to https://wireless.wiki.kernel.org/en/users/drivers/brcm80211 it is a BCM43143. No? :)

JamesH65, 2017-07-27:
BRCM43438 (from the 4343x family)

wschlich, 2017-07-27:
@JamesH65 Yes, the USB dongle :-)
Ah, 43438 is the RPi 3 one w/ BT, right?
Would be great to get a better firmware for the USB dongle... for users of older Pis...

JamesH65, 2017-07-27:
I'll look in to that - not sure if the firmware is built in to the dongle
or if it is loaded at run time. If the former, then I doubt it can be
upgraded. If the latter, I'll try and find out more details.

JamesH65, 2017-07-27:
OK, we think it's loaded at runtime. I'll see if we can get newer firmware.

6by9, 2017-07-27:
The normal upstream repo for firmware is at https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/
Checking in https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/log/brcm/brcmfmac43143.bin says that there has only been one commit (June 2014), so currently it appears to not have been updated since then.

wschlich, 2017-07-28:
@JamesH65, @6by9 Exactly what I wrote before... :)

JamesH65, 2017-09-13:
I'm trying to remember if we have heard back from Cypress re: firmware for the dongle. Memory has failed! I'll try and get more information direct from them.

jmnel, 2018-02-17:
Has any progress been made to this issue? I see this message on my Macbook Pro with kernel 4.15.

`04:00.0` Network controller: Broadcom Limited BCM43602 802.11ac Wireless LAN SoC (rev 01)`

JamesH65, 2018-02-19:
Confused. What distro are you using on your Macboox pro? This thread is specifically about the Raspberry Pi USB weifreless dongle, so I am not sure of the relevence.

dickontoo, 2018-02-19:
I've just hit this problem on a non-Pi 32b ARM machine -- development project for work -- using a Pi dongle.  It reliably errors with that error message, and won't associate.  4.12 kernel.  No idea if that helps...

JamesH65, 2018-02-19:
@dickontoo Any idea of which dongle firmware you are using?

dickontoo, 2018-02-19:
It reports:

```crate login: [  754.729295] usb 1-1: new high-speed USB device number 2 using orion-ehci
[  754.922674] usb 1-1: New USB device found, idVendor=0a5c, idProduct=bd1e
[  754.929423] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  754.936602] usb 1-1: Product: Remote Download Wireless Adapter
[  754.942471] usb 1-1: Manufacturer: Broadcom
[  754.946674] usb 1-1: SerialNumber: 000000000001
[  755.271534] brcmfmac: brcmf_c_preinit_dcmds: Firmware version = wl0: Apr  3 2014 04:43:32 version 6.10.198.66 (r467479) FWID 01-32bd010e
[  755.322069] usb 1-1 wlx086a0a975a64: renamed from wlan0
```

when I enable the USB port.

JamesH65, 2018-02-19:
I've emailed Cypress to see if they have more recent firmware - 2014 seems a bit old. We have recently got updated FW for the onboard device, so hopefully they have something for this too.

wschlich, 2018-05-04:
@JamesH65 any response from Cypress to your mail from February?!

JamesH65, 2018-05-04:
Can't remember. I need to do some Cypress stuff this afternoon - will take a look.

wschlich, 2018-05-08:
@JamesH65 Any luck so far? :)

wschlich, 2018-05-09:
@JamesH65 Thank you!

deblauwetom, 2018-06-07:
I would also be verrrry interested :)

JamesH65, 2018-06-08:
We are actively working with Cypress on this, and trying to get a recent dongle firmware (that works)!

joxer, 2018-06-18:
Hi! Do you have any updates? I've got the same error on this wifi dongle I bought. This is branded as Official RPI wifi dongle, but it seems to not be so good as it seems.

JamesH65, 2018-07-13:
There is now beta driver firmware available for the Official Dongle. Please see this thread  https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=187256&start=25#p1340347

However, I do not think it will 'fix' this arp error message, although I suspect it's actually benign, and may only occur under low reception conditions.

deblauwetom, 2018-07-13:
It works here, I tested the AP mode.

However, it still is not acting properly after a watchdog reboot, soooo, it is not worse than the 2014 firmware ;)

kind regards,
Tom,

JamesH65, 2018-07-13:
@deblauwetom Waht's is the problem with watchdog reset?  Don't remember seeing anything about that in this issue.

deblauwetom, 2018-07-13:
Sorry that I replied about this in this issue, but the forum thread on raspberrypi.org <http://raspberrypi.org/> is locked, and I wanted to reply there in the first place but could not :)

But the problem with the watchdog is this: https://www.raspberrypi.org/forums/viewtopic.php?f=98&t=216240 <https://www.raspberrypi.org/forums/viewtopic.php?f=98&t=216240>

Best regards,
Tom,

akiernan, 2018-07-13:
I'm still seeing the ARP message (which I'd agree looks like it's just cosmetic). The extreme packet latency when I was connecting to an Ubiquiti AP is gone and I no longer have to block it from those, so this looks like a big improvement!

andrum99, 2018-08-05:
I'm seeing this bug on the official Raspberry Pi wifi adaptor as well:

`[613007.105332] brcmfmac: brcmf_inetaddr_changed: fail to get arp ip table err:-23
[613062.460124] brcmfmac: brcmf_inetaddr_changed: fail to get arp ip table err:-23
[613066.490755] brcmfmac: brcmf_inetaddr_changed: fail to get arp ip table err:-23
[613067.055044] brcmfmac: brcmf_inetaddr_changed: fail to get arp ip table err:-23
[613075.684197] brcmfmac: brcmf_inetaddr_changed: fail to get arp ip table err:-23
[613269.174398] brcmfmac: brcmf_inetaddr_changed: fail to get arp ip table err:-23
[613272.204291] brcmfmac: brcmf_inetaddr_changed: fail to get arp ip table err:-23
[613272.818290] brcmfmac: brcmf_inetaddr_changed: fail to get arp ip table err:-23
[942895.987961] systemd: 32 output lines suppressed due to ratelimiting
[1636595.676930] brcmfmac: brcmf_inetaddr_changed: fail to get arp ip table err:-23
[1636656.994104] brcmfmac: brcmf_inetaddr_changed: fail to get arp ip table err:-23
[1636661.301159] brcmfmac: brcmf_inetaddr_changed: fail to get arp ip table err:-23
[1636661.921151] brcmfmac: brcmf_inetaddr_changed: fail to get arp ip table err:-23
[1636677.970549] brcmfmac: brcmf_inetaddr_changed: fail to get arp ip table err:-23
[1637711.974910] brcmfmac: brcmf_inetaddr_changed: fail to get arp ip table err:-23
[1777259.879331] brcmfmac: brcmf_inetaddr_changed: fail to get arp ip table err:-23
[1777269.534104] brcmfmac: brcmf_inetaddr_changed: fail to get arp ip table err:-23
[1780186.344044] brcmfmac: brcmf_inetaddr_changed: fail to get arp ip table err:-23
[1780197.724651] brcmfmac: brcmf_inetaddr_changed: fail to get arp ip table err:-23
[1780231.458370] brcmfmac: brcmf_inetaddr_changed: fail to get arp ip table err:-23
[1780241.594340] brcmfmac: brcmf_inetaddr_changed: fail to get arp ip table err:-23
[1780578.370638] brcmfmac: brcmf_inetaddr_changed: fail to get arp ip table err:-23
[1780588.594778] brcmfmac: brcmf_inetaddr_changed: fail to get arp ip table err:-23
[1780839.052178] brcmfmac: brcmf_inetaddr_changed: fail to get arp ip table err:-23
[1780847.756426] brcmfmac: brcmf_inetaddr_changed: fail to get arp ip table err:-23
[1781132.714396] brcmfmac: brcmf_inetaddr_changed: fail to get arp ip table err:-23
[1781140.844141] brcmfmac: brcmf_inetaddr_changed: fail to get arp ip table err:-23
[1781309.165828] brcmfmac: brcmf_inetaddr_changed: fail to get arp ip table err:-23
[1781315.683604] brcmfmac: brcmf_inetaddr_changed: fail to get arp ip table err:-23
[1793532.967755] brcmfmac: brcmf_inetaddr_changed: fail to get arp ip table err:-23
[1793544.975584] brcmfmac: brcmf_inetaddr_changed: fail to get arp ip table err:-23
`
Is there any update? If the message is benign then perhaps it should be changed to warn or debug. I won't be trying the beta firmware as I only have one of these adaptors and it's on a Pi that is acting as a network audio player for my dad, so can't risk beta on it.

Overc1ocker, 2018-08-30:
This is what i get when attempting to use the RPI official dongle. The messages dont exist in the 4.14 kernel, but do exist in 4.17, 4.18, and the RC1 of 4.19. Any comments?

```
[   14.321870] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43143 for chip BCM43143/2
[   14.321907] usbcore: registered new interface driver brcmfmac
[   14.625988] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43143 for chip BCM43143/2
[   14.626042] usb 2-6: Direct firmware load for brcm/brcmfmac43143.clm_blob failed with error -2
[   14.626046] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available
[   14.626608] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43143/2 wl0: Apr  3 2014 04:43:32 version 6.10.198.66 (r467479) FWID 01-32bd010e
[   23.147371] brcmfmac: brcmf_inetaddr_changed: fail to get arp ip table err:-52

```

Overc1ocker, 2018-08-30:
The interesting part is that the dongle still seems to work. Can these messages be ignored?

DougieLawson, 2019-02-03:
I've got the following errors from four of my various Raspberries

Model Zero falcon    4.14.94+ (with dongle)
```
[   23.396903] brcmfmac: brcmf_fw_map_chip_to_name: using brcm/brcmfmac43143.bin for chip 0x00a887(43143) rev 0x000002
[   23.397265] usbcore: registered new interface driver brcmfmac
[   24.875142] brcmfmac: brcmf_c_preinit_dcmds: Firmware version = wl0: Apr  3 2014 04:43:32 version 6.10.198.66 (r467479) FWID 01-32bd010e
[   24.876002] brcmfmac: brcmf_c_preinit_dcmds: CLM version = API: 10.0 Data: 6.7.8 Compiler: 1.15.2 ClmImport: 1.15.1 Inc Data: 7.10.12 Inc Compiler: 1.19.6 Inc ClmImport: 1.19.6
[   30.481906] brcmfmac: power management disabled
[   37.516911] brcmfmac: brcmf_inetaddr_changed: fail to get arp ip table err:-23
```

Model 3B+ hitchhiker 4.14.94-v7+
```
[    7.585375] brcmfmac: F1 signature read @0x18000000=0x15264345
[    7.595174] brcmfmac: brcmf_fw_map_chip_to_name: using brcm/brcmfmac43455-sdio.bin for chip 0x004345(17221) rev 0x000006
[    7.596235] usbcore: registered new interface driver brcmfmac
[    7.903713] brcmfmac: brcmf_c_preinit_dcmds: Firmware version = wl0: Feb 27 2018 03:15:32 version 7.45.154 (r684107 CY) FWID 01-4fbe0b04
[    7.904383] brcmfmac: brcmf_c_preinit_dcmds: CLM version = API: 12.2 Data: 9.10.105 Compiler: 1.29.4 ClmImport: 1.36.3 Creation: 2018-03-09 18:56:28
[    8.108403] brcmfmac mmc1:0001:1 wlxb827eb698d91: renamed from wlan0
[   10.630422] brcmfmac: power management disabled
[103858.556117] brcmfmac: brcmf_run_escan: error (-25)
[103858.556134] brcmfmac: brcmf_cfg80211_scan: scan error (-25)
```

Model 1B ranger    4.14.59+ (with Wifi dongle) - 171 days of uptime.
```[14129722.918439] brcmfmac: brcmf_inetaddr_changed: fail to get arp ip table err:-23```

Model 3A+ ulysses   4.19.17-v7+
```
[    4.974954] brcmfmac: F1 signature read @0x18000000=0x15264345
[    4.986083] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    4.986968] usbcore: registered new interface driver brcmfmac
[    5.312538] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    5.330935] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Feb 27 2018 03:15:32 version 7.45.154 (r684107 CY) FWID 01-4fbe0b04
[    6.447425] brcmfmac: power management disabled
[13518.282172] brcmfmac: brcmf_run_escan: error (-52)
[13518.282186] brcmfmac: brcmf_cfg80211_scan: scan error (-52)
```

All four of those are still connected to my network and don't exhibit any symptoms other than filling the ```dmesg``` log with noise.

stefson, 2019-02-20:
It would be great to have a firmware working with kernel 4.14, the one from firmware.git certainly does _not_: 
```Direct firmware load for brcm/brcmfmac43143.bin failed with error -2```

while with the beta firmware it seems kind of okish: 

```
[   12.199694] brcmfmac: brcmf_fw_map_chip_to_name: using brcm/brcmfmac43143.bin for chip 0x00a887(43143) rev 0x000002
[   12.522255] brcmfmac: brcmf_c_preinit_dcmds: Firmware version = wl0: Jan 29 2018 22:57:27 version 6.20.198.26 (r683471) FWID 01-32bd010e
[   12.522939] brcmfmac: brcmf_c_preinit_dcmds: CLM version = API: 10.0 Data: 6.7.8 Compiler: 1.15.2 ClmImport: 1.15.1 Inc Data: 7.8.9 Inc Compiler: 1.18.3 Inc ClmImport: 1.18.3 
```

Alirezaies, 2019-03-28:
> on my MacBookPro I install ubuntu desktop 18.04.1
> Linux MacBookPro 4.15.0-45-generic #48-Ubuntu SMP Tue Jan 29 16:28:13 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
> 
> [ 9.366002] usbcore: registered new interface driver brcmfmac
> [ 9.494733] brcmfmac: brcmf_fw_map_chip_to_name: using brcm/brcmfmac43602-pcie.bin for chip 0x00aa52(43602) rev 0x000001
> [ 9.495816] brcmfmac 0000:04:00.0: Direct firmware load for brcm/brcmfmac43602-pcie.txt failed with error -2
> [ 10.006693] brcmfmac 0000:04:00.0: Direct firmware load for brcm/brcmfmac43602-pcie.clm_blob failed with error -2
> [ 10.006695] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available(err=-2), device may have limited channels available
> [ 10.007342] brcmfmac: brcmf_c_preinit_dcmds: Firmware version = wl0: Nov 10 2015 06:38:10 version 7.35.177.61 (r598657) FWID 01-ea662a8c
> [ 10.057024] brcmfmac 0000:04:00.0 wlp4s0: renamed from wlan0
> [ 16.797117] brcmfmac: brcmf_inetaddr_changed: fail to get arp ip table err:-23

have same issue on my MacBook Pro 2015 with Ubuntu 18.04
I have a working wifi but this error causes my MacBook to not shutdown or restart, can you confirm the same issue?

JamesH65, 2019-03-28:
This is not a Macbook support repo, its for the Raspberry Pi SBC.

Alirezaies, 2019-03-29:
ok I'm sorry, but in my case it was `firewalld` that preventing to shutdown or restart so I just needed to replace it with `ufw` and everything was fine :) @mmh891113

wschlich, 2019-05-23:
@JamesH65 any news here?

JamesH65, 2019-05-23:
No, right down the list of priorities at the moment.

JoseHervas, 2019-06-06:
Don't know if this info is useful but... I have the same issue on my rpi 3b+ with Kali Linux with kernel 4.19.29, however same raspi with Raspbian and kernel version 4.19.42-v7+ works perfectly.

pinonpierre, 2019-06-27:
Exactly the same issue on XPS 15 9550 on Debian.

brcmfmac: brcmf_inetaddr_changed: fail to get arp ip table err:-52

tres-pitt, 2019-08-06:
Ubuntu 18.04 on MBP - `brcmfmac: brcmf_inetaddr_changed: fail to get arp ip table err:-52`. Can't boot - repeatedly asks me for Login credentials i can't crack (maybe I haven't tried enough options or I selected an unusual u/pw I"ve forgotten). looks like I'll have to wipe and revert to 16.04 LTS

xxauroraxx, 2019-09-04:
Raspberry Pi 3 Model B Rev 1.2

`[    5.325792] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[    5.326148] usbcore: registered new interface driver brcmfmac
[    5.569101] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[    5.569206] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available
[    5.569961] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Oct 23 2017 03:55:53 version 7.45.98.38 (r674442 CY) FWID 01-e58d219f
`

Johnny-quattro, 2020-05-12:
Still seeing error with original white Raspberry Pi dongle (official):

C3PI5:/home/pi $ dmesg -T | grep inetaddr
[Tue May 12 11:38:25 2020] brcmfmac: brcmf_inetaddr_changed: fail to get arp ip table err:-52
[Tue May 12 16:01:32 2020] brcmfmac: brcmf_inetaddr_changed: fail to get arp ip table err:-52
[Tue May 12 16:01:39 2020] brcmfmac: brcmf_inetaddr_changed: fail to get arp ip table err:-52
C3PI5:/home/pi $ /home/pi/scripts/pi_identify
Raspbian GNU/Linux 9 (stretch)
  Type: Pi 2, Revision: 01, Memory: 1024MB, Maker: Sony
  *--> Raspberry Pi 2 Model B Rev 1.1
C3PI5:/home/pi $ uname -a
Linux C3PI5 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux