raspberrypi/linux #2090: mmc_block quirks not possibly correctly in 4.9

jorisvergeer, 2017-06-27:
https://github.com/raspberrypi/linux/commit/04ddf9031a62eaf7048e5920d1e3a621aa7fcade
Fixed some typos in (did I interpret that correctly?) quirk presets in the mmc block driver in 4.4

This patch is not done in 4.9

Some kingston cards of type "broken" SD16G ended up in my production and were corrupted after an update. They do not get corrupted if add to cmdline.txt:
mmcblk.mmc_quirks=0x80000000 mmc_block.mmc_quirks=0x80000000 

I do not know the history of the driver between 4.4 and 4.9 but maybe this patch should also be done in 4.9.

pelwell, 2017-06-27:
Yes, that fixing commit got lost in the switch to 4.9. The two should really be squashed together.

It's in rpi-4.9.y now.

pelwell, 2017-07-03:
This patch is in the latest rpi-update kernel. Please test, then close if appropriate.

jorisvergeer, 2017-07-03:
@pelwell I can confirm that in 4.9.35 the quirk works without specifying it explicitly in the cmdline.txt

closed 2017-07-03 by jorisvergeer