raspberrypi/linux #2093: brcmfmac: ad-hoc mode doesn't work properly

skoehler, 2017-06-30:
Create a wpa_supplicant.conf containing the following:

```
network={
    ssid="TestAdHoc"
    frequency=2412
    key_mgmt=NONE
    mode=1
}
```
Then run `wpa_supplicant -i wlan0 -c wpa_supplicant.conf` on two Raspberry Pi 3 running latest Raspbian Jessy. The ad hoc network wont work. I also tried different values for ap_scan, namely 0, 1, and 2.

Running `wpa_supplicant -D wext -i wlan0 -c wpa_supplicant.conf` on both Raspberry Pi creates a working adhoc network, but produces a lot of error messages and in fact I believe that brcmfmac is a cfg80211 based driver, so -D nl80211 (which is the default) should work just fine. But it doesn't.

This seems like a bug in the brcmfmac driver or maybe a firmware issue.

lategoodbye, 2017-07-01:
Can you please provide the kernel version and some of these error messages?

JamesH65, 2017-09-13:
@skoekler Please retry with the latest Raspbian. There has been a new release of Wifi firmware as well as driver changes that may affect what you are trying.

JamesH65, 2017-12-04:
@skoehler Anything further to add to this issue? Can it be closed?

closed 2018-06-27 by JamesH65