raspberrypi/linux #2095: how to reboot into partition with new firmware but no systemd?

matthuisman, 2017-07-03:
I am trying to get Batocera OS rebooting into another partition.
They have just updated to 4.9.
The /sys/module/bcm2709 path is no longer present.

If I try the **reboot [part]**, it gives the error.
So, it looks like it's using an "old" reboot that doesn't know about the extra partition parameter?

Is there any other way to set the reboot part?
Or, is their a newer "reboot" I can compile?
Or, is there any simple C code I could compile to achieve it?
I can't fully change the system, but I could copy in a executable and run it if need be to get the result.

I see most systems have switched to systemd and it appears systemd takes care of the reboot now?

This OS is using systemvinit.
And I assume the reboot program is from busybox or something..

pelwell, 2017-07-03:
The RPi firmware still uses the same mechanism as before, but the board support code now uses the standard reboot handling mechanism - passing a command string containing the partition number to the kernel_restart function.

It sounds like Batocera is using a command (shell script?) that writes the supplied partition number into a module parameter exposed by the old BCM2708/9 board support code. This code no longer exists, hence the failure you are seeing. Instead it needs a small utility - conventionally called reboot under Linux - that invokes the reboot system call (http://man7.org/linux/man-pages/man2/reboot.2.html) passing in the required magic numbers, LINUX_REBOOT_CMD_RESTART2 and the string with the partition number.

You'll have to raise this with the Batocera devs.

closed 2017-07-03 by pelwell

matthuisman, 2017-07-03:
Hi,
yes, it already has the reboot utility (which symlinks to halt).
But, it seems this utility is older and doesn't know about the extra parameter.

I was asking is there some source code for the reboot / halt utility (with the extra parameter) that I can compile? Or even that I can modify with the extra parameter myself.

I'm just unsure where the source code for reboot / halt is kept?

matthuisman, 2017-07-04:
Ended up with this so far

`#include <linux/reboot.h>
#include <sys/reboot.h>
#include <sys/syscall.h>
#include <unistd.h>

int main(int argc, char *argv[])  {
    char *param = NULL;
    if( argc == 2 ) {
        param = argv[1];
    }

    sync();
    syscall(SYS_reboot, LINUX_REBOOT_MAGIC1, LINUX_REBOOT_MAGIC2, LINUX_REBOOT_CMD_RESTART2, param);
    
    return 0;
}`

However, it doesn't stop init or anything like that.
I wonder if I can call "halt", and then run this...

pelwell, 2017-07-04:
Raspbian uses systemd, and reboot, poweroff etc. are symlinked to systemctl, the source for which can be found here: https://github.com/systemd/systemd/blob/master/src/systemctl/systemctl.c

I found the source to a simpler shutdown command here: http://util-linux.sourcearchive.com/documentation/2.14/shutdown_8c-source.html
It includes killing init, turning off swap etc.