raspberrypi/linux #2096: pi zero - kernel oops on boot with dtoverlay lirc-rpi+w1_gpio+iqaudio-dacplus

MaFL2k, 2017-07-04:
I have a problem with a pi zero:
As soon as I activate all 3 dtoverlays I get a kernel oops on boot and the system stops working. Disabling one of the 3 and it works.
It also works on a rpi3. 

Tested it with fresh raspbian lite 4.9.28-v7+ and rpi-update to 4.9.35-v7+
 
```
[    7.320600] Unable to handle kernel NULL pointer dereference at virtual address 0000001b
[    7.334209] pgd = da60c000
[    7.339641] [0000001b] *pgd=00000000
[    7.345960] Internal error: Oops: 5 [#1] ARM
[    7.352840] Modules linked in: bcm2835_gpiomem lirc_rpi(C+) lirc_dev rc_core w1_gpio wire cn fixed uio_pdrv_genirq uio ipv6
[    7.369740] CPU: 0 PID: 114 Comm: systemd-journal Tainted: G         C      4.9.35+ #1014
[    7.383659] Hardware name: BCM2835
[    7.389987] task: da5e8000 task.stack: da678000
[    7.397493] PC is at kernfs_iop_get_link+0x90/0x1c0
[    7.405340] LR is at mutex_lock+0x1c/0x48
[    7.412314] pc : [<c01b910c>]    lr : [<c05d24ec>]    psr: a0000013
[    7.412314] sp : da679ef0  ip : da679ed8  fp : da679f24
[    7.429666] r10: dac3b720  r9 : 00000013  r8 : da679f5c
[    7.437771] r7 : da602000  r6 : da602000  r5 : be9d5704  r4 : dad4fdc0
[    7.447239] r3 : 00000013  r2 : c0733d48  r1 : 00000000  r0 : 00000000
[    7.456601] Flags: NzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[    7.466600] Control: 00c5387d  Table: 1a60c008  DAC: 00000055
[    7.475165] Process systemd-journal (pid: 114, stack limit = 0xda678188)
[    7.484690] Stack: (0xda679ef0 to 0xda67a000)
[    7.491787] 9ee0:                                     595b7e76 da679f28 dab73140 00000400
[    7.505539] 9f00: be9d5704 ffffff9c be9d5304 da679f5c dab73140 00000400 da679f44 da679f28
[    7.519559] 9f20: c0155b44 c01b9088 00000000 00000000 ffffffea 00004000 da679f94 da679f48
[    7.533633] 9f40: c014a384 c0155acc da679f5c c00262d8 da679f6c be9d5704 c00262d8 00000000
[    7.547880] 9f60: daf6e910 dab7a198 c005e248 801c6db8 b6f7bf10 be9d5304 00000055 c000ffc4
[    7.562466] 9f80: da678000 00000000 da679fa4 da679f98 c014a3fc c014a2ec 00000000 da679fa8
[    7.577308] 9fa0: c000fe40 c014a3dc 801c6db8 b6f7bf10 be9d5304 be9d5704 00000400 e9df6c00
[    7.592442] 9fc0: 801c6db8 b6f7bf10 be9d5304 00000055 be9da020 00000000 00000003 be9da895
[    7.607884] 9fe0: 7f610dd4 be9d50f4 7f5f097c b6e9226c 60000010 be9d5304 55555551 55115555
[    7.623513] [<c01b910c>] (kernfs_iop_get_link) from [<c0155b44>] (generic_readlink+0x84/0x94)
[    7.639658] [<c0155b44>] (generic_readlink) from [<c014a384>] (SyS_readlinkat+0xa4/0xf0)
[    7.655545] [<c014a384>] (SyS_readlinkat) from [<c014a3fc>] (SyS_readlink+0x2c/0x30)
[    7.671174] [<c014a3fc>] (SyS_readlink) from [<c000fe40>] (ret_fast_syscall+0x0/0x1c)
[    7.686897] Code: ea000002 e1530004 e1a03001 0a00000b (e5931008) 
[    7.697088] ---[ end trace 8093b0d380080e13 ]---
```

boot/config:
```
dtoverlay=iqaudio-dacplus
dtoverlay=w1-gpio,gpiopin=4,pullup=on
dtoverlay=lirc-rpi,gpio_out_pin=17,gpio_in_pin=25
```

complete boot log:
[bootlog.txt](https://github.com/raspberrypi/linux/files/1121954/bootlog.txt)

Any ideas?
Matt

pelwell, 2017-07-04:
1. Are the errors fairly consistent or do they change from run to run?
2. With only two of the overlays loaded, are you seeing any power supply warnings? What does `vcgencmd get_throttled` report?

pelwell, 2017-07-04:
I couldn't lay my hands on the same sound card, but my test with a very similar card worked as expected, albeit without any w1 or IR hardware.

pelwell, 2017-07-05:
If power supply doesn't seem to be an issue, please try your tests again but with some or all of the Onewire and IR hardware removed.

MaFL2k, 2017-07-05:
The messages are very consistent, stack and registers values alter a bit, but this is normal.

```
vcgencmd get_throttled
throttled=0x0
```

I don´t think it´s a power problem. It makes no difference if I remove the hardware. 
I only have one zero, but I´ll test it also on a pi1 and zero-w

JamesH65, 2017-09-13:
@MaFL2K Did you have a chance to try out @pelwell suggestions? Or can this issue be closed?

MaFL2k, 2017-09-14:
@JamesH65 The issue is still unsolved.

@pelwell's tendency was a power problem. I don´t think so. As I wrote above it makes no difference if the hardware is connected or not, also no throttling occured, despite it´s a zero and not 2/3 -> so no.
The only thing I need to do to let this zero panic on boot is to activate the 3 dtos.

But It works on a zero-w

pelwell, 2017-09-14:
I've just tried again on a rev 1.3 Zero with no external hardware and it still boots without incident.

Comparing the crash dump with a kernel disassembly I think it's crashing in kernfs_get_target_path dereferencing kn->parent [here](http://elixir.free-electrons.com/linux/latest/source/fs/kernfs/symlink.c#L60) where the parent pointer is 0x13 (i.e. garbage).

JamesH65, 2017-12-04:
@MaFL2k  @pelwell Any progress on this? Can it be closed?