raspberrypi/linux #2097: Problem when plugging several USNB

slaadvak, 2017-07-04:
I make a test bench using a lot of USB peripherals. Everything was working fine with kernel 4.4.50-v7+, except for a sporadic problem (https://github.com/raspberrypi/linux/issues/830). So, I've decided to upgrade my kernel with the `dist-upgrade` command, because of this recent pull request in kernel 4.9.28: https://github.com/raspberrypi/linux/pull/2010. But since then, I'm running into important issues which have for effect of preventing using most of my USB peripherals, keyboard included.

I'm using a RasPi 3 with Raspian.

cat /etc/os-release:

```
PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

uname -r:
```
4.9.35-v7+
```

My setup is made of two USB hubs and a keyboard. Each hub have 10 ports and each port is filled with a USB ACM device.
The idVendor of the hubs is 1a40. The idVendor of the ACM devices is 15a2.
All the devices, except for the keyboard are initialy unplugged.

If I only plug the first hub (path is `1.1.2`) everything works fine. You can see this at around 46 seconds in the `dmesg` excerpt below.

Then, if I plug the second USB hub (path is `1.1.4`) the USB keyboard stops working. I can however still connect to my RasPi with SSH. I see that all the USB peripherals appear corrrectly in `/dev`, but slowly than usual. When I try to connect to these device (through some serial API) it doesn't work, or if it finally works, I receive a lot of garbage.

You can see in the `dmesg` below the connection of the second USB hub at around 68 seconds. You will see a lot of errors about clock problems and then, finally an error in `assign_and_init_hc`. If I try to unplug/replug the hubs, in the same or in another port, I won't be able to recover from that weird state.

```
[   47.636830] usb 1-1.2: new high-speed USB device number 5 using dwc_otg
[   47.767285] usb 1-1.2: New USB device found, idVendor=1a40, idProduct=0201
[   47.767298] usb 1-1.2: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[   47.767306] usb 1-1.2: Product: USB 2.0 Hub [MTT]
[   47.768243] hub 1-1.2:1.0: USB hub found
[   47.768342] hub 1-1.2:1.0: 7 ports detected
[   48.086839] usb 1-1.2.1: new full-speed USB device number 6 using dwc_otg
[   48.219908] usb 1-1.2.1: New USB device found, idVendor=15a2, idProduct=0300
[   48.219920] usb 1-1.2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   48.219928] usb 1-1.2.1: Product: IO BOARD
[   48.219936] usb 1-1.2.1: Manufacturer: ZIP HYPERION.
[   48.243494] cdc_acm 1-1.2.1:1.0: ttyACM0: USB ACM device
[   48.244392] usbcore: registered new interface driver cdc_acm
[   48.244397] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
[   48.316833] usb 1-1.2.2: new full-speed USB device number 7 using dwc_otg
[   48.449907] usb 1-1.2.2: New USB device found, idVendor=15a2, idProduct=0300
[   48.449918] usb 1-1.2.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   48.449926] usb 1-1.2.2: Product: IO BOARD
[   48.449934] usb 1-1.2.2: Manufacturer: ZIP HYPERION.
[   48.452067] cdc_acm 1-1.2.2:1.0: ttyACM1: USB ACM device
[   48.546832] usb 1-1.2.3: new full-speed USB device number 8 using dwc_otg
[   48.679906] usb 1-1.2.3: New USB device found, idVendor=15a2, idProduct=0300
[   48.679918] usb 1-1.2.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   48.679926] usb 1-1.2.3: Product: IO BOARD
[   48.679934] usb 1-1.2.3: Manufacturer: ZIP HYPERION.
[   48.683154] cdc_acm 1-1.2.3:1.0: ttyACM2: USB ACM device
[   48.776840] usb 1-1.2.4: new full-speed USB device number 9 using dwc_otg
[   48.909907] usb 1-1.2.4: New USB device found, idVendor=15a2, idProduct=0300
[   48.909918] usb 1-1.2.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   48.909926] usb 1-1.2.4: Product: IO BOARD
[   48.909934] usb 1-1.2.4: Manufacturer: ZIP HYPERION.
[   48.912172] cdc_acm 1-1.2.4:1.0: ttyACM3: USB ACM device
[   49.006856] usb 1-1.2.5: new full-speed USB device number 10 using dwc_otg
[   49.139908] usb 1-1.2.5: New USB device found, idVendor=15a2, idProduct=0300
[   49.139919] usb 1-1.2.5: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   49.139927] usb 1-1.2.5: Product: IO BOARD
[... 109 lines cut ...]
[   72.511702] bcm2835-cpufreq:bcm2835_cpufreq_driver_target_index:175: Error occurred setting a new frequency (1200000)
[   72.511706] cpufreq: __target_index: Failed to change cpu frequency: -22
[   73.216894] bcm2835-cpufreq:bcm2835_cpufreq_set_clock:76: Failed to set clock: 1200000 (-12)
[   73.216900] bcm2835-cpufreq:bcm2835_cpufreq_driver_target_index:175: Error occurred setting a new frequency (1200000)
[   73.216905] cpufreq: __target_index: Failed to change cpu frequency: -22
[   73.926904] bcm2835-cpufreq:bcm2835_cpufreq_set_clock:76: Failed to set clock: 1200000 (-12)
[   73.926910] bcm2835-cpufreq:bcm2835_cpufreq_driver_target_index:175: Error occurred setting a new frequency (1200000)
[   73.926914] cpufreq: __target_index: Failed to change cpu frequency: -22
[   74.636902] bcm2835-cpufreq:bcm2835_cpufreq_set_clock:76: Failed to set clock: 1200000 (-12)
[   74.636909] bcm2835-cpufreq:bcm2835_cpufreq_driver_target_index:175: Error occurred setting a new frequency (1200000)
[   74.636914] cpufreq: __target_index: Failed to change cpu frequency: -22
[   75.346904] bcm2835-cpufreq:bcm2835_cpufreq_set_clock:76: Failed to set clock: 1200000 (-12)
[   75.346911] bcm2835-cpufreq:bcm2835_cpufreq_driver_target_index:175: Error occurred setting a new frequency (1200000)
[   75.346916] cpufreq: __target_index: Failed to change cpu frequency: -22
[   76.056917] bcm2835-cpufreq:bcm2835_cpufreq_set_clock:76: Failed to set clock: 1200000 (-12)
[   76.056924] bcm2835-cpufreq:bcm2835_cpufreq_driver_target_index:175: Error occurred setting a new frequency (1200000)
[   76.056929] cpufreq: __target_index: Failed to change cpu frequency: -22
[   76.406927] usb 1-1.4.7.2: new full-speed USB device number 28 using dwc_otg
[   76.540037] usb 1-1.4.7.2: New USB device found, idVendor=15a2, idProduct=0300
[   76.540050] usb 1-1.4.7.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   76.540059] usb 1-1.4.7.2: Product: IO BOARD
[   76.540067] usb 1-1.4.7.2: Manufacturer: ZIP HYPERION.
[   76.542555] cdc_acm 1-1.4.7.2:1.0: ttyACM17: USB ACM device
[   76.542644] ERROR::assign_and_init_hc:1377: assign_and_init_hc: Failed to allocate memory to handle non-dword aligned buffer case
```

P33M, 2017-07-12:
>My setup is made of two independently powered USB hubs and a keyboard. Each hub have 10 ports and each port is filled with a USB ACM device.

This is unlikely to be supportable due to the inherent limitations of the DWC USB host on Pi. That said, what is the first firmware revision that fails in a worse way than previously?

Run `sudo rpi-update <HASH>` where hash is from this repository's commits:

https://github.com/Hexxeh/rpi-firmware/commits/master 

and report back with the first "bad" version.

slaadvak, 2017-07-13:
This is the firmware version of the first "bad" version, corresponding to Kernel version 4.4.50-v7+:
`sudo rpi-update 1fdc3a06af325159cf1ed88b5bbd1bae85346d0a`

https://github.com/hexxeh/rpi-firmware/tree/1fdc3a06af325159cf1ed88b5bbd1bae85346d0a

With this one, I am able to poll up to 30 devices for around 20 hours, once every minute. The longer between polling, the longer the rpi will run correctly. After that, I encounter the issue https://github.com/raspberrypi/linux/issues/830 (I guess!). Here is a screenshot of a typical kernel stack trace that I get when this situation occurs: 
[image]

slaadvak, 2017-07-13:
Sorry, I misunderstood what you asked me. I'm trying to find the first commit where it fails in the worse way, as described at the top of this issue...

JamesH65, 2017-09-13:
@slaadvak Did you have a chance to determine which commit caused a performance regression? Or is the issue solved?

JamesH65, 2017-12-04:
Closing due to lack of activity. Reopen if you feel this issue is still relevant.

closed 2017-12-04 by JamesH65

PocketSam, 2018-12-07:
In my case the problem was caused by a USB keyboard. I've disconnected it and everything became fine. Active USB hub and USB-modem wasn't a problem for rPi3.