raspberrypi/linux #2098: rpi-4.12.y non-fatal errors on boot

asavah, 2017-07-05:
When booting fresh 4.12.y (bcm2709_defconfig) on pi3 I get 2 errors (which actually seem to be warnings):
```
[    0.791045] Error: Driver 'bcm2835_thermal' is already registered, aborting...
[    0.839276] Error: Driver 'sdhost-bcm2835' is already registered, aborting...
```
The system actually boots and works fine.
dtb seems to load fine too
```
001680.997: brfs: File read: /mfs/sd/bcm2710-rpi-3-b.dtb
```

```
vcgencmd version
Jul  3 2017 14:20:31
Copyright (c) 2012 Broadcom
version 4139c62f14cafdb7d918a3eaa0dbd68cf434e0d8 (tainted) (release)
```

pelwell, 2017-07-05:
I get no errors booting to top-of-tree rpi-12.y apart from the transient error from bcm2835-aux-uart because its clock isn't ready.

Both of the drivers listed are built into the kernel so there's no reason why anything should be trying to load them. Are you perhaps using an initramfs that hasn't been updated?

asavah, 2017-07-05:
@pelwell Thank, and sorry for the noise.
I found the issue and it's in my (semiautomated) build system, it did not detect properly that I switched kernel branch and "upgraded" (make olddefconfig) 4.11.y config instead of running bcm2709_defconfig.
This resulted in both "downstream" and "upstream" ?? drivers being enabled, thus resulting in this little conflict.

"upgraded" config (with error)
```
CONFIG_MMC_BCM2835_MMC=y
CONFIG_MMC_BCM2835=y

CONFIG_THERMAL_BCM2835=y
CONFIG_BCM2835_THERMAL=y
```

bcm2709_defconfig (clean boot)
```
CONFIG_MMC_BCM2835_MMC=y
# CONFIG_MMC_BCM2835 is not set

# CONFIG_THERMAL_BCM2835 is not set
CONFIG_BCM2835_THERMAL=y
```

Feel free to close, I'm leaving this info here for the people that may stumble into same error.

pelwell, 2017-07-05:
No problem.

closed 2017-07-05 by pelwell