raspberrypi/linux #2099: w1_gpio/w1_therm can return incorrect data

andrew01144, 2017-07-05:
Problem:
Using w1_gpio + w1_therm in python3 on Raspberry Pi, the driver occasionally returns the value from the previous sample instead of the current sample.
Important: This is a 'silent problem', you won't usually know that it has occurred unless you look at the smoothness of a temp/time graph (see [Summary.pdf](https://github.com/raspberrypi/linux/files/1124208/Summary.pdf)).

Linux RPi02 4.9.28+ #998 Mon May 15 16:50:35 BST 2017 armv6l GNU/Linux

I previously posted it here: https://bugs.launchpad.net/raspbian/+bug/1702243, but it was suggested that github was the best place for it.

My findings:
This problem happens on both RPi V3 and RPi Zero W, but happens much more frequently on the Zero.
What seems to be happening:
The 1-wire bit stream has occasional pauses in it, as if the driver is being interrupted. I have seen pauses from 150us to 1ms.
If the pin is high when the pause occurs, it's ok.
If low, it breaks the protocol and the packet will be ignored.
If that happens during the 'read scratchpad' command (0xBE), I expect it will cause a CRC and the driver will re-read.
If that happens in the 'convert' command (0x44), we have a problem.
 The driver will assume the convert has been issued, and will issue a read command (0xBE) approx 780ms later.
 However, the convert command was not successfully issued, so the data read will be from an older point in time.
[Summary.pdf](https://github.com/raspberrypi/linux/files/1124208/Summary.pdf) has examples from the pulseview logic analyzer.

In my tests:
RPi V3: Executed 477 samples, 3 of which were old data (0.6%).
RPi Zero: Executed 522 samples, 45 of which were old data (8.6%).

Possible remedy:
 driver should disable interrupts during critical operations, such as sending a low bit.
 driver should issue a readBit() after 'convert' to check that the DS18B20 has gone busy.

Testing:
You need to apply a changing temperature to the DS18B20 in order to spot what seems to be an old sample. I have built a reasonably simple test rig to do this. I am happy to test any new code on my rig, or I can describe how to build the test rig.

Many thanks!

pelwell, 2017-07-05:
BCM2835 has no w1 interface hardware, so the w1_gpio driver has to bit-bash it using GPIOs. I consider this a best-effort service, so I would expect reduced performance under load. Having said that, I see that the w1 module (called `wire`) has a parameter called "disable_irqs" that may do what you want - try setting it to 1:
```
sudo sh -c "echo options wire disable_irqs=1 > /etc/modprobe.d/wire.conf"
```
Then reboot and see if the quality of readings has improved.

andrew01144, 2017-07-05:
Thanks, pelwell.
disable_irqs=1 improves it hugely.
There are still plenty of pauses in the bit stream, but none while the pin is low, so the protocol is not broken.

I do still see very occasional instances of returning [what appears to be] data from the previous sample, but I can't see anything wrong in the protocol analyzer trace. I'll leave the issue open for a couple of days, in case I discover anything relevant. 

Many thanks!

andrew01144, 2017-08-10:
I have done some more tests, taken a closer look at the protocol analyzer traces, and here are my findings.

1. disable_irqs=1 improves it hugely.
2. My Pi-Zero-W shows the problem much more than my Pi3B. I don't know why.
3. I do still see a few mis-converts. IMHO, I think these could be eliminated with a change to the driver logic. I'll describe this change below.

What should I do with this "issue" now?
I guess I should close it, but I would like someone to consider my suggestion to improve the robustness of the driver.
Sadly, I'm not clever enough to implement it myself!

###############################################################

1) Suggestion to improve the robustness of the driver.
The problem is simple: If the 'convert' command gets corrupted in any way, such that the device does not see it, the user will be unaware. The user then goes on to read the result from the device. If the convert command was not seen by the device, the user will get the previous temperature reading, which could be seconds, minutes or days old.
I think a simple fix would avoid this: When the driver issues the 'convert' command, it should immediately check that the device goes busy. If it does not, it should retry the convert command a few times.

2) Suggestion: Should wire disable_irqs=1 be default?
Or, maybe /etc/modprobe.d/wire.conf be provided with a commented out wire disable_irqs=1, to draw attention to it.

################################################################

Results from my tests:

RunNumber				 Samples	Bad	     %
31	PiZ, irq-disable=1		1024	 1	0.10%   (Other runs have been as high as 0.40%.)
32	Arduino				1068	 0	0.00%
33	PiZ, without irq-disable	1058	29	2.74%

Description of corruption.
The 1-wire pulses are being stretched when Raspian interrupts the 1-wire driver to do something else.
The 1-wire protocol is immune to stretching a high, but stretching a low causes corruption.
wire.conf: "wire disable_irqs=1" makes a good job of stopping the lows getting interrupted.
It does not seem to stop the highs being interrupted, but AFAIK that is ok.

Normal timing of the 1-wire protocol as seen from my Pi-Zero:
Zero is represented as a long low of 62us.
One is represented as a short low of 7us.
I have seen two kinds of error:
1) The low for a One is stretched to approx 30us, causing the One to appear as a Zero.
2) The signal seems to be overwritten by a low of 165us, causing one or two bits to be obliterated.

Test methodology:

I have two RPis.
RPi01 is the controller, and has a logic analyzer connected to a USB port.
RPi02 is a Pi-Zero-W, and has 2x DS18B20s connected to a GPIO pin.

RPi01 has a python program that does the following every 15sec:
	Start sigrok-cli to capture the 1-wire protocol.
	Take a temperature reading by "ssh RPi02 cat /sys/bus/w1/devices/28-04169085b3ff/w1_slave"
	If the reading is the same as the previous reading, I suspect that the convert command may have failed.
	So, if duplicate, I take another reading. If the new reading is within 2 LSBs (+/- 0.12 deg C), I assume the duplicate reading was genuine.
	If greater, then I assume the convert failed, and take a copy of the sigrok trace for later viewing.
I use a lamp over the DS18B20s to cycle the temperature between 27 and 41 degrees, so that most readings should be different from the previous reading.

P33M, 2017-08-10:
You will see an increase in raw error rate with pi0w even with irq_disable=1 because of the relative latency for memory accesses when the SoC is under load. If the Pi's notion of time is smeared out, then it will either
a) assert 0 for longer than necessary when transmitting (the cause of 1. above) or
b) mis-time when it next needs to signal a start bit, trampling a device's response (probably the cause of 2. above).

Bit times with a resolution of 7uS are hard to guarantee if there is other system bus traffic that could cause the ARM CPU to contend for either SDRAM cycles or cycles to the GPIO registers. Similarly even if the driver performs the transfer with IRQs disabled, the first time around the bitbash loop will incur latency as instruction and data cache lines are filled. As a general rule of thumb, the hard realtime horizon for Pi 1 is about 10uS.

Pi3 has a 512k L2 cache which serves to hide a lot of these effects, whereas BCM2835-based products have to make do with 16k Level 1 I and D cache.

andrew01144, 2017-08-10:
Thanks for the explanation.
What do you think about adding a check-for-busy to improve the robustness of the driver?
Should I close this now? I don't know what the etiquette is.

JamesH65, 2017-09-13:
Closing this issue as questions answered/issue resolved.

closed 2017-09-13 by JamesH65