raspberrypi/linux #2105: no rpi3 onboard wlan (brcmfmac) using kernel 4.12.0

mkreisl, 2017-07-07:
Building new 4.12.0 kernel, using kernel config from working 4.11 kernel.

Kernel is booting, but no wlan is there. Module brcmfmac is not loaded, no (error) message can be found in dmesg

In my kernel config, there is BRCMFMAC configured as follows:

```
CONFIG_BRCMUTIL=m
# CONFIG_BRCMSMAC is not set
CONFIG_BRCMFMAC=m
CONFIG_BRCMFMAC_PROTO_BCDC=y
CONFIG_BRCMFMAC_SDIO=y
CONFIG_BRCMFMAC_USB=y
```

If I run `sudo modprobe brcmfmac manually`, I get 
```
[  222.258843] usbcore: registered new interface driver brcmfmac
```

Do I miss anything?

lategoodbye, 2017-07-08:
Yes, the Wifi chip is connected via SDIO. You'll need to make sure that the sdhost (sd card) and MMC driver (wifi) are running and properly setup in devicetree. So the complete dmesg would be helpful here.

pelwell, 2017-07-08:
Our advice is always to start with one of the RPi defconfigs appropriate for the kernel version, then once that works to make the required config changes.

Dependencies between config settings change with each kernel version, so it is usually a good idea to use the savedefconfig feature to distil a .config down to the bare essentials before switching to a new kernel.

mkreisl, 2017-07-08:
@lategoodbye @pelwell 
Thanks for your quick answer.
I was running safedefconfig, built kernel again and now wlan is back 😄 
But strange thing is, I can not see any significant difference between old (not working) config and new config. Very confusing. But anyway, it's working again

closed 2017-07-08 by mkreisl