raspberrypi/linux #2106: Linux wifi doesn't work due to express vpn

Darkblood618, 2017-07-08:
Alright so I don't really know where else to post this because the torvald Linus one doesn't have an issues tab. So I use express on and it expired a few weeks ago. But I always had the problem that I have to use the vpn while I use the wifi or it won't work. But since it expired I can't use the wifi. When I am on chrome it gives the message that there is no dns address or something like that. Any fix would be great thanks

JamesH65, 2017-07-08:
You are going to have to provide a LOT more information before we can make any sense of this.

What is your setup, which Pi version, which version of Raspbian, etc. It also might be a better bet to post on the Pi forums instead of here, as that has more viewers.

JamesH65, 2017-09-13:
Closing due to lack of activity/no response to information requests. Reopen if you feel this issue is still relevant.

closed 2017-09-13 by JamesH65