raspberrypi/linux #2111: Excessive rt8192eu dropped packets.

cobaltdr, 2017-07-12:
Problem:
Hello. I was asked by someone in #raspbian on freenode to open a ticket on this here. I have about 2-4% of all packets going through wlan1 being dropped (Rx)

Background:
I have a Pi Zero W running jessie with kernel 4.9.35+. wlan1 is a dongle I bought containing the rt8192eu chipset (UK); this seems to be fairly common. The reason to use a second wireless interface was as an access point, while the onboard wlan0 connects to the internet via the intranet.

2-4% of all packets being dropped seems exceedingly high, when considering wlan0 drops none. The kernel module had to be compiled and was sourced from https://github.com/Mange/rtl8192eu-linux-driver

Comments on the forums I was reading eventually pointed me either to a .ko file compiled by people there, or to the source there which was already patched to work with the Pi. I had to go that route because the sources supplied by my manufacturer (TPlink) would not compile, and I didn't trust a .ko file just off the net. Comments also mentioned the source released by Realtek was quite buggy. I had to edit some of the debug messages out of the source to avoid my logs being flooded (about 4MB in one hour). While not completely gone, this is more acceptable now, and only a minor annoyance.

Question:
I would be happy to supply more information, but I'm not entirely sure how to proceed, and what to give. Any suggestions welcome.

lsmod | grep 8192eu:
8192eu                967936  0

sudo ifconfig
wlan0     Link encap:Ethernet  HWaddr b8:27:eb:6d:2e:24  
          inet addr:192.168.2.101  Bcast:255.255.255.255  Mask:255.255.255.0
          inet6 addr: fe80::ad2c:a3aa:7ad7:f853/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:126201 errors:0 dropped:0 overruns:0 frame:0
          TX packets:103634 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:63724146 (60.7 MiB)  TX bytes:16648689 (15.8 MiB)

wlan1     Link encap:Ethernet  HWaddr 84:16:f9:11:28:06  
          inet addr:172.24.1.1  Bcast:172.24.1.255  Mask:255.255.255.0
          inet6 addr: fe80::8616:f9ff:fe11:2806/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:128810 errors:0 dropped:4110 overruns:0 frame:0
          TX packets:68220 errors:0 dropped:26 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:13058922 (12.4 MiB)  TX bytes:62884087 (59.9 MiB)

rodizio1, 2017-07-12:
Not sure if it's related, but I also get quite high packetloss with the rtl8192cu kernel module in 4.9.28 and 4.9.35.

JamesH65, 2017-07-12:
Does any one have any figures for older kernel version? Is this something
that has started happening recently, or has it always been there?

cobaltdr, 2017-07-21:
Posted in https://github.com/raspberrypi/linux/issues/1342#issuecomment-317083600 another issue with this chipset/module where having it plugged in on a Pi Zero W also causes disconnects on the wlan0 every 2 hours or so.

JamesH65, 2017-09-15:
I wonder if this is related to #2134? (as suggested here https://github.com/raspberrypi/linux/issues/1954#issuecomment-329780032)  In which case updating the to latest kernel would be a good test.

JamesH65, 2017-09-15:
If anyone with this dongle is able to test with the latest kernel, it would be useful to know if the issue is still present.

JamesH65, 2017-12-04:
Since there have been no responses, I'll close this. Please reopen if anyone has any further information.

closed 2017-12-04 by JamesH65