raspberrypi/linux #2112: kernel BUG in ext4/mballoc

bavay, 2017-07-12:
Hi!

I am running raspbian (stretch) on a Raspberry pi 2B rev 1.1. It currently uses the following kernel: 4.9.28-v7+ #998 SMP Mon May 15 16:55:39 BST 2017 armv7l GNU/Linux

Today, I had curl (a cron job runs curl in order to update my dyndns service) triggering a kernel bug (Unable to handle kernel NULL pointer dereference at virtual address 00000044). When curl ran again, it triggered a kernel bug in EXT4-fs (kernel BUG at fs/ext4/mballoc.c:3988).
[kernel_bug.txt](https://github.com/raspberrypi/linux/files/1142190/kernel_bug.txt)

Feel free to ask me for more information if anything is missing!
Mathias

popcornmix, 2017-07-12:
Obvious questions first:
Any custom cmdline.txt changes?
Any custom config.txt settings?
Any peripherals (HAT/USB etc) plugged in?
What does ``vcgencmd get_throttled`` report after running under load for a while?

bavay, 2017-07-12:
Here is my cmdline.txt:
dwc_otg.lpm_enable=0 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline cgroup_enable=memory swapaccount=1 fsck.repair=yes rootwait

And in my config.txt: I overclocked (using raspi-config and then pushed the over voltage to 4) and turned off several features (audio, i2c, i2s, spi).
[config.txt](https://github.com/raspberrypi/linux/files/1142360/config.txt)

I have one USB hard disk (powered through the USB port) and no hats. The power supply has so far never given me any trouble and was sold as "good quality" for the raspberry and peripherals.

The system is mostly sitting idle except for short burst when serving pages / files or when indexing (it also runs minidlna). I've so far never seen the temperature above 42C. And I will try loading it to check the output of "vcgencmd get_throttled" as soon as I'll have restarted it...

bavay, 2017-07-12:
So, I compiled a 68k lines C++ library that is usually quite heavy (quite a few templates, etc) on 4 cores (make -j4) and that took several minutes. It put quite a heavy load on the system and the cpu temperature went up to 62.7 C. But "vcgencmd get_throttled" still reports "throttled=0x0". Should I try to put such a load for a longer time so as to force the throttle to kick in? (I can just force it to recompile several times)

lategoodbye, 2017-07-17:
Are you able to reproduce the issue with a fresh raspbian without changes to the config?

popcornmix, 2017-07-17:
Yes, at first sign of unexpected behaviour you should remove any overclocking settings.

bavay, 2017-07-17:
I am currently running some tests (without any overclocking) but I still have some weird behavior from nextcloud regarding reading some files (it fails to load some images). I am starting to wonder if there is not a hardware failure somehow (I've just recently moved all the files to an external, usb drive)... I'll update this thread when I have more results!

bavay, 2017-07-24:
I confirm that I could reproduce some issues without any overclocking (default config). I could not ssh to the system, its webserver did not answer, when connecting a keyboard and a screen, I could not get any signal (black screen, although I normally get a console). In all my logs but one, there is a gap of several hours. But in syslog, I still see some activity during this unresponsive time. I was hoping for a nice dump, but so far there is none. But at least, I confirm that there is a problem somewhere (but it can be hardware as well as software).

JamesH65, 2017-09-13:
@bavay Have you any further information to add to this issue? Is it still happening?

bavay, 2017-09-26:
It still happens from time to time (twice a month? on a 24/7 system), but I've never been able to get any meaningful logs. My gut feeling is that it is actually a hardware problem...

JamesH65, 2017-12-04:
@bavay Anything further on this one? Does the latest kernel help? Have you tried alternative HW?

bavay, 2017-12-11:
I have been running on alternative HW (actually, a raspberry 3) also running 24/7 since early October without any problems (I cloned the system on another sdcard before starting the new HW, so it was the exact same software to start with).

closed 2018-06-27 by JamesH65