raspberrypi/linux #2113: Kernel panic with 4.9.28 and 4.9.35 on a Pi1A+

rodizio1, 2017-07-13:
It crashes quite hard ;)

https://youtu.be/90Auf6_oaq0

This happens on both Kernel 4.9.28 and 4.9.35 (from the latest two Raspbian releases) on a Pi1A+. I have tested the same kernel on a Pi2 and Pi3, there it's stable.

pelwell, 2017-07-14:
I just started with the latest Raspbian Desktop download (a 4.9.35+ kernel) and it booted in an A+ without incident.

If you have a spare card, download https://downloads.raspberrypi.org/raspbian_latest and try it in your A+.

If you can't get it to work (or you don't have spare card), it would help if you could extract the kernel crash log. The best way is with a serial port attached to GPIOs 14 and 15 - use raspi-config to enable the serial console and the UART, then add `loglevel=8` to the end of `cmdline.txt`.

Failing that, try taking a few photographs of the screen at the point where it crashes - stills are easier to read than video. You can get more on screen by adding `framebuffer_height=2160` to config.txt.

rodizio1, 2017-07-16:
Did some more testing, it's caused by the firmware, not the kernel (I always update kernel and firmware together to avoid compatibility issues). Sorry for the confusion.

Will do more testing to narrow it down some more and open a new issue in the firmware repository. So far I can only tell it works with the firmware in 2017-02-27, but not with 2017-06-21 and above.

closed 2017-07-16 by rodizio1