raspberrypi/linux #2114: PiCamera 2.1 module not working with rpi-4.12.y branch

alexxy, 2017-07-14:
Hi!

I get picamera 2.1 module and it seems its not working via bcm2835-v4l2

> [10749.357227] mmal_vchiq: received invalid message context 132!
[37556.267576] bcm2835-v4l2: Clipping/aligning 800x600 format 47504A4D
[37556.267593] bcm2835-v4l2: Now 800x600 format 47504A4D
[37556.267608] bcm2835-v4l2: vidioc_try_fmt_vid_cap: w 800 h 600 field 1 pfmt 0x47504a4d bpl 0 sz_img 486400 colorspace 0x1 priv 0
[37556.267617] bcm2835-v4l2: vid_cap - disconnect previous tunnel
[37556.268334] bcm2835-v4l2: vid_cap - set up encode comp
[37556.268991] bcm2835-v4l2: vid_cap - cur_buf.size set to 486400
[37556.269003] bcm2835-v4l2: Set dev->capture.fmt 47504A4D, 800x600, stride 0, size 486400
[37556.269044] bcm2835-v4l2: Set fps range to 24/1 to 24/1
[37556.269397] bcm2835-v4l2: queue_setup: dev:baff2800
[37556.272543] bcm2835-v4l2: buffer_prepare: dev:baff2800
[37556.272569] bcm2835-v4l2: buffer_prepare: dev:baff2800
[37556.272586] bcm2835-v4l2: buffer_prepare: dev:baff2800
[37556.272600] bcm2835-v4l2: buffer_prepare: dev:baff2800
[37556.272662] bcm2835-v4l2: vidioc_g_fmt_vid_cap: w 800 h 600 field 1 pfmt 0x47504a4d bpl 0 sz_img 486400 colorspace 0x1 priv 0
[37556.276729] bcm2835-v4l2: buffer_queue: dev:baff2800 buf:baa00e00
[37556.276747] bcm2835-v4l2: buffer_queue: dev:baff2800 buf:baa01800
[37556.276756] bcm2835-v4l2: buffer_queue: dev:baff2800 buf:baa01a00
[37556.276765] bcm2835-v4l2: buffer_queue: dev:baff2800 buf:baa00c00
[37556.276774] bcm2835-v4l2: start_streaming: dev:baff2800
[37559.355993] mmal_vchiq: error 0 waiting for sync completion
[37559.356014] bcm2835-v4l2: Failed enabling camera, ret -62
[37559.356021] bcm2835-v4l2: Failed to enable camera
[37559.356029] ------------[ cut here ]------------
[37559.356088] WARNING: CPU: 1 PID: 30338 at drivers/media/v4l2-core/videobuf2-core.c:1342 vb2_start_streaming+0xe4/0x160 [videobuf2_core]
[37559.356095] Modules linked in: bcm2835_v4l2(C) videobuf2_vmalloc videobuf2_memops v4l2_common videobuf2_v4l2 videobuf2_core videodev media ipv6 8021q garp stp llc binfmt_misc uio_pdrv_genirq fixed uio
[37559.356193] CPU: 1 PID: 30338 Comm: mjpg_streamer Tainted: G        WC      4.12.0-raspberrypi-rpi+ #2
[37559.356199] Hardware name: BCM2835
[37559.356240] [<8011073c>] (unwind_backtrace) from [<8010c56c>] (show_stack+0x20/0x24)
[37559.356262] [<8010c56c>] (show_stack) from [<80586398>] (dump_stack+0xc8/0x10c)
[37559.356284] [<80586398>] (dump_stack) from [<8011e77c>] (__warn+0xf8/0x110)
[37559.356301] [<8011e77c>] (__warn) from [<8011e838>] (warn_slowpath_null+0x30/0x38)
[37559.356339] [<8011e838>] (warn_slowpath_null) from [<7f153ca4>] (vb2_start_streaming+0xe4/0x160 [videobuf2_core])
[37559.356395] [<7f153ca4>] (vb2_start_streaming [videobuf2_core]) from [<7f155a30>] (vb2_core_streamon+0x138/0x180 [videobuf2_core])
[37559.356458] [<7f155a30>] (vb2_core_streamon [videobuf2_core]) from [<7f163068>] (vb2_streamon+0x40/0x60 [videobuf2_v4l2])
[37559.356504] [<7f163068>] (vb2_streamon [videobuf2_v4l2]) from [<7f1630d4>] (vb2_ioctl_streamon+0x4c/0x50 [videobuf2_v4l2])
[37559.356686] [<7f1630d4>] (vb2_ioctl_streamon [videobuf2_v4l2]) from [<7f118d64>] (v4l_streamon+0x2c/0x30 [videodev])
[37559.356874] [<7f118d64>] (v4l_streamon [videodev]) from [<7f11e9b0>] (__video_do_ioctl+0x2b8/0x330 [videodev])
[37559.357060] [<7f11e9b0>] (__video_do_ioctl [videodev]) from [<7f11ed04>] (video_usercopy+0x2dc/0x630 [videodev])
[37559.357245] [<7f11ed04>] (video_usercopy [videodev]) from [<7f11f078>] (video_ioctl2+0x20/0x24 [videodev])
[37559.357427] [<7f11f078>] (video_ioctl2 [videodev]) from [<7f11765c>] (v4l2_ioctl+0xb0/0xe8 [videodev])
[37559.357531] [<7f11765c>] (v4l2_ioctl [videodev]) from [<802abff0>] (do_vfs_ioctl+0xc0/0x820)
[37559.357549] [<802abff0>] (do_vfs_ioctl) from [<802ac794>] (SyS_ioctl+0x44/0x6c)
[37559.357568] [<802ac794>] (SyS_ioctl) from [<80108360>] (ret_fast_syscall+0x0/0x1c)
[37559.357577] ---[ end trace 07789c701c79b1a4 ]---
[37560.281833] mmal_vchiq: received invalid message context 150!

6by9, 2017-07-14:
Does it work with raspistill or raspivid?
Could you try adding ```start_debug=1``` to the end of /boot/config.txt, and then run ```sudo vcdbg log assert``` after recreating the failure? Unfortunately the GPU isn't always very forthcoming about why it can't enable the camera with the standard build.

I thought I'd fixed the videobuf2 warning you've got logged, but that bit is non-fatal anyway - https://github.com/raspberrypi/linux/blob/rpi-4.9.y/drivers/media/v4l2-core/videobuf2-core.c#L1337

alexxy, 2017-07-19:
I can try rpi-4.9.y kernel and see if its a regression or not

JamesH65, 2017-09-13:
@alexxy Is this now working correctly?

alexxy, 2017-09-26:
It doesnt work on 4.9 and 4.13

6by9, 2017-09-26:
As asked [earlier](https://github.com/raspberrypi/linux/issues/2114#issuecomment-315392728), does it work with raspistill or raspivid? Use those to check for a hardware issue before hunting through V4L2.

What kernel config are you using when building the kernel?
I'm assuming you're not building a 64 bit kernel - that's known not to work.

Something very odd is going on if you're seeing "mmal_vchiq: received invalid message context"

alexxy, 2017-09-28:
Ok. I tryed with rpi-4.9.x rpi-4.12.x and rpi-4.13.x

`
rpi ~ # mmal_vc_diag camerainfo
cameras  : 1
flashes  : 2
camera 0 : port 0: 2592 x 1944 lens absent
flash 0  : flash type LED
flash 1  : flash type LED
rpi ~ # raspi
raspistill   raspivid     raspividyuv  raspiyuv     
rpi ~ # raspistill -o tmp.jpg  
mmal: mmal_vc_component_enable: failed to enable component: ENOSPC
mmal: camera component couldn't be enabled
mmal: main: Failed to create camera component
mmal: Failed to run camera app. Please check for firmware updates
`

alexxy, 2017-09-28:
rpi ~ # vcdbg log assert
001173.037: assert( source ) failed; ../../../../../middleware/confzilla/cp_front_fdt.c::cp_front_fdt_load_builtin line 115 rev f5217b8
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
003029.454: assert( card not initialised ) failed; ../../../../../filesystem/media/sdcard/sdcard.c::sdhost_close line 1045 rev f5217b8
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
009337.828: assert( limit_turbo == last_limit_turbo && arm_freq_cap == last_arm_freq_cap ) failed; ../../../../../helpers/arm_loader/arm_loader.c::tempcheck_task line 3059 rev f5217b8
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
012641.833: assert( limit_turbo == last_limit_turbo && arm_freq_cap == last_arm_freq_cap ) failed; ../../../../../helpers/arm_loader/arm_loader.c::tempcheck_task line 3059 rev f5217b8
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
016042.645: assert( limit_turbo == last_limit_turbo && arm_freq_cap == last_arm_freq_cap ) failed; ../../../../../helpers/arm_loader/arm_loader.c::tempcheck_task line 3059 rev f5217b8
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
019646.941: assert( limit_turbo == last_limit_turbo && arm_freq_cap == last_arm_freq_cap ) failed; ../../../../../helpers/arm_loader/arm_loader.c::tempcheck_task line 3059 rev f5217b8
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
106462.806: assert( camera_subsystem_state.power_enable_ref_count[port] > 0 ) failed; ../../../../../vcfw/drivers/device/camera_subsystem/camera_subsystem.c::camera_disable_ctrl line 1116 rev f5217b8
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------

JamesH65, 2017-09-28:
Not at desk, but that looks like a hardware issue to me.

I presume you have checked all the connections, both ends of the cab le are the right way round, and the connector on the camera PCB is firmly seated.

Do you have another Pi or camera module to test?

alexxy, 2017-09-29:
Yes. I have another one. So i'll test it

6by9, 2017-09-29:
Something doesn't add up if you're reporting "2592 x 1944" when you claim this is a V2.1 camera (should be 3280x2464)

```vcgencmd get_camera``` is the best command to determine how many cameras the firmware has detected, although ```mmal_vc_diag camerainfo``` seems to have confirmed it (a command I wasn't aware of!)
Have you got gpu_mem set to something sensible? ```sudo vcdbg reloc``` to confirm how much free heap you have.

alexxy, 2017-09-29:
So firmware at least see this camera

rpi ~ # vcgencmd get_camera
supported=1 detected=1

rpi ~ # vcdbg reloc

Relocatable heap version 4 found at 0x2c400000
total space allocated is 296M, with 16M relocatable, 0 legacy and 280M offline
0 legacy blocks of size 2359296

free list at 0x3eaef520
15M free memory in 1 free block(s)
largest free block is 15M bytes

0x2c400000: offline 280M
0x3dc00000: free 15M
[   4] 0x3eaef540: used  576 (refcount 1 lock count 0, size      512, align    4, data 0x3eaef560, d0rual) 'ILCS VC buffer pool'
[   3] 0x3eaef780: used 1.0M (refcount 1 lock count 8, size  1091584, align 4096, data 0x3eaf0000, d1rual) 'ARM FB'
[   2] 0x3ebfafa0: used  16K (refcount 1 lock count 0, size    16384, align   32, data 0x3ebfafc0, d0ruAl) 'audioplus_tmp_buf'
[   1] 0x3ebfefe0: used 4.0K (refcount 1 lock count 0, size        0, align 4096, data 0x3ebff000, d1rual) 'camera fast alloc arena'
small allocs not requested

alexxy, 2017-09-29:
No!
 I use default config for rpi kernel with some options removed/added (like btrfs)

$ zcat /proc/config.gz | grep CMA
CONFIG_CMA=y
# CONFIG_CMA_DEBUG is not set
# CONFIG_CMA_DEBUGFS is not set
CONFIG_CMA_AREAS=7
# CONFIG_NETWORK_SECMARK is not set
CONFIG_DMA_CMA=y
CONFIG_CMA_SIZE_MBYTES=5
CONFIG_CMA_SIZE_SEL_MBYTES=y
# CONFIG_CMA_SIZE_SEL_PERCENTAGE is not set
# CONFIG_CMA_SIZE_SEL_MIN is not set
# CONFIG_CMA_SIZE_SEL_MAX is not set
CONFIG_CMA_ALIGNMENT=8
CONFIG_INPUT_CMA3000=m
# CONFIG_INPUT_CMA3000_I2C is not set
# CONFIG_BCM_VC_CMA is not set
CONFIG_BCMA_POSSIBLE=y
CONFIG_BCMA=m
CONFIG_BCMA_HOST_SOC=y
CONFIG_BCMA_SFLASH=y
# CONFIG_BCMA_DRIVER_GMAC_CMN is not set
CONFIG_BCMA_DRIVER_GPIO=y
# CONFIG_BCMA_DEBUG is not set
CONFIG_DRM_GEM_CMA_HELPER=y
CONFIG_DRM_KMS_CMA_HELPER=y
# CONFIG_USB_HCD_BCMA is not set
CONFIG_CRYPTO_CMAC=m

JamesH65, 2017-09-29:
So, does the camera work correctly with a completely vanilla Raspbian?

pelwell, 2017-09-29:
OK - the vcdbg utility must be misinterpreting some of the heap state.

alexxy, 2017-09-29:
I dont have raspbian on it. It runs gentoo. However i can try raspbian kernel (mine uses raspbian config with some features enabled)

alexxy, 2017-09-29:
May it be related to firmware config? (/boot/firmware/config.txt)

pelwell, 2017-09-29:
Post your config.txt.

alexxy, 2017-09-29:
```
rpi ~ # cat /boot/firmware/config.txt
#kernel=u-boot.bin

#from archlinux arm sdcard image
# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# for more options see http://elinux.org/RPi_config.txt

## Some over clocking settings, govenor already set to ondemand

##None
#arm_freq=700
#core_freq=250
#sdram_freq=400
#over_voltage=0

##Modest
#arm_freq=800
#core_freq=300
#sdram_freq=400
#over_voltage=0

##Medium
#arm_freq=900
#core_freq=333
#sdram_freq=450
#over_voltage=2

##High
#arm_freq=950
#core_freq=450
#sdram_freq=450
#over_voltage=6

##Turbo
#arm_freq=1000
#core_freq=500
#sdram_freq=500
#over_voltage=6

gpu_mem_512=316
gpu_mem_256=128
cma_lwm=16
cma_hwm=32
cma_offline_start=16

start_file=start_x.elf
fixup_file=fixup_x.dat
start_x=1             # essential
gpu_mem=144           # at least, or maybe more if you wish
start_debug=1
#disable_camera_led=1  # optional, if you don't want the led to glow

kernel=vmlinuz-4.9.51-raspberrypi-rpi+
#kernel=u-boot.bin

```

pelwell, 2017-09-29:
```
gpu_mem_512=316
gpu_mem_256=128
```
The settings above look OK.

```
cma_lwm=16
cma_hwm=32
cma_offline_start=16
```
What's are these cma settings doing here? They are responsible for the offline block seen above (which wasn't a glitch after all) - please delete them.
```
start_file=start_x.elf
fixup_file=fixup_x.dat
```
These lines are unnecessary because `start_x=1` does this automatically - delete them too.
```
start_x=1             # essential
gpu_mem=144           # at least, or maybe more if you wish
start_debug=1
kernel=vmlinuz-4.9.51-raspberrypi-rpi+
```
`start_debug` overrides `start_x`, but since `start_debug` is effectively `start_x_debug` this isn't a problem.

alexxy, 2017-09-29:
Its funny, but now camera seems works at least with raspistill

JamesH65, 2017-09-30:
It's not funny at all. You had a custom config.txt , and didn't mention it, you are running a non-supported OS, and didn't mention it, and running a custom kernel and didn't mention it, and yet immediately assumed there was something wrong with our code branch, rather than your setup.

Three Raspberry Pi engineers have spent time commenting on this non-issue. Had you tried the Raspbian OS and raspistill straight away as suggested, this could have been closed ages ago.

closed 2017-09-30 by JamesH65

alexxy, 2017-10-17:
So you suggest that only supported distro is raspbian and only one kernel config? That's realy funny. 

Arent there should be documentation about camera setup? Or arent this repo for rpi kernel? Only difference of my config and vanilla one is enabling BTRFS and some USBtoSerial modules and disabling some netfilter stuff. This shouldnt influence camera at all. Config above was taken from some forum post that described similar problems with camera.

JamesH65, 2017-10-17:
Our supported distro is Raspbian, its the one we do all the testing on. The camera works on every version of Raspbian. We cannot guarantee the camera will work on any other distro, it probably will, but we have no control over the firmware version (very important), kernel version (not so important but still relevant), config. txt (very important) etc of other peoples distro's. 

Does the camera work correct on a vanilla Raspbian?

6by9, 2017-10-17:
I'll sort a PR to remove vc-cma from both firmware and kernel - it only confuses matters.