raspberrypi/linux #2115: PCF8523 RTC: hwclock: Cannot access the Hardware Clock via any known method.

amak79, 2017-07-14:
I get the following error when attempting to access an Adafruit PiRTC PCF8523 installed on a Raspberry Pi Model B (Rev 1.0, 256 MB):
```
# hwclock --debug
hwclock from util-linux 2.30
Trying to open: /dev/rtc0
No usable clock interface found.
hwclock: Cannot access the Hardware Clock via any known method.
```
I'm using Void Linux with kernel 4.9.29 and firmware 20170427 (git-08abc0bd01). I have the following in `/boot/config.txt`:
```
dtparam=i2c_arm=on
dtoverlay=i2c-rtc,pcf8523
```
The required kernel modules have been loaded:
```
# lsmod
Module                  Size  Used by
rtc_pcf8523             3619  1
i2c_bcm2835             6401  0
bcm2835_gpiomem         3751  0
uio_pdrv_genirq         3718  0
fixed                   3029  0
uio                    10281  1 uio_pdrv_genirq
i2c_dev                 6802  0
```
The kernel reports that the PCF852390 has been registered:
```
# dmesg | grep rtc
[    4.023141] rtc-pcf8523 0-0068: rtc core: registered rtc-pcf8523 as rtc0
```
The I2C address used by the PCF852390 is shown as in use:
```
# i2cdetect -y 0
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- UU -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --
```
The device attributes appear to be correct:
```
# ls -l /dev/rtc0 
crw------- 1 root root 253, 0 Jan  1  1970 /dev/rtc0
```
However if I disable the `i2c-rtc,pcf8523` device tree overlay in `/boot/config.txt` and register the PCF8523 manually with;
```
# echo pcf8523 0x68 > /sys/class/i2c-adapter/i2c-0/new_device
```
I can then read/write time with `hwclock`. How do I get `hwclock` to work when using the device tree overlay?

pelwell, 2017-07-14:
It works for me on a Pi 3 running Raspbian, but your system differs in quite a few ways. With the overlay reinstated, try this (as root) and report the output:
```
rmmod rtc_pcf8523
i2cdetect -y 0
i2cdump -y -r 0-31 0 0x68
hwclock --debug
modprobe rtc_pcf8523
hwclock --debug
```

amak79, 2017-07-14:
Thank you for the reply.

I'm unable to remove the `rtc_pcf8523` module with the overlay enabled:
```
# rmmod rtc_pcf8523
rmmod: ERROR: Module rtc_pcf8523 is in use
```
`lsmod` shows that the `rtc_pcf8523` module is in use, but it doesn't list which module is using it:
```
# lsmod
Module                  Size  Used by
rtc_pcf8523             3619  1
i2c_bcm2835             6401  0
bcm2835_gpiomem         3751  0
uio_pdrv_genirq         3718  0
fixed                   3029  0
uio                    10281  1 uio_pdrv_genirq
i2c_dev                 6802  0
```
Is that normal?

With the overlay disabled:
```
# i2cdetect -y 0
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --
# i2cdump -y -r 0-31 0 0x68
No size specified (using byte-data access)
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f    0123456789abcdef
00: 80 00 08 18 51 14 14 05 07 17 80 80 80 80 00 00    ?.??Q?????????..
10: 07 00 07 6e 00 66 00 00 00 00 00 00 00 00 00 00    ?.?n.f..........
```

pelwell, 2017-07-14:
I think something in the OS has opened /dev/rtc0 at start of day and is preventing the hwclock command from reading it. You can go some way to verifying this by loading the overlay at runtime instead. Remove the `dtoverlay=` from config.txt, reboot the Pi, then run (as root, again):
```
dtoverlay i2c-rtc pcf8523
```
With luck you should end up with /dev/rtc0 and be able to run `hwclock --debug` successfully.

If this turns to be the case, you may be able to use `lsof` to find the culprit. Without rebooting:
```
apt-get install lsof
cat /dev/rtc0 &
lsof /dev/rtc0
```
You should see the `cat` command listed as having the RTC open. Now put the overlay back in config.txt, reboot and run lsof again (you seem to need to be root to guarantee that lsof works).

amak79, 2017-07-14:
The `/dev/rtc0` device was being opened by chrony:
```
# lsof /dev/rtc0
COMMAND PID   USER   FD   TYPE DEVICE SIZE/OFF NODE NAME
chronyd 210 chrony    3u   CHR  253,0      0t0 5180 /dev/rtc0
```
`hwclock` ran successfully once the chrony service was stopped. I think `hwclock` could really do with more informative error messages.

Thank you for the help.

closed 2017-07-14 by amak79