raspberrypi/linux #2116: 4.4.x: kernel hangs with 'dtoverlay=vc4-kms-v3d' while writing to /dev/fb0

piso77, 2017-07-14:
I've been testing a small program (psplash - see below) to display a bitmap on framebuffer when i hit a serious condition - if i enable the vc4-kms-v3d overlay, and then open the fb0 to display the bitmap, the board hangs and the only way to recover is to pull the plug:

pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.4.50-v7+ #1 SMP Fri Jul 14 15:42:03 CEST 2017 armv7l GNU/Linux

pi@raspberrypi:~ $ sudo ./psplash/psplash

and on serial console:

raspberrypi login: [ 5376.327290] NMI watchdog: BUG: soft lockup - CPU#0 stuck for 789s! [psplash:766]
[ 5376.327296] NMI watchdog: BUG: soft lockup - CPU#3 stuck for 789s! [swapper/3:0]

This on a up to date Raspbian system (but a 4.4.y / bcm2709_defconfig kernel + SOFTLOCKUP + CONFIG_PREEMPT).

Since it appears all the cpu are stuck somewhere in kernel code, i tried to dump the lock list with sysrq over serial, but it's not working, and i only get garbage:

Raspbian GNU/Linux 8 raspberrypi ttyS0

raspberrypi login: ����������������������������������������������

and the serial appears to be stuck after this - i probably ended up in the sysrq console, but something is borked with the serial line itself.

Then i tried to narrow down the breakage window, but it didn't go really well (these refer to raspbian's github kernel branches - rpi-4.4.y, rpi-4.5.y, ..., rpi-4.9.y using bcm2709_defconfig)

rpi-4.5.y: same as rpi-4.4.y
rpi-4.6.y: hangs during boot when enabling 'dtoverlay=vc4-kms-v3d', while it's fine if i don't apply it
rpi-4.7.y: 
...
[   16.608976] vc4-drm soc:gpu: bound 3f902000.hdmi (ops vc4_hdmi_ops [vc4])
[   16.609186] vc4-drm soc:gpu: bound 3f206000.pixelvalve (ops vc4_crtc_ops [vc4])
[   16.609356] vc4-drm soc:gpu: bound 3f207000.pixelvalve (ops vc4_crtc_ops [vc4])
[   16.609545] vc4-drm soc:gpu: bound 3f807000.pixelvalve (ops vc4_crtc_ops [vc4])
[   16.609621] vc4-drm soc:gpu: bound 3f400000.hvs (ops vc4_hvs_ops [vc4])
[   16.609702] [drm:vc4_v3d_bind [vc4]] *ERROR* V3D_IDENT0 read 0xdeadbeef instead of 0x02443356
[   16.609771] vc4-drm soc:gpu: failed to bind 3fc00000.v3d (ops vc4_v3d_ops [vc4]): -22
[   16.610020] vc4-drm soc:gpu: master bind failed: -22
[   16.610055] vc4-drm: probe of soc:gpu failed with error -22

it appears vc4 fails to attach, notice the '0xdeadbeef' ptr, and there was no /dev/fb0 after that.

rpi-4.8.y and rpi-4.9.y: 

i can execute psplash, the framebuffer turns black and no bitmap is displayed, but the system is alive and i can CTRL+C the program with no ill effects (except for fb remaning full black)

Psplash can be found here git://git.yoctoproject.org/psplash:

$ git clone git://git.yoctoproject.org/psplash
$ cd psplash
$ autoreconf --install
$ ./configure
$ make
$ sudo ./psplash

and if you don't load the vc4 overlay above, ift will display a bitmap on fb0.

So, does this problem sound familiar? Or do you have any idea how to debug this further?
I checked drivers/gpu/drm/vc4* and i couldn't find debug options.

Generic system info:

pi@raspberrypi:~ $ cat /proc/cmdline 
8250.nr_uarts=1 cma=256M@256M bcm2708_fb.fbwidth=1680 bcm2708_fb.fbheight=1050 bcm2708_fb.fbswap=1 dma.dmachans=0x7f35 bcm2709.boardrev=0xa02082 bcm2709.serial=0x9710fb23 bcm2709.uart_clock=48000000 smsc95xx.macaddr=B8:27:EB:10:FB:23 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait

pi@raspberrypi:~ $ cat /boot/config.txt | grep -v ^#

enable_uart=1

dtoverlay=vc4-kms-v3d
dtoverlay=i2c-rtc,ds3231

pi@raspberrypi:~ $ dpkg -l | grep raspi
ii  raspi-config                   20170518                         all          Raspberry Pi configuration tool
ii  raspi-copies-and-fills         0.5-1                            armhf        ARM-accelerated versions of selected functions from string.h

pi@raspberrypi:~ $ dpkg -l | grep boot
ii  insserv                        1.14.0-5                         armhf        boot sequence organizer using LSB init.d script dependency information
ii  klibc-utils                    2.0.4-2+rpi1                     armhf        small utilities built with klibc for early boot
ii  libplymouth4:armhf             0.9.2-3+rpi1                     armhf        graphical boot animation and logger - shared libraries
pi  plymouth                       0.9.2-3+rpi1                     armhf        boot animation, logger and I/O multiplexer
ii  raspberrypi-bootloader         1.20170703-1                     armhf        Raspberry Pi bootloader
ii  raspberrypi-kernel             1.20170703-1                     armhf        Raspberry Pi bootloader

piso77, 2017-07-14:
by adding drm/debug=0x1e i get this during boot:

...
[   15.054996] [drm:drm_fb_helper_hotplug_event]                                                                                              [109/9400]
[   15.055035] [drm:drm_helper_probe_single_connector_modes_merge_bits] [CONNECTOR:19:HDMI-A-1]
[   15.080582] [drm:drm_helper_probe_single_connector_modes_merge_bits] [CONNECTOR:19:HDMI-A-1] probed modes :
[   15.080613] [drm:drm_mode_debug_printmodeline] Modeline 54:"1680x1050" 60 146250 1680 1784 1960 2240 1050 1053 1059 1089 0x48 0x6
[   15.080633] [drm:drm_mode_debug_printmodeline] Modeline 60:"1280x1024" 75 135000 1280 1296 1440 1688 1024 1025 1028 1066 0x40 0x5
[   15.080654] [drm:drm_mode_debug_printmodeline] Modeline 59:"1280x1024" 60 108000 1280 1328 1440 1688 1024 1025 1028 1066 0x40 0x5
[   15.080673] [drm:drm_mode_debug_printmodeline] Modeline 58:"1280x960" 60 108000 1280 1376 1488 1800 960 961 964 1000 0x40 0x5
[   15.080693] [drm:drm_mode_debug_printmodeline] Modeline 61:"1152x864" 75 108000 1152 1216 1344 1600 864 865 868 900 0x40 0x5
[   15.080712] [drm:drm_mode_debug_printmodeline] Modeline 67:"1024x768" 75 78800 1024 1040 1136 1312 768 769 772 800 0x40 0x5
[   15.080732] [drm:drm_mode_debug_printmodeline] Modeline 57:"1024x768" 75 78750 1024 1040 1136 1312 768 769 772 800 0x40 0x5
[   15.080751] [drm:drm_mode_debug_printmodeline] Modeline 68:"1024x768" 70 75000 1024 1048 1184 1328 768 771 777 806 0x40 0xa
[   15.080770] [drm:drm_mode_debug_printmodeline] Modeline 69:"1024x768" 60 65000 1024 1048 1184 1344 768 771 777 806 0x40 0xa
[   15.080789] [drm:drm_mode_debug_printmodeline] Modeline 70:"832x624" 75 57284 832 864 928 1152 624 625 628 667 0x40 0xa
[   15.080808] [drm:drm_mode_debug_printmodeline] Modeline 56:"800x600" 75 49500 800 816 896 1056 600 601 604 625 0x40 0x5
[   15.080828] [drm:drm_mode_debug_printmodeline] Modeline 62:"800x600" 60 40000 800 840 968 1056 600 601 605 628 0x40 0x5
[   15.080847] [drm:drm_mode_debug_printmodeline] Modeline 63:"640x480" 75 31500 640 656 720 840 480 481 484 500 0x40 0xa
[   15.080866] [drm:drm_mode_debug_printmodeline] Modeline 64:"640x480" 60 25200 640 656 752 800 480 490 492 525 0x40 0xa
[   15.080885] [drm:drm_mode_debug_printmodeline] Modeline 73:"640x480" 60 25175 640 656 752 800 480 490 492 525 0x40 0xa
[   15.080904] [drm:drm_mode_debug_printmodeline] Modeline 65:"720x400" 70 28320 720 738 846 900 400 412 414 449 0x40 0x6
[   15.080923] [drm:drm_mode_debug_printmodeline] Modeline 55:"640x350" 70 25170 640 656 752 800 350 387 389 449 0x40 0x9
[   15.080953] [drm:drm_setup_crtcs] 
[   15.080969] [drm:drm_enable_connectors] connector 19 enabled? yes
[   15.080983] [drm:drm_target_preferred] looking for cmdline mode on connector 19
[   15.080996] [drm:drm_target_preferred] looking for preferred mode on connector 19 0
[   15.081008] [drm:drm_target_preferred] found mode 1680x1050
[   15.081021] [drm:drm_setup_crtcs] picking CRTCs for 2048x2048 config
[   15.081038] [drm:drm_setup_crtcs] desired mode 1680x1050 set on crtc 43 (0,0)
[   15.081077] [drm:drm_atomic_state_init] Allocated atomic state b9019500
[   15.081098] [drm:drm_atomic_get_plane_state] Added [PLANE:20] b98e4d00 state to b9019500
[   15.081114] [drm:drm_atomic_get_plane_state] Added [PLANE:22] b98e4a00 state to b9019500
[   15.081128] [drm:drm_atomic_set_crtc_for_plane] Link plane state b98e4a00 to [NOCRTC]
[   15.081141] [drm:drm_atomic_set_fb_for_plane] Set [NOFB] for plane state b98e4a00
[   15.081158] [drm:drm_atomic_get_plane_state] Added [PLANE:23] b9008900 state to b9019500
[   15.081170] [drm:drm_atomic_set_crtc_for_plane] Link plane state b9008900 to [NOCRTC]
[   15.081182] [drm:drm_atomic_set_fb_for_plane] Set [NOFB] for plane state b9008900
[   15.081197] [drm:drm_atomic_get_plane_state] Added [PLANE:24] b9008d00 state to b9019500
[   15.081209] [drm:drm_atomic_set_crtc_for_plane] Link plane state b9008d00 to [NOCRTC]
[   15.081221] [drm:drm_atomic_set_fb_for_plane] Set [NOFB] for plane state b9008d00
[   15.081236] [drm:drm_atomic_get_plane_state] Added [PLANE:25] b9008e00 state to b9019500
[   15.081248] [drm:drm_atomic_set_crtc_for_plane] Link plane state b9008e00 to [NOCRTC]
[   15.081259] [drm:drm_atomic_set_fb_for_plane] Set [NOFB] for plane state b9008e00
[   15.081274] [drm:drm_atomic_get_plane_state] Added [PLANE:26] b9008100 state to b9019500
[   15.081286] [drm:drm_atomic_set_crtc_for_plane] Link plane state b9008100 to [NOCRTC]
[   15.081297] [drm:drm_atomic_set_fb_for_plane] Set [NOFB] for plane state b9008100
[   15.081313] [drm:drm_atomic_get_plane_state] Added [PLANE:27] b9008200 state to b9019500
[   15.081324] [drm:drm_atomic_set_crtc_for_plane] Link plane state b9008200 to [NOCRTC]
[   15.081336] [drm:drm_atomic_set_fb_for_plane] Set [NOFB] for plane state b9008200
[   15.081350] [drm:drm_atomic_get_plane_state] Added [PLANE:28] b9008c00 state to b9019500
[   15.081362] [drm:drm_atomic_set_crtc_for_plane] Link plane state b9008c00 to [NOCRTC]
[   15.081374] [drm:drm_atomic_set_fb_for_plane] Set [NOFB] for plane state b9008c00
[   15.081389] [drm:drm_atomic_get_plane_state] Added [PLANE:29] b9008a00 state to b9019500
[   15.081401] [drm:drm_atomic_set_crtc_for_plane] Link plane state b9008a00 to [NOCRTC]
[   15.081412] [drm:drm_atomic_set_fb_for_plane] Set [NOFB] for plane state b9008a00
[   15.081427] [drm:drm_atomic_get_plane_state] Added [PLANE:30] b9008800 state to b9019500
[... cut ...]
[   15.082675] [drm:drm_atomic_connectors_for_crtc] State b9019500 has 0 connectors for [CRTC:32]
[   15.082689] [drm:drm_atomic_connectors_for_crtc] State b9019500 has 1 connectors for [CRTC:43]
[   15.082710] [drm:drm_atomic_commit] commiting b9019500
[   15.082758] [drm:drm_atomic_state_default_clear] Clearing atomic state b9019500
[   15.082817] [drm:drm_atomic_state_free] Freeing atomic state b9019500
pi@raspberrypi:~ $ 
pi@raspberrypi:~ $ sudo ./psplash/psplash
[freeze here and no more msgs]

popcornmix, 2017-07-14:
Can you report this here: https://github.com/anholt/linux/issues/
Eric Anholt is developer of the vc4-kms-v3d driver and handles issues there.

piso77, 2017-07-25:
4.7 was the last branch with this problem, while 4.8 was fine, so i bisected between the two, and after every bisect i rebased the bradcom BSP on top of that to enable the bcm2709 soc.

And i found this to be the fix for the above board crash:

[https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/drivers/gpu/drm/drm_fb_cma_helper.c?id=ce0c5757681052eb8965209129f28362c3a4d8a7](url)

it implements the mmap() syscall on fb devices using the dma API, though it's not entirely clear to me why or what it really does - any chance anyone can shed some light? @popcornmix @anholt ???

Anyhow, with this fix backported to 4.4 (it applies cleanly BTW), i can paint on fb0 even when the vc4-kms-v3d overlay is applied, without any ill effects.

popcornmix, 2017-07-25:
This isn't really my area, and we we have moved on from the 4.4 kernel so it's not something we'll backport.
But the information could be useful for someone else in the same situation.
Any reason why you haven't moved to 4.9?
You won't get any updates, bug fixes, performance improvements, new features etc on 4.4.

piso77, 2017-07-25:
Because for Ubuntu / Xenial, we picked up 4.4 and all the topic branches (e.g. raspi2 that actually supports rpi2 and rpi3) are based off that:

https://git.launchpad.net/~ubuntu-kernel/ubuntu/+source/linux/+git/xenial/log/?h=raspi2

so we'll stick to 4.4 until Xenial expires, and i'm the mantainer of that kernel.