raspberrypi/linux #2117: 64-bit version of bcm2835-v4l2?

UMRnInside, 2017-07-15:
I compiled 64-bit kernel (and its modules),but I **CANNOT** find `bcm2835-v4l2`
I wonder **WHY** is it missing and **HOW** to compile a 64-bit version of `bcm2835-v4l2`

6by9, 2017-07-17:
64bit kernels are not currently being supported by Pi Towers. Some drivers have been updated (often by various members of the community), but not all.

There are various issues with switching the IPC with the firmware to use 64 bit pointers whilst maintaining backwards compatibility. The GPU itself is inherently 32bit, and some of the IPC structures got defined using void* so need to be modified to u32/uint32_t and then mapped somewhere.

Some of the work has been done upstream in 4.12, so if you look at https://github.com/raspberrypi/linux/tree/rpi-4.13.y/drivers/staging/vc04_services/bcm2835-camera you'll find what has been done.

JamesH65, 2017-09-13:
Closing this issue as questions answered/issue resolved.

closed 2017-09-13 by JamesH65