raspberrypi/linux #2122: 'LOCAL BUILDING' with Raspberry Pi 3 failed

AZO234, 2017-07-17:
I tried to 'LOCAL BUILDING' with Raspberry Pi 3.

git clone, cd linux, is OK.
When
make bcm2709_defconfig
occurs Error.

*** Error during writing of the configuration.

scripts/kconfig/Makefile:112: recipe for target 'bcm2709_defconfig' failed
make[1]: *** [bcm2709_defconfig] Error 1
Makefile:546: recipe for target 'bcm2709_defconfig' failed
make: *** [bcm2709_defconfig] Error 2

popcornmix, 2017-07-17:
https://www.raspberrypi.org/documentation/linux/kernel/building.md

pelwell, 2017-07-17:
I think you may have a filesystem or permissions problem - run `sudo make clean` then try again.

AZO234, 2017-07-17:
To pelwell

I ran script with sudo.
Directory's parmissions is not for root.
Sorry.

closed 2017-07-17 by AZO234