raspberrypi/linux #2124: WLAN interface disappeared on rpi-4.12.y arm64

UMRnInside, 2017-07-19:
Part of the dmesg file:
```
[    0.000000] Virtual kernel memory layout:
[    0.000000]     modules : 0xffffff8000000000 - 0xffffff8008000000   (   128 MB)
[    0.000000]     vmalloc : 0xffffff8008000000 - 0xffffffbebfff0000   (   250 GB)
[    0.000000]       .text : 0xffffff9804480000 - 0xffffff9804b60000   (  7040 KB)
[    0.000000]     .rodata : 0xffffff9804b60000 - 0xffffff9804da0000   (  2304 KB)
[    0.000000]       .init : 0xffffff9804da0000 - 0xffffff9805040000   (  2688 KB)
[    0.000000]       .data : 0xffffff9805040000 - 0xffffff9805113a00   (   847 KB)
[    0.000000]        .bss : 0xffffff9805113a00 - 0xffffff98051c1bbc   (   697 KB)
[    0.000000]     fixed   : 0xffffffbefe7fd000 - 0xffffffbefec00000   (  4108 KB)
[    0.000000]     PCI I/O : 0xffffffbefee00000 - 0xffffffbeffe00000   (    16 MB)
[    0.000000]     vmemmap : 0xffffffbf00000000 - 0xffffffc000000000   (     4 GB maximum)
[    0.000000]               0xffffffbf4c000000 - 0xffffffbf4cdc0000   (    13 MB actual)
[    0.000000]     memory  : 0xffffffd300000000 - 0xffffffd337000000   (   880 MB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 25021 entries in 98 pages
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000] NR_IRQS:64 nr_irqs:64 0
[    0.000000] Failed to get local register map. FIQ is disabled for cpus > 1
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000007] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000238] Console: colour dummy device 80x25
[    0.000262] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=19200)
[    0.000282] pid_max: default: 32768 minimum: 301
[    0.000585] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes)
[    0.000600] Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes)
[    0.005080] ASID allocator initialised with 65536 entries
[    0.009354] EFI services will not be available.
[    0.011140] smp: Bringing up secondary CPUs ...
[    0.018340] Detected VIPT I-cache on CPU1
[    0.018407] CPU1: Booted secondary processor [410fd034]
[    0.025441] Detected VIPT I-cache on CPU2
[    0.025485] CPU2: Booted secondary processor [410fd034]
[    0.032576] Detected VIPT I-cache on CPU3
[    0.032617] CPU3: Booted secondary processor [410fd034]
[    0.032715] smp: Brought up 1 node, 4 CPUs
[    0.032760] SMP: Total of 4 processors activated.
[    0.032779] CPU features: detected feature: 32-bit EL0 Support
[    0.033012] CPU: All CPU(s) started at EL2
[    0.033052] alternatives: patching kernel code
[    0.034315] devtmpfs: initialized
[    0.047504] DMI not present or invalid.
[    0.047817] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275000 ns
[    0.047847] futex hash table entries: 1024 (order: 5, 131072 bytes)
[    0.048478] pinctrl core: initialized pinctrl subsystem
[    0.049165] NET: Registered protocol family 16
[    0.055333] cpuidle: using governor menu
[    0.055614] vdso: 2 pages (1 code @ ffffff9804b67000, 1 data @ ffffff9805044000)
[    0.055644] hw-breakpoint: found 6 breakpoint and 4 watchpoint registers.
[    0.059667] DMA: preallocated 256 KiB pool for atomic allocations
[    0.059831] Serial: AMBA PL011 UART driver
[    0.062469] bcm2835-mbox 3f00b880.mailbox: mailbox enabled
[    0.063204] uart-pl011 3f201000.serial: could not find pctldev for node /soc/gpio@7e200000/uart0_pins, deferring probe
[    0.103411] bcm2835-dma 3f007000.dma: DMA legacy API manager at ffffff800804b000, dmachans=0x1
[    0.105362] SCSI subsystem initialized
[    0.105534] usbcore: registered new interface driver usbfs
[    0.105622] usbcore: registered new interface driver hub
[    0.105756] usbcore: registered new device driver usb
[    0.105975] dmi: Firmware registration failed.
[    0.107157] raspberrypi-firmware soc:firmware: Attached to firmware from 2017-04-27 17:27
[    0.108535] clocksource: Switched to clocksource arch_sys_counter
[    0.211980] VFS: Disk quotas dquot_6.6.0
[    0.212089] VFS: Dquot-cache hash table entries: 512 (order 0, 4096 bytes)
[    0.212303] FS-Cache: Loaded
[    0.212585] CacheFiles: Loade
```
I'm on **arm64** .
`sudo modprobe brcmfmac` doesn't work.

pelwell, 2017-07-19:
Sorry, but we don't support 64-bit kernels so I can't help much. The WiFi interface is connected via SDIO on mmc1 using GPIOs 34-39; you could check that all those things are in place. "dmesg | grep mmc" and "raspi-gpio get" might give some clues.

skoehler, 2017-07-28:
Same here. It does work with rpi-4.9.y and rpi-4.10.y. Didn't test rpi-4.11.y yet. I'm also using arm64 kernels. Unfortunately, I don't have the raspi-gpio tool (I'm using Arch Linux userland) and "dmesg|grep mmc" doesn't show anything suspicious.

pelwell, 2017-07-28:
resolution can easily be built from source - the repo is here: https://github.com/RPi-Distro/raspi-gpio

UMRnInside, 2017-07-28:
There _seems_ to be a solution:
* `make menuconfig`
* _Device Drivers_ -> _MMC/SD/SDIO card support_
* Enable everything related to BCM2835

@pelwell @skoehler

skoehler, 2017-07-28:
@UMRnInside  I used `make bcmrpi3_defconfig` so I expected MMC support to be enabled. Also booting from SD card was no problem. However, I will check tonight. Was there a particular CONFIG_* option that made things work for you?

UMRnInside, 2017-07-28:
I `make bcmrpi3_defconfig` first,then I "Enable everything related to BCM2835".
I haven't edited _.config_ file **on my own**.

skoehler, 2017-07-28:
@UMRnInside All the options in `make menuconfig` have names. You can find out about their names and their description by pressing the ? on your keyboard. You can also search for options by pressing / on your keyboard.

It turns out that 4.12.y has a new option called CONFIG_MMC_BCM2835_MMC which is not enabled by `make bcmrpi3_defconfig`. I activated that option and that option only in addition to `make bcmrpi3_defconfig` and the integrated wifi is working now. Hooray!

skoehler, 2017-07-28:
Hold on a minute, it's not simple. There are 3 MMC-related drivers:
- `CONFIG_MMC_BCM2835`
- `CONFIG_MMC_SDHCI_IPROC`
- `CONFIG_MMC_BCM2835_MMC`

I believe that the first 2 are from the mainline kernel while the third is a driver by the raspberry pi foundation. The driver by the raspberry pi foundation has to be renamed, as it conflicts with the driver from mainline. At least that's my best guess.

Now you should only enable EITHER the first two OR the third. Dont mix them. It seems that the driver by the raspberry pi foundation is superior as it mentions DMA support.

skoehler, 2017-07-28:
There is a bug that `CONFIG_MMC_BCM2835_PIO_DMA_BARRIER`, which is a sub-item of `CONFIG_MMC_BCM2835_MMC`, actually disappears when `CONFIG_MMC_BCM2835` is deselected. But it really should disappear. I suspect this is an artifact of renaming the driver by the raspberry pi foundation.

UMRnInside, 2017-07-29:
@skoehler 

* _bcmrpi3_defconfig_ is **original** .

* I ran `sudo modprobe configs && zcat /proc/config.gz | grep MMC | grep BCM2835`
Then I got these:
```
CONFIG_MMC_BCM2835_MMC=m
CONFIG_MMC_BCM2835_DMA=y
CONFIG_MMC_BCM2835_PIO_DMA_BARRIER=2
CONFIG_MMC_BCM2835_SDHOST=y
CONFIG_MMC_BCM2835=y
```

* BTW, _bcmrpi3_defconfig_ does **NOT** exist in _arch/arm/configs_
They _probably_ use these defconfig files:
```
$ ls arch/arm/configs | grep bcm
bcm2709_defconfig
bcm2835_defconfig
bcmrpi_defconfig
```

kukabu, 2017-07-29:
try it
CONFIG_MMC_BCM2835_MMC=y
CONFIG_MMC_BCM2835=n

skoehler, 2017-07-29:
@kukabu I am running a kernel with `CONFIG_MMC_BCM2835_MMC=y` and `CONFIG_MMC_BCM2835=n` right now. It's working great. But I had to patch Kconfig so that the DMA_PIO_BARRIER option doesn't disappear.

JamesH65, 2017-09-13:
Closing this issue as questions answered/issue resolved.

closed 2017-09-13 by JamesH65

raidensnake, 2018-02-07:
@skoehler you mean this right in drivers/mmc/host/Kconfig?

`config MMC_BCM2835_PIO_DMA_BARRIER
	int "Block count limit for PIO transfers"
-	depends on MMC_BCM2835_MMC && MMC_BCM2835_DMA
+	depends on MMC_BCM2835_MMC`