raspberrypi/linux #2128: Alignment trap

andersthomson, 2017-07-22:
Hi,

As can be seen in the attached log, I've stated to get alignment trap errors. This is possibly related to #2107, as it's the same process that triggers it. It's /usr/sbin/run-crons (attached too), which yields this in the email cron sends out:

/usr/sbin/run-crons: line 120: 30083 Bus error               (core dumped) find "${LASTRUNDIR}/" -name cron.$BASE $TIME -exec rm {} \; 2> /dev/null
/usr/sbin/run-crons: line 120: 30109 Bus error               (core dumped) find "${LASTRUNDIR}/" -name cron.$BASE $TIME -exec rm {} \; 2> /dev/null
/usr/sbin/run-crons: line 120: 30115 Bus error               (core dumped) find "${LASTRUNDIR}/" -name cron.$BASE $TIME -exec rm {} \; 2> /dev/null
/usr/sbin/run-crons: line 178: 30128 Bus error               (core dumped) find "${LASTRUNDIR}/" -newer "${LASTRUNDIR}" -exec /bin/rm -f {} \; 2> /dev/null

I haven't touched the cron setup in ages and other cron tasks run fine. 

find(1) works perfectly in a ton of other scripts I have. It just sigbuses on when run from run-crons.

The machine is still up, and giving this error each hour. Anything I should test?

FYI:
/opt/vc/bin/vcgencmd get_throttled
throttled=0x0

[alignment_trap_bus_error.txt](https://github.com/raspberrypi/linux/files/1167143/alignment_trap_bus_error.txt)
[run-crons.txt](https://github.com/raspberrypi/linux/files/1167145/run-crons.txt)

JamesH65, 2017-09-13:
@andersthomson Is this still a live issue?

andersthomson, 2017-09-18:
Nope. For whatever reason the box is now behaving rather well. The odd issue with us writes the tv sticks, but other than that it's happy.

closed 2017-12-04 by JamesH65