raspberrypi/linux #2130: Can't read ID EEPROM on Pi Zero W

Naegolus, 2017-07-23:
Hi,

the following works in Pi Zero but not Pi Zero W:
`root@raspberrypi:/sys/class/i2c-adapter/i2c-0/0-0050# hd eeprom 
hd: eeprom: Connection timed out`

Using kernel version:
`# uname -a
Linux raspberrypi 4.9.39+ #1020 Sat Jul 22 14:08:01 BST 2017 armv6l GNU/Linux`

pelwell, 2017-07-24:
We don't recommend using I2C0 (i2c_vc) because it isn't guaranteed to be safe on all platforms. The eepflash shell script we provide for HAT developers uses a software I2C implementation instead:
```
sudo dtoverlay i2c-gpio i2c_gpio_sda=0 i2c_gpio_scl=1
```
This creates /dev/i2c-3 on GPIOs 0 and 1.

Having said that, in the Pi Zero W case there's actually no reason why it shouldn't be OK - it's just that the in-built dt-blob that the firmware uses to configure the pins is leaving the I2C0 alt-function enabled on GPIOs 28 & 29 at the same time as you are trying to use it on GPIOs 0 & 1 (mapping an I2C interface to multiple pin groups breaks it).

All camera-capable platforms used to statically enable I2C0 on 28&29, but then the firmware was changed to enable it as needed, allowing the same I2C interface to be multiplexed between several pin groups. At the same time, the static mapping was removed, but because of parallel development in a side branch the Zero W pin settings were missed, leading to the problem you see now.

I've tested with a modified dt-blob on a Zero W and /dev/i2c0 works as expected. The fixed dt-blob will appear in a future firmware release, but until then you have a few options:
a) Use `i2c-gpio` as show above.
b) Run the following after booting:
```
raspi-gpio set 28 ip
raspi-gpio set 29 ip
```
c) Download a replacement dt-blob.bin from [here](https://drive.google.com/file/d/0B_P-i4u-SLBXT3h6OE1CNXA0STA/view?usp=sharing), copy it into the boot partition (/boot) and reboot.
This external dt-blob will be used in preference to the internal one, so by installing it you will masking any future updates to the built-in version.

I'll add a comment here when a new firmware is released, at which point you can update and remove your workaround.

Naegolus, 2017-07-24:
Wow. Thank you!

Tested the modified blob => Working

closed 2017-07-24 by Naegolus