raspberrypi/linux #2131: tpa6130a2 module need patch update to work with raspodac3 module

jgrulich, 2017-07-24:
The present kernel 4.x need the tpa6130a2 patch to work again with the raspidac3. The tpa6130a2 driver modification was removed when the kernel was updated from version 4.4 and was not applied again for new version. Now the tpa6130a2 is missing the "Headphone" control.

pelwell, 2017-07-24:
Please provide an updated patch, as a Pull Request, that applies to the 4.9 tree.

JamesH65, 2017-09-13:
@jgrulich  I cannot see a PR for this. Are you intending to provide one or can this issue be closed?

JamesH65, 2017-12-04:
Closing due to lack of activity. Reopen if you feel this issue is still relevant.

closed 2017-12-04 by JamesH65

czechnorris, 2021-03-09:
Hi, I have just encountered this issue when trying to make raspidac3 work.
This is the original commit from @jgrulich that was included in 4.4 and is missing since then: 570d2d99cbf2b830e0d1be01dcc6fdd339b4d157
Can we include it in the latest tree? Should I create a PR? New issue maybe?

jgrulich, 2021-03-10:
@czechnorris the tpa6130a2 module patching is no more supported and raspidac3 was deprecated a few years ago due to other competitors like IQaudio.
But if you have one and like to use it, you may use direct I2C control: https://www.grulich.eu/files/hp.tar.gz