raspberrypi/linux #2132: wlan0 don't exist with kernel  4.13.0-rc1-v7  in RPi3

dakota72, 2017-07-24:
Hi,
I compilated the last kernel version for my rpi3
but The interface wlan0 is missing.
the result of ifconfig
eth0      Link encap:Ethernet  HWaddr b8:27:eb:08:b7:16
          inet addr:10.53.1.60  Bcast:10.53.1.255  Mask:255.255.255.0
          inet6 addr: fe80::bd69:11c7:c80c:9e21/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1870 errors:0 dropped:13 overruns:0 frame:0
          TX packets:1079 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:137445 (134.2 KiB)  TX bytes:186082 (181.7 KiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:8 errors:0 dropped:0 overruns:0 frame:0
          TX packets:8 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:480 (480.0 B)  TX bytes:480 (480.0 B)

wlan0 isn't in the list...why?

JamesH65, 2017-07-24:
Just tried this, seems to work fine. Are you sure all the modules and dtb stuff has been correctly installed on the SD card? 

Note though that this is a very early 4.13, so support is limited and there are likely to be some issues.

lategoodbye, 2017-07-24:
@dakota72 Please provide a dmesg output.

dakota72, 2017-07-24:
I have followed this guide:
http://ithink.ch/blog/2013/09/08/patching_the_linux_kernel_to_install_the_xin-mo_dual_arcade_driver_on_a_raspberry_pi.html

after compile I did this:

sudo cp arch/arm/boot/Image /boot/kernel_new.img
mkdir ~/modules
export MODULES_TEMP=~/modules
make INSTALL_MOD_PATH=${MODULES_TEMP} modules_install
mkdir ~/modules
export MODULES_TEMP=~/modules
make INSTALL_MOD_PATH=${MODULES_TEMP} modules_install
cd /lib
sudo cp -R ~/modules/lib/firmware/ .
sudo cp -R ~/modules/lib/modules/ .

it's possible to my .config are wrong?
which .config I must be used?

dakota72, 2017-07-24:
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.13.0-rc1-v7 (pi@retropie) (gcc version 4.9.2 (Raspbian 4.9.2-10)) #6 SMP Tue Jul 18 22:57:48 UTC 2017
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 8 MiB at 0x2e400000
[    0.000000] On node 0 totalpages: 192512
[    0.000000] free_area_init_node: node 0, pgdat 80c724c0, node_mem_map add64000
[    0.000000]   Normal zone: 1692 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 192512 pages, LIFO batch:31
[    0.000000] percpu: Embedded 17 pages/cpu @aef95000 s38848 r8192 d22592 u69632
[    0.000000] pcpu-alloc: s38848 r8192 d22592 u69632 alloc=17*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 190820
[    0.000000] Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=1920 bcm2708_fb.fbheight=1080 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet loglevel=3 consoleblank=0 usbhid.quirks=0x16c0:0x75e1:0x040
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 741380K/770048K available (7168K kernel code, 497K rwdata, 1976K rodata, 1024K init, 704K bss, 20476K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xaf800000 - 0xff800000   (1280 MB)
    lowmem  : 0x80000000 - 0xaf000000   ( 752 MB)
    modules : 0x7f000000 - 0x80000000   (  16 MB)
      .text : 0x80008000 - 0x80800000   (8160 kB)
      .init : 0x80b00000 - 0x80c00000   (1024 kB)
      .data : 0x80c00000 - 0x80c7c73c   ( 498 kB)
       .bss : 0x80c83f7c - 0x80d34374   ( 705 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 24268 entries in 72 pages
[    0.000000] Hierarchical RCU implementation.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000009] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000022] Switching to timer-based delay loop, resolution 52ns
[    0.000327] Console: colour dummy device 80x30
[    0.000350] console [tty1] enabled
[    0.000373] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.000389] pid_max: default: 32768 minimum: 301
[    0.000701] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000722] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001596] CPU: Testing write buffer coherency: ok
[    0.002106] CPU0: update cpu_capacity 1024
[    0.002115] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.002556] Setting up static identity map for 0x100000 - 0x10003c
[    0.002700] Hierarchical SRCU implementation.
[    0.003392] smp: Bringing up secondary CPUs ...
[    0.004179] CPU1: update cpu_capacity 1024
[    0.004187] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.004932] CPU2: update cpu_capacity 1024
[    0.004939] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.005629] CPU3: update cpu_capacity 1024
[    0.005637] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.005721] smp: Brought up 1 node, 4 CPUs
[    0.005731] SMP: Total of 4 processors activated (153.60 BogoMIPS).
[    0.005736] CPU: All CPU(s) started in HYP mode.
[    0.005740] CPU: Virtualization extensions available.
[    0.006509] devtmpfs: initialized
[    0.017643] VFP support v0.3: implementor 41 architecture 3 part 40 variant 3 rev 4
[    0.017902] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.017919] futex hash table entries: 1024 (order: 4, 65536 bytes)
[    0.018468] pinctrl core: initialized pinctrl subsystem
[    0.019185] random: get_random_u32 called from bucket_table_alloc+0xfc/0x24c with crng_init=0
[    0.019297] NET: Registered protocol family 16
[    0.022022] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[... 153 lines cut ...]
[    2.378187] fuse init (API version 7.26)
[    2.380156] scsi host0: usb-storage 1-1.3:1.0
[    2.656689] systemd-udevd[129]: starting version 215
[    2.990589] gpiomem-bcm2835 3f200000.gpiomem: Initialised: Registers at 0x3f200000
[    3.272855] snd_bcm2835: module is from the staging directory, the quality is unknown, you have been warned.
[    3.276539] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
[    3.278106] bcm2835_alsa bcm2835_alsa: card created with 8 channels
[... cut ...]
[    4.111834] systemd-journald[108]: Received request to flush runtime journal from PID 1
[    4.969377] uart-pl011 3f201000.serial: no DMA platform data
[    5.439084] Adding 102396k swap on /var/swap.  Priority:-1 extents:7 across:14016508k SSFS
[    5.663741] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[    5.664096] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[    7.722798] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[    7.723252] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, half-duplex, lpa 0x0080
[    8.100919] random: crng init done

lategoodbye, 2017-07-24:
Looks like the mmc (not sdhost) driver isn't enabled.

JamesH65, 2017-07-24:
The instructions referenced are talking about a 3.x series kernel, and are 4 years old. Might be worth checking these instructions..

https://www.raspberrypi.org/documentation/linux/kernel/building.md

dakota72, 2017-07-24:
it's possible that I was wrong a .config?

popcornmix, 2017-07-24:
Always use the defconfig for that kernel tree - don't continue using an old config from an older (e.g. 4.9) kernel. i.e.
```
ARCH=arm make bcm2709_defconfig
```
Also make sure you 4.13 tree is up to date - a couple of days ago wifi wasn't working.

dakota72, 2017-07-25:
I updated 4.13 tree but the problem is the same... wlan0 don't exist.
which is the patch for wifi?

pelwell, 2017-07-25:
Your dmesg log is lacking any mention of mmc1, which is why WiFi isn't working. Build a vanilla RPi kernel following the instructions James linked to above (https://www.raspberrypi.org/documentation/linux/kernel/building.md) and prove that it works, at which point we know there is something wrong with your configuration.

dakota72, 2017-07-25:
I updated 4.13 tree and I followed the instrunctions of:
https://www.raspberrypi.org/documentation/linux/kernel/building.md
now this is my dmesg:
dmesg | grep mmc
[    0.000000] Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=1920 bcm27fbheight=1080 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=/dev/mmcblk0p2 root=ext4 elevator=deadline fsck.repair=yes rootwait quiet loglevel=3 consoleblank=0 .quirks=0x16c0:0x75e1:0x040
[    0.749616] mmc-bcm2835 3f300000.mmc: could not get clk, deferring probe
[    0.749941] sdhost-bcm2835 3f202000.mmc: could not get clk, deferring probe
[    0.772611] mmc-bcm2835 3f300000.mmc: mmc_debug:0 mmc_debug2:0
[    0.772622] mmc-bcm2835 3f300000.mmc: DMA channel allocated
[    0.867875] mmc1: queuing unknown CIS tuple 0x80 (2 bytes)
[    0.869450] mmc1: queuing unknown CIS tuple 0x80 (3 bytes)
[    0.871023] mmc1: queuing unknown CIS tuple 0x80 (3 bytes)
[    0.873822] mmc1: queuing unknown CIS tuple 0x80 (7 bytes)
[    0.911625] mmc0: sdhost-bcm2835 loaded - DMA enabled (>1)
[    0.912461] Waiting for root device /dev/mmcblk0p2...
[    1.028855] mmc0: host does not support reading read-only switch, assuming wrible
[    1.030762] mmc0: new high speed SDXC card at address 0001
[    1.031289] mmcblk0: mmc0:0001 00000 59.6 GiB
[    1.032877]  mmcblk0: p1 p2
[    1.052771] mmc1: new high speed SDIO card at address 0001
[    1.070577] EXT4-fs (mmcblk0p2): mounted filesystem with ordered data mode. Opull)
[    2.968921] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)

pelwell, 2017-07-25:
That's looking much better. What do you get from `dmesg | grep -E "(brcmfmac|wlan)"`?

dakota72, 2017-07-25:
dmesg | grep -E "(brcmfmac|wlan)"
nothing! :(

pelwell, 2017-07-25:
More typing:
```
lsmod
modprobe brcmfmac
```

dakota72, 2017-07-25:
lsmod
Module                  Size  Used by
rfcomm                 49152  0
bluetooth             364544  3 rfcomm
ecdh_generic           28672  1 bluetooth
cfg80211              569344  0
rfkill                 28672  2 bluetooth,cfg80211
sg                     28672  0
snd_bcm2835            32768  0
snd_pcm                98304  1 snd_bcm2835
snd_timer              32768  1 snd_pcm
snd                    69632  3 snd_timer,snd_bcm2835,snd_pcm
uio_pdrv_genirq        16384  0
uio                    20480  1 uio_pdrv_genirq
fixed                  16384  0
uinput                 20480  0
fuse                  106496  1
ecryptfs               98304  0
ipv6                  430080  34

---------------------------
sudo modprobe brcmfmac
---------------------------
lsmod
Module                  Size  Used by
brcmfmac              233472  0
brcmutil               16384  1 brcmfmac
rfcomm                 49152  0
bluetooth             364544  3 rfcomm
ecdh_generic           28672  1 bluetooth
cfg80211              569344  1 brcmfmac
rfkill                 28672  3 bluetooth,cfg80211
sg                     28672  0
snd_bcm2835            32768  0
snd_pcm                98304  1 snd_bcm2835
snd_timer              32768  1 snd_pcm
snd                    69632  3 snd_timer,snd_bcm2835,snd_pcm
uio_pdrv_genirq        16384  0
uio                    20480  1 uio_pdrv_genirq
fixed                  16384  0
uinput                 20480  0
fuse                  106496  1
ecryptfs               98304  0
ipv6                  430080  30

dakota72, 2017-07-25:
ifconfig
...
wlan0     Link encap:Ethernet  HWaddr b8:27:eb:5d:e2:43
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

work!!!! 
now, How do I make permanent changes?

pelwell, 2017-07-25:
Two more commands - the output may be lengthy (especially from the first command) so you might want to use pastebinit (`sudo apt-get install pastebinit` - use `cmd | pastebinit`):
```
grep brcmgrep -r . /sys/class/mmc_host/mmc1/mmc1:0001/
grep brcmfmac /lib/modules/`uname -r`/modules.alias
```

dakota72, 2017-07-25:
grep brcmgrep -r  /sys/class/mmc_host/mmc1/mmc1:0001
grep: /sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/power/autosuspend_delay_ms: Input/output error
grep: /sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0/phys_port_id: Operation not supported
grep: /sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0/power/autosuspend_delay_ms: Input/output error
grep: /sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0/duplex: Invalid argument
grep: /sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0/phys_switch_id: Operation not supported
grep: /sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0/speed: Invalid argument
grep: /sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0/name_assign_type: Invalid argument
grep: /sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0/phys_port_name: Operation not supported
grep: /sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/ieee80211/phy0/rfkill0/power/autosuspend_delay_ms: Input/output error
grep: /sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/ieee80211/phy0/power/autosuspend_delay_ms: Input/output error
grep: /sys/class/mmc_host/mmc1/mmc1:0001/power/autosuspend_delay_ms: Input/output error
grep: /sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:2/power/autosuspend_delay_ms: Input/output error

grep brcmfmac /lib/modules/`uname -r`/modules.alias
alias usb:v0A5Cp0BDCd*dc*dsc*dp*ic*isc*ip*in* brcmfmac
alias usb:v043Ep3101d*dc*dsc*dp*ic*isc*ip*in* brcmfmac
alias usb:v13B1p0039d*dc*dsc*dp*ic*isc*ip*in* brcmfmac
alias usb:v0A5CpBD27d*dc*dsc*dp*ic*isc*ip*in* brcmfmac
alias usb:v0A5CpBD1Fd*dc*dsc*dp*ic*isc*ip*in* brcmfmac
alias usb:v0A5CpBD17d*dc*dsc*dp*ic*isc*ip*in* brcmfmac
alias usb:v0A5CpBD1Ed*dc*dsc*dp*ic*isc*ip*in* brcmfmac
alias sdio:c*v02D0d4356* brcmfmac
alias sdio:c*v02D0d4354* brcmfmac
alias sdio:c*v02D0dA9BF* brcmfmac
alias sdio:c*v02D0d4345* brcmfmac
alias sdio:c*v02D0dA9A6* brcmfmac
alias sdio:c*v02D0d4339* brcmfmac
alias sdio:c*v02D0d4335* brcmfmac
alias sdio:c*v02D0dA962* brcmfmac
alias sdio:c*v02D0dA94D* brcmfmac
alias sdio:c*v02D0dA94C* brcmfmac
alias sdio:c*v02D0d4334* brcmfmac
alias sdio:c*v02D0d4330* brcmfmac
alias sdio:c*v02D0d4329* brcmfmac
alias sdio:c*v02D0d4324* brcmfmac
alias sdio:c*v02D0dA887* brcmfmac

pelwell, 2017-07-25:
Sorry, I mangled the first command. It's fixed now - can you try again?

dakota72, 2017-07-25:
the command:
grep brcmgrep -r . /sys/class/mmc_host/mmc1/mmc1:0001/
don't work, he is waiting a standard input

pelwell, 2017-07-25:
It's supped to be:
```
grep -r . /sys/class/mmc_host/mmc1/mmc1:0001/
```

dakota72, 2017-07-25:
ah ok sorry!
grep -r . /sys/class/mmc_host/mmc1/mmc1:0001/
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/vendor:0x02d0
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/device:0xa9a6
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/power/runtime_suspended_time:0
grep: /sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/power/autosuspend_delay_ms: Input/output error
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/power/runtime_active_time:0
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/power/control:auto
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/power/runtime_status:unsupported
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/class:0x00
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0/netdev_group:0
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0/addr_len:6
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0/flags:0x1003
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0/statistics/rx_nohandler:0
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0/statistics/tx_fifo_errors:0
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0/statistics/rx_frame_errors:0
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0/statistics/rx_missed_errors:0
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0/statistics/collisions:0
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0/statistics/tx_aborted_errors:0
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0/statistics/tx_dropped:0
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0/statistics/tx_carrier_errors:0
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0/statistics/rx_crc_errors:0
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0/statistics/tx_errors:0
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0/statistics/tx_packets:93
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0/statistics/rx_compressed:0
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0/statistics/rx_fifo_errors:0
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0/statistics/tx_bytes:14946
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0/statistics/rx_over_errors:0
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0/statistics/rx_length_errors:0
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0/statistics/rx_dropped:0
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0/statistics/rx_errors:0
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0/statistics/multicast:0
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0/statistics/tx_window_errors:0
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0/statistics/rx_packets:16
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0/statistics/tx_heartbeat_errors:0
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0/statistics/rx_bytes:2223
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0/statistics/tx_compressed:0
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0/gro_flush_timeout:0
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0/mtu:1500
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0/dormant:0
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0/link_mode:1
grep: /sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0/phys_port_id: Operation not supported
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0/power/runtime_suspended_time:0
grep: /sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0/power/autosuspend_delay_ms: Input/output error
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0/power/runtime_active_time:0
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0/power/control:auto
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0/power/runtime_status:unsupported
grep: /sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0/duplex: Invalid argument
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0/dev_port:0
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0/queues/tx-0/traffic_class:0
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0/queues/tx-0/byte_queue_limits/limit_min:0
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0/queues/tx-0/byte_queue_limits/inflight:0
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0/queues/tx-0/byte_queue_limits/limit:0
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0/queues/tx-0/byte_queue_limits/limit_max:1879048192
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0/queues/tx-0/byte_queue_limits/hold_time:1000
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0/queues/tx-0/tx_timeout:0
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0/queues/tx-0/xps_cpus:0
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0/queues/tx-0/tx_maxrate:0
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0/queues/rx-0/rps_cpus:0
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0/queues/rx-0/rps_flow_cnt:0
grep: /sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0/phys_switch_id: Operation not supported
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0/proto_down:0
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0/operstate:up
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0/broadcast:ff:ff:ff:ff:ff:ff
grep: /sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0/speed: Invalid argument
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0/addr_assign_type:0
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0/type:1
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0/tx_queue_len:1000
[... cut ...]
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:2/modalias:sdio:c00v02D0dA9A6
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:2/uevent:DRIVER=brcmfmac
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:2/uevent:SDIO_CLASS=00
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:2/uevent:SDIO_ID=02D0:A9A6
/sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:2/uevent:MODALIAS=sdio:c00v02D0dA9A6
/sys/class/mmc_host/mmc1/mmc1:0001/uevent:MMC_TYPE=SDIO
/sys/class/mmc_host/mmc1/mmc1:0001/uevent:MMC_NAME=
/sys/class/mmc_host/mmc1/mmc1:0001/uevent:MODALIAS=mmc:block

pelwell, 2017-07-25:
That all looks OK, which is odd. The grep in modules.alias shows that the brcmfmac module is registered, we already know that it is loadable and functional, and the MODALIAS line in the other output matches a value in the module so the window where things could be failing is now very small.

When the SDIO device is detected (which we know it is), the kernel generates an event that is intercepted by the userspace "udev" process. udev uses a large set of rules to determine how to handle each event, which in this case should result in an alias lookup and the brcmfmac module being loaded.

I may have overlooked something, but for now I'm going to say that threethings could be going wrong:
1) The kernel isn't emitting the right events.
2) The udev rule is missing or broken.
3) The brcmfmac module is blacklisted, which will prevent any automatic loading, requiring a manual modprobe to load it.

We can check for case 3 with:
```
grep brcmfmac /etc/modprobe.d/*
```
The others are harder.

dakota72, 2017-07-25:
grep brcmfmac /etc/modprobe.d/*
/etc/modprobe.d/brcmfmac.conf:options brcmfmac debug=0x100000
/etc/modprobe.d/raspi-blacklist.conf:blacklist brcmfmac
yess brcmfmac is blasklisted...but why?

pelwell, 2017-07-25:
I'm guessing your followed some tutorial, or perhaps you wanted to disable WiFi for an experiment. Well, it worked!
```
sudo rm /etc/modprobe.d/raspi-blacklist.conf
sudo reboot
```

dakota72, 2017-07-26:
It's worked. thanks so much for your support! :)

closed 2017-07-26 by dakota72