raspberrypi/linux #2133: 64 bits device tree for CM3?

Minipada, 2017-07-27:
I have my rpi3b running in 64 bits mode without problem, using devicetree files `bcm2837-rpi-3-b.dts` and `bcm2710-rpi-3-b.dtb` located in arm64/broadcom folder.

I am now trying to use the CM3 in 64 bits mode. I use those

    bcm2708-rpi-0-w.dtb 
    bcm2708-rpi-b.dtb 
    bcm2708-rpi-b-plus.dtb 
    bcm2709-rpi-2-b.dtb 
    bcm2710-rpi-3-b.dtb 
    bcm2708-rpi-cm.dtb 
    bcm2710-rpi-cm3.dtb

I am having the kernel booting, can read/write by serial until  and read the log by serial but I am losing the console as soon as the system boots. I have the assumption I may need one more dts, linked to the bcm2837, like the one needed for the rpi3b 64bits, but I don't understand why it is already needed in the first place.

JamesH65, 2017-09-13:
We don't specifically support 64 bit OS's, but maybe @pelwell the "DT Master" can help.

Minipada, 2017-09-13:
Hello James, 
Thanks for answering, I started working on it and started a PR. The work is not done since I'm mostly learning about DT and the rpi itself

JamesH65, 2018-06-27:
This issue will be closed within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested.

SteveBetter, 2018-06-29:
I am using the mainline kernel 4.14.40 aarch64 with preempt-rt. I can get only one dtb file bcm2837-rpi-3-b.dtb. When I was using "device_tree=bcm2837-rpi-3-b.dtb" in the config.txt, the boot process just stuck with none messages. When I delete "device_tree=...", booting is successful, and I can log in via UART, but no HDMI, USB, NETWORK.

Can anyone help? Thanks.

lategoodbye, 2018-07-21:
@SteveBetter There is no support for Compute Module 3 in Mainline yet. I'm currently working on that but there is no patch yet. You can try 4.17/4.18 plus Martin Sperl's [patch](http://lists.infradead.org/pipermail/linux-rpi-kernel/2017-November/007166.html).

SteveBetter, 2018-08-09:
@lategoodbye Thanks. I am sorry that I didn't make it clear my board is Raspberry Pi 3B. I downloaded the .config file for arm64 from this repository, but it just cannot boot up.