raspberrypi/linux #2136: CANNOT get CPU temperature in sysfs on rpi-4.12.y ARM64

UMRnInside, 2017-07-29:
One day I tried to get CPU temperature via _sysfs_:
```
pi@raspberry:~ $ cat /sys/class/thermal/thermal_zone0/temp
cat: /sys/class/thermal/thermal_zone0/temp: Invalid argument
```

However,I can get **correct** temperature via **VCHIQ**:
```
pi@raspberry:~ $ vcgencmd measure_temp
temp=42.9'C
```

dmesg:
```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.12.3-v8+ (umaru@mUMR) (gcc version 6.3.0 20170516 (Debian 6.3.0-18) ) #3 SMP PREEMPT Fri Jul 28 20:47:30 CST 2017
[    0.000000] Boot CPU: AArch64 Processor [410fd034]
[    0.000000] Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] efi: Getting EFI parameters from FDT:
[    0.000000] efi: UEFI not found.
[    0.000000] cma: Reserved 8 MiB at 0x0000000036800000
[    0.000000] On node 0 totalpages: 225280
[    0.000000]   DMA zone: 3520 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 225280 pages, LIFO batch:31
[    0.000000] percpu: Embedded 25 pages/cpu @ffffffeeb6771000 s63768 r8192 d30440 u102400
[    0.000000] pcpu-alloc: s63768 r8192 d30440 u102400 alloc=25*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: enabling workaround for ARM erratum 845719
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 221760
[    0.000000] Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=640 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000  VERBOSE dwc_otg.lpm_enable=0 console=ttyS0,115200 root=/dev/mmcblk0p2 rootfstype=f2fs elevator=deadline fsck.repair=yes rootwait
[    0.000000] PID hash table entries: 4096 (order: 3, 32768 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes)
[    0.000000] Memory: 860848K/901120K available (6908K kernel code, 846K rwdata, 2296K rodata, 2688K init, 696K bss, 32080K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     modules : 0xffffff8000000000 - 0xffffff8008000000   (   128 MB)
[    0.000000]     vmalloc : 0xffffff8008000000 - 0xffffffbebfff0000   (   250 GB)
[    0.000000]       .text : 0xffffff866c880000 - 0xffffff866cf40000   (  6912 KB)
[    0.000000]     .rodata : 0xffffff866cf40000 - 0xffffff866d190000   (  2368 KB)
[    0.000000]       .init : 0xffffff866d190000 - 0xffffff866d430000   (  2688 KB)
[    0.000000]       .data : 0xffffff866d430000 - 0xffffff866d503a00   (   847 KB)
[    0.000000]        .bss : 0xffffff866d503a00 - 0xffffff866d5b1bbc   (   697 KB)
[    0.000000]     fixed   : 0xffffffbefe7fd000 - 0xffffffbefec00000   (  4108 KB)
[    0.000000]     PCI I/O : 0xffffffbefee00000 - 0xffffffbeffe00000   (    16 MB)
[    0.000000]     vmemmap : 0xffffffbf00000000 - 0xffffffc000000000   (     4 GB maximum)
[    0.000000]               0xffffffbfba000000 - 0xffffffbfbadc0000   (    13 MB actual)
[    0.000000]     memory  : 0xffffffee80000000 - 0xffffffeeb7000000   (   880 MB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 24988 entries in 98 pages
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000] NR_IRQS:64 nr_irqs:64 0
[    0.000000] Failed to get local register map. FIQ is disabled for cpus > 1
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000006] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
...
[    0.732223] dwc_otg: FIQ enabled
[    0.732234] dwc_otg: NAK holdoff enabled
[    0.732245] dwc_otg: FIQ split-transaction FSM enabled
[    0.732260] Module dwc_common_port init
[    0.732812] usbcore: registered new interface driver usb-storage
[    0.733094] bcm2835_thermal 3f212000.thermal: Missing firmware node
[    0.733128] bcm2835_thermal: probe of 3f212000.thermal failed with error -2
[    0.733759] bcm2835-wdt 3f100000.watchdog: Broadcom BCM2835 watchdog timer
[    0.734088] bcm2835-cpufreq: min=600000 max=1200000
```

I have no idea how _bcm2835_thermal_ failed to probe 3f212000.thermal.

* I built _rpi-4.11.y_ and _rpi-4.12.y_ both using `make bcmrpi3_defconfig`.
On _rpi-4.11.y_ `/sys/class/thermal/thermal_zone0/temp` **works properly** but VCHIQ is **broken**.

UMRnInside, 2017-07-29:
A **possible** solution:
Use _bcm2710-rpi-3-b.dtb_ from branch **rpi.4.11.y** on kernel **rpi-4.12.y**
Since there's a 
>Missing firmware node

kukabu, 2017-07-29:
try it
CONFIG_THERMAL_BCM2835=n
CONFIG_BCM2835_THERMAL=y
rpi-4.12.y uses upstream thermal driver.

lategoodbye, 2017-07-29:
As explanation to @kukabu suggestion look at this [thread](http://lists.infradead.org/pipermail/linux-rpi-kernel/2017-July/006632.html).

kukabu, 2017-07-30:
@lategoodbye it's another issue.

lategoodbye, 2017-07-30:
Sorry, you're right.

aswild, 2017-08-30:
The upstream driver works in the current rpi-4.12.y branch, but CONFIG_BCM2835_THERMAL can only be set with CONFIG_ARCH_BCM, which is enabled for 32-bit but not 64-bit.

The Kconfig patch below allows using CONFIG_BCM2835_THERMAL when ARCH=arm64, which works for me on the pi3.

```diff
diff --git a/drivers/thermal/Kconfig b/drivers/thermal/Kconfig
index 257e2d59d1b0..61353fdcc995 100644
--- a/drivers/thermal/Kconfig
+++ b/drivers/thermal/Kconfig
@@ -415,7 +415,7 @@ config MTK_THERMAL
 	  controller present in Mediatek SoCs
 
 menu "Broadcom thermal drivers"
-depends on ARCH_BCM || COMPILE_TEST
+depends on ARCH_BCM || ARCH_BCM2835 || COMPILE_TEST
 source "drivers/thermal/broadcom/Kconfig"
 endmenu
 
```

lategoodbye, 2017-08-30:
@aswild Thanks this fixes commit 6892cf07e7337776c0b006db3b96a96bf071acb7 . 

Could you please send this to upstream?

aswild, 2017-08-30:
Sure, I'll get a patch sent out.

For the downstream linux-rpi kernel, I think we should also change bcmrpi3_defconfig to use the (working) upstream driver. Presumably both CONFIG_THERMAL_BCM2835 and CONFIG_BCM2835_THERMAL shouldn't be set at the same time, but it seems to work when that's the case.

I need to double-check the configs for 32-bit mode and update bcm2709_defconfig if needed for the rpi-4.12.y branch.

JamesH65, 2017-09-13:
@aswild Did that patch make it to upstream? Can this issue be closed?

lategoodbye, 2017-09-13:
There wasn't any feedback from the thermal maintainers. So it's likely that we need to resend after merge window has closed.

lategoodbye, 2017-09-27:
I resend the patch and hope it gets merged soon:

https://patchwork.kernel.org/patch/9972399/

lategoodbye, 2017-10-17:
Eduardo applied the patch today in his repo, so it should be in linux-next tomorrow.

JamesH65, 2017-12-04:
@lategoodbye Can this be closed?

lategoodbye, 2017-12-04:
@pelwell I don't know how long it takes until the upstream patch (currently in 4.15rc1) goes into 4.14.y.
Should i send a backport of this patch for rpi-4.14?

pelwell, 2017-12-04:
I've just cherry-picked it.

closed 2017-12-04 by pelwell