raspberrypi/linux #2139: USB Webcam not enumerating after firmware update

roerch, 2017-07-31:
As mentioned in #1709 there seems to be a regression between commits 528b516135111e0184336d3c290001c2681a675a  and a86bfee5b47a74c13056997f1e4d8b9d8090b398.

Problem:
I have (besides other things) a webcam connected to my Raspberry via a externally powered hub. The hub itself is Single TT, but today I verified that the same issue happens with a MultiTT hub. After booting (if at all) the webcam can be used once, after which it is being reset by dwc_otg and not enumerating anymore. With kernel 4.9.24 this was working fine, with kernel 4.9.25 and above, the above problem occurs.
Another thing maybe worth to mention: The webcam works with 4.9.25 and above if I edit the cmdline.txt to set dwc_otg.fiq_fsm_enable to 0 (NOP FIQ) - however this then breaks an USB 1.0 device by delivering garbled data and is no solution for me either. Let me know, if you need further input from my side.

Output of dmesg:
```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.9.25+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #994 Fri Apr 28 16:46:40 BST 2017
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] OF: fdt:Machine model: Raspberry Pi Model B Rev 2
[    0.000000] cma: Reserved 8 MiB at 0x1d400000
[    0.000000] Memory policy: Data cache writeback
[    0.000000] On node 0 totalpages: 122880
[    0.000000] free_area_init_node: node 0, pgdat c0914814, node_mem_map dcfc8000
[    0.000000]   Normal zone: 1080 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 122880 pages, LIFO batch:31
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 121800
[    0.000000] Kernel command line: bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  smsc95xx.turbo_mode=N dwc_otg.lpm_enable=0 console=ttyAMA0,115200 kgdboc=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait
[    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 468428K/491520K available (5943K kernel code, 489K rwdata, 1948K rodata, 404K init, 734K bss, 14900K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xde800000 - 0xff800000   ( 528 MB)
    lowmem  : 0xc0000000 - 0xde000000   ( 480 MB)
    modules : 0xbf000000 - 0xc0000000   (  16 MB)
      .text : 0xc0008000 - 0xc05d5e1c   (5944 kB)
      .init : 0xc083f000 - 0xc08a4000   ( 404 kB)
      .data : 0xc08a4000 - 0xc091e6c8   ( 490 kB)
       .bss : 0xc091e6c8 - 0xc09d5ec8   ( 734 kB)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000029] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[    0.000072] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[    0.000172] bcm2835: system timer (irq = 27)
[    0.000636] Console: colour dummy device 80x30
[    0.001410] console [tty1] enabled
[    0.001466] Calibrating delay loop... 697.95 BogoMIPS (lpj=3489792)
[    0.060343] pid_max: default: 32768 minimum: 301
[    0.060801] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.060871] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.061973] Disabling cpuset control group subsystem
[    0.062248] CPU: Testing write buffer coherency: ok
[    0.062359] ftrace: allocating 21709 entries in 64 pages
[    0.177741] Setting up static identity map for 0x8200 - 0x8238
[    0.179797] devtmpfs: initialized
[    0.188630] VFP support v0.3: implementor 41 architecture 1 part 20 variant b rev 5
[    0.189142] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.189242] futex hash table entries: 256 (order: -1, 3072 bytes)
[    0.190558] pinctrl core: initialized pinctrl subsystem
[    0.192070] NET: Registered protocol family 16
[    0.194542] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.204512] hw-breakpoint: found 6 breakpoint and 1 watchpoint registers.
[    0.204607] hw-breakpoint: maximum watchpoint size is 4 bytes.
[    0.204751] Serial: AMBA PL011 UART driver
[    0.207659] bcm2835-mbox 2000b880.mailbox: mailbox enabled
[    0.254324] bcm2835-dma 20007000.dma: DMA legacy API manager at de80d000, dmachans=0x1
[    0.257022] SCSI subsystem initialized
[    0.257354] usbcore: registered new interface driver usbfs
[    0.257529] usbcore: registered new interface driver hub
[    0.257862] usbcore: registered new device driver usb
[    0.267813] raspberrypi-firmware soc:firmware: Attached to firmware from 2017-04-27 17:19
[... 1065 lines cut ...]
          Usage Type               Data
        wMaxPacketSize     0x0044  1x 68 bytes
        bInterval               4
        bRefresh                0
        bSynchAddress           0
        AudioControl Endpoint Descriptor:
          bLength                 7
          bDescriptorType        37
          bDescriptorSubtype      1 (EP_GENERAL)
          bmAttributes         0x01
            Sampling Frequency
          bLockDelayUnits         0 Undefined
          wLockDelay              0 Undefined
Device Qualifier (for other device speed):
  bLength                10
  bDescriptorType         6
  bcdUSB               2.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 ?
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  bNumConfigurations      1
Device Status:     0x0000
  (Bus Powered)
```

P33M, 2017-08-01:
Your webcam is a high-speed device, so should be unaffected by that change. What's even more odd is that swapping the FIQ causes a change in the cam's behaviour.

Can you try with  `dwc_otg.fiq_fsm_mask=0x3` and remove the `dwc_otg.fiq_fsm_enable=0` parameter?

roerch, 2017-08-01:
This is an interesting hint. With `dwc_otg.fiq_fsm_mask=0x3` the webcam seems to operate normally again, even on the newest kernel. It didn't initially enumerate during bootup, but after unplugging and plugging it, it immediatly worked and during use, it was reset once, but not as previously until it was removed. However dmesg shows after short runtime a worrying (?) message at the very end.

I will leave this setup running for some time, to see if it has implications on any device.

dmesg here:
```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.9.40+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #1022 Sun Jul 30 11:11:04 BST 2017
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] OF: fdt:Machine model: Raspberry Pi Model B Rev 2
[    0.000000] cma: Reserved 8 MiB at 0x1d400000
[    0.000000] Memory policy: Data cache writeback
[    0.000000] On node 0 totalpages: 122880
[    0.000000] free_area_init_node: node 0, pgdat c09168d4, node_mem_map dcfc8000
[    0.000000]   Normal zone: 1080 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 122880 pages, LIFO batch:31
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 121800
[    0.000000] Kernel command line: bcm2708_fb.fbwidth=1824 bcm2708_fb.fbheight=984 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  smsc95xx.turbo_mode=N dwc_otg.lpm_enable=0 dwc_otg.fiq_fsm_mask=0x3 console=ttyAMA0,115200 kgdboc=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait
[    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 468416K/491520K available (5949K kernel code, 489K rwdata, 1952K rodata, 400K init, 734K bss, 14912K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xde800000 - 0xff800000   ( 528 MB)
    lowmem  : 0xc0000000 - 0xde000000   ( 480 MB)
    modules : 0xbf000000 - 0xc0000000   (  16 MB)
      .text : 0xc0008000 - 0xc05d771c   (5950 kB)
      .init : 0xc0842000 - 0xc08a6000   ( 400 kB)
      .data : 0xc08a6000 - 0xc0920788   ( 490 kB)
       .bss : 0xc0920788 - 0xc09d7f88   ( 734 kB)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000030] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[    0.000074] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[    0.000171] bcm2835: system timer (irq = 27)
[    0.000643] Console: colour dummy device 80x30
[    0.001414] console [tty1] enabled
[    0.001469] Calibrating delay loop... 697.95 BogoMIPS (lpj=3489792)
[    0.060346] pid_max: default: 32768 minimum: 301
[    0.060812] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.060880] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.061990] Disabling cpuset control group subsystem
[    0.062267] CPU: Testing write buffer coherency: ok
[    0.062373] ftrace: allocating 21716 entries in 64 pages
[    0.178766] Setting up static identity map for 0x8200 - 0x8238
[    0.180847] devtmpfs: initialized
[    0.189603] VFP support v0.3: implementor 41 architecture 1 part 20 variant b rev 5
[    0.190132] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.190230] futex hash table entries: 256 (order: -1, 3072 bytes)
[    0.191551] pinctrl core: initialized pinctrl subsystem
[    0.193054] NET: Registered protocol family 16
[    0.195556] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.205161] hw-breakpoint: found 6 breakpoint and 1 watchpoint registers.
[    0.205254] hw-breakpoint: maximum watchpoint size is 4 bytes.
[    0.205397] Serial: AMBA PL011 UART driver
[    0.208205] bcm2835-mbox 2000b880.mailbox: mailbox enabled
[    0.256036] bcm2835-dma 20007000.dma: DMA legacy API manager at de80d000, dmachans=0x1
[    0.258884] SCSI subsystem initialized
[    0.259222] usbcore: registered new interface driver usbfs
[    0.259390] usbcore: registered new interface driver hub
[    0.259596] usbcore: registered new device driver usb
[    0.268836] raspberrypi-firmware soc:firmware: Attached to firmware from 2017-07-28 20:15
[    0.270839] clocksource: Switched to clocksource timer
[    0.326679] VFS: Disk quotas dquot_6.6.0
[... 197 lines cut ...]
[   80.115324] uvcvideo 1-1.3.2:1.0: Entity type for entity Extension 10 was not initialized!
[   80.115344] uvcvideo 1-1.3.2:1.0: Entity type for entity Extension 12 was not initialized!
[   80.115364] uvcvideo 1-1.3.2:1.0: Entity type for entity Extension 8 was not initialized!
[   80.115380] uvcvideo 1-1.3.2:1.0: Entity type for entity Extension 11 was not initialized!
[   80.115396] uvcvideo 1-1.3.2:1.0: Entity type for entity Processing 2 was not initialized!
[... cut ...]
Bus 001 Device 012: ID 046d:09a1 Logitech, Inc. QuickCam Communicate MP/S5500
Bus 001 Device 005: ID 05e3:0608 Genesys Logic, Inc. USB-2.0 4-Port HUB
Bus 001 Device 004: ID 1941:8021 Dream Link WH1080 Weather Station / USB Missile Launcher
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9512 Standard Microsystems Corp. LAN9500 Ethernet 10/100 Adapter / SMSC9512/9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

P33M, 2017-08-02:
What program are you using to read from the webcam? As it's a standard UVC cam I may be able to reproduce the condition.

roerch, 2017-08-02:
I use the standard `uvccapture` to capture a single frame.
The exact command I use is `uvccapture -S90 -G60 -B150 -C60 -x960 -y720 -m -o/tmp/snap.jpg`.
This command is embedded into a cron job running every 10 minutes. But it either works all the time, or not at all.

JamesH65, 2018-06-27:
@roerch I knows it's been a while, but it would be worth updated to the latest kernel to see if anything gets fixed. If this is no longer an issue, please close the report.

This issue will be closed within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested.

roerch, 2018-06-28:
Ok, I updated now to the most recent 4.14 kernel via rpi-update (commit: 963a31330613a38e160dc98b708c662dd32631d5).
Spontaneously, all seems fine, but I will monitor this for some days and if it doesn't occur again, close the issue.

roerch, 2018-07-04:
After nearly a week of running the current kernel, I can confirm that my issue is not occurring anymore. Thanks for solving this.

closed 2018-07-04 by roerch