raspberrypi/linux #2140: Pi 3 B drops data on USB CDC with kernel 4.9.35-v7+

jsiverskog, 2017-08-01:
Possibly related to https://github.com/raspberrypi/linux/issues/2134, however, that issue clearly states that my hardware+kernel combination works.

Basically I have a custom board based on a PSoC 5LP that acts as an USB device implementing CDC and HID. Everything works as expected on my x86-64 running Linux 4.4.0. When I hook that board up to a Raspberry Pi 3 B, running 4.9.35-v7+, CDC data intermittently gets lost.

My test case is set up so that the Raspberry repeatedly sends 64 bytes of data (+ 7 bytes header) at a time, and then waits for an ack from the PSoC device. After some iterations (usually in the tens) some data gets lost - the PSoC doesn't receive a full packet.

I have hooked up a Totalphase Beagle USB 480 sniffer to see what gets sent between the units, and it confirms that data seems to be lost/incorrect (data dump exported and attached as [beagle_dump.txt](https://github.com/raspberrypi/linux/files/1190372/beagle_dump.txt)). Looking at line 330, where data is incorrect, there is a SPLIT_NAK instead of SPLIT as the data preceding it. Not sure exactly what that means though.

I tried to do an usbmon dump via tshark (on the Pi). As far as I can see (with my limited USB knowledge), the data looks correct. Dump attached as [tshark_dump.zip](https://github.com/raspberrypi/linux/files/1190386/tshark_dump.zip).

lsusb -v:
```
Bus 001 Device 026: ID 2367:babe
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            2 Communications
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0         8
  idVendor           0x2367
  idProduct          0xbabe
  bcdDevice            0.01
  iManufacturer           1 XXX
  iProduct                2 i/o unit
  iSerial               128 350202171B0B2164
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           92
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          1 XXX
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         2 Communications
      bInterfaceSubClass      2 Abstract (modem)
      bInterfaceProtocol      1 AT-commands (v.25ter)
      iInterface              3 CDC Communication Interface
      CDC Header:
        bcdCDC               1.10
      CDC ACM:
        bmCapabilities       0x02
          line coding and serial state
      CDC Union:
        bMasterInterface        0
        bSlaveInterface         1
      CDC Call Management:
        bmCapabilities       0x00
        bDataInterface          1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              10
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass        10 CDC Data
      bInterfaceSubClass      0 Unused
      bInterfaceProtocol      0
      iInterface              4 CDC Data Interface
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x03  EP 3 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      0 None
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      63
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              10
Device Status:     0x0000
  (Bus Powered)
```
I have tried to remove the HID class, as well as changing some parameters on the CDC class. Nothing I have tried so far has helped.

I have not tried other hardware (Pi:s) or kernels. What's the best way forward?

P33M, 2017-08-01:
It's more likely to be related to https://github.com/raspberrypi/linux/issues/1709 as a specific issue concerning the same type of transfer was reported (and fixed) there.

Are you plugging the device directly into the hub on the Pi?

Please also post a full dmesg since boot (with the device plugged in).

jsiverskog, 2017-08-01:
I tried with dwc_otg.nak_holdoff=0 without any improvement.

The device has a built-in hub on its board that I am unable to bypass. So it's Pi-(built-in hub on Pi)-(built-in hub on PSoC board)-PSoC.

dmesg since boot:
```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.9.35-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #1014 SMP Fri Jun 30 14:47:43 BST 2017
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt:Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] cma: Reserved 8 MiB at 0x3a800000
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 241664
[    0.000000] free_area_init_node: node 0, pgdat 80c6ee40, node_mem_map b9faa000
[    0.000000]   Normal zone: 2124 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 241664 pages, LIFO batch:31
[    0.000000] percpu: Embedded 14 pages/cpu @b9f64000 s25600 r8192 d23552 u57344
[    0.000000] pcpu-alloc: s25600 r8192 d23552 u57344 alloc=14*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 239540
[    0.000000] Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000  dwc_otg.nak_holdoff=0 dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=8b66d917-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 936296K/966656K available (7168K kernel code, 484K rwdata, 2012K rodata, 1024K init, 778K bss, 22168K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xbb800000 - 0xff800000   (1088 MB)
    lowmem  : 0x80000000 - 0xbb000000   ( 944 MB)
    modules : 0x7f000000 - 0x80000000   (  16 MB)
      .text : 0x80008000 - 0x80800000   (8160 kB)
      .init : 0x80b00000 - 0x80c00000   (1024 kB)
      .data : 0x80c00000 - 0x80c79014   ( 485 kB)
       .bss : 0x80c7b000 - 0x80d3da64   ( 779 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] 	Build-time adjustment of leaf fanout to 32.
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000000] arm_arch_timer: Architected cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000008] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000025] Switching to timer-based delay loop, resolution 52ns
[    0.000336] Console: colour dummy device 80x30
[    0.001245] console [tty1] enabled
[    0.001293] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.001362] pid_max: default: 32768 minimum: 301
[    0.001721] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001764] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.002795] Disabling cpuset control group subsystem
[    0.002985] CPU: Testing write buffer coherency: ok
[    0.003052] ftrace: allocating 22390 entries in 66 pages
[    0.055595] CPU0: update cpu_capacity 1024
[    0.055664] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.055761] Setting up static identity map for 0x100000 - 0x100034
[    0.057999] CPU1: update cpu_capacity 1024
[    0.058007] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.058725] CPU2: update cpu_capacity 1024
[    0.058732] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.059396] CPU3: update cpu_capacity 1024
[    0.059403] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.059486] Brought up 4 CPUs
[    0.059658] SMP: Total of 4 processors activated (153.60 BogoMIPS).
[    0.059687] CPU: All CPU(s) started in HYP mode.
[    0.059713] CPU: Virtualization extensions available.
[    0.060536] devtmpfs: initialized
[    0.071652] VFP support v0.3: implementor 41 architecture 3 part 40 variant 3 rev 4
[... 182 lines cut ...]
[    5.015114] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
[    5.251562] usbcore: registered new interface driver brcmfmac
[    5.399606] brcmfmac: Firmware version = wl0: May 27 2016 00:13:38 version 7.45.41.26 (r640327) FWID 01-df77e4a7
[    5.949445] systemd-journald[135]: Received request to flush runtime journal from PID 1
[    6.388845] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[    6.388870] brcmfmac: power management disabled
[    7.351397] uart-pl011 3f201000.serial: no DMA platform data
[    7.690585] Adding 102396k swap on /var/swap.  Priority:-1 extents:4 across:184320k SSFS
[    7.858675] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[    9.462716] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0xCDE1
[... cut ...]
[   11.810212] Bluetooth: SCO socket layer initialized
[   11.820663] Bluetooth: HCI UART driver ver 2.3
[   11.820674] Bluetooth: HCI UART protocol H4 registered
[   11.820679] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   11.820868] Bluetooth: HCI UART protocol Broadcom registered
[   12.071507] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   12.071518] Bluetooth: BNEP filters: protocol multicast
[   12.071536] Bluetooth: BNEP socket layer initialized
[   23.635296] random: crng init done
```

P33M, 2017-08-01:
Is there any way of analyzing the text dump? It appears that individual transfers are being aggregated with information being lost (i.e. what tokens were transmitted at what time). The "record" field is also not particularly illuminating.

As you're sending 71 bytes for an OUT transfer, the split transaction has to be completed in two stages. The first is a 64-byte packet and the second is a 7-byte transfer. This is why seeing individual tokens is critical - one failure mode could be trying to start the second transfer before completing the first, with the hub cancelling the first transfer.

jsiverskog, 2017-08-01:
Does this help? It's a screenshot of the program used for capturing ("Total Phase Data Center") with the last packets expanded (this is from another run). Of course I can share the entire dump if you want to install the program yourself. [image]

P33M, 2017-08-02:
Do you have a link for the analyzer software? If you upload the capture dump somewhere then it'll be easier for me to find out what's going wrong.

jsiverskog, 2017-08-03:
There are download links at https://www.totalphase.com/products/data-center/ . Attached is
[beagle_dump_rpi.tdc.zip](https://github.com/raspberrypi/linux/files/1196250/beagle_dump_rpi.tdc.zip).

To get a clear view of the issue:
1. Start "Data Center"
2. Open the file
3. In the navigator on the right-hand side, expand USB 2.0, right-click "i/o unit" and select "Filter: Show Only"

Thanks.

P33M, 2017-08-04:
Can you also post traces (about 20 seconds of traffic) for 
- Pi running a 4.4 kernel (there were several USB patches between 4.4 and latest 4.9)
- An x86-64 machine?

jsiverskog, 2017-08-07:
Sure. Is there an easy way to install the 4.4 kernel on this Raspbian system?

P33M, 2017-08-07:
Running `sudo rpi-update 52241088c1da59a359110d39c1875cda56496764` will get 4.4.50 which was the last shipped 4.4 kernel.

jsiverskog, 2017-08-07:
Thanks.

The same (or a similar) issue appears on 4.4.50: [beagle_dump_io_unit_rpi_4.4.50.tdc.zip](https://github.com/raspberrypi/linux/files/1204784/beagle_dump_io_unit_rpi_4.4.50.tdc.zip)
But on x86-64 I can run at least 4000 iterations: [beagle_dump_io_unit_x86-64.tdc.zip](https://github.com/raspberrypi/linux/files/1204785/beagle_dump_io_unit_x86-64.tdc.zip)

jsiverskog, 2017-08-11:
@P33M: Any idea? Thanks.

jsiverskog, 2017-08-14:
I can confirm the same issue appears on
- Pi 3 B running 4.9.41
- Pi Zero 1.3 running 4.4.50
- Pi Zero 1.3 running 4.9.41

P33M, 2017-08-14:
OK, that confirms that it's not a regression in the recent slew of fixes. I'll see if anything stands out from looking at the captures.

jsiverskog, 2017-08-14:
I am trying to setup a system with another PSoC on the other end - without the hub.

jsiverskog, 2017-08-14:
I can confirm that it works with a PSoC unit not connected through the hub.

I will get in touch with the hub manufacturer and see whether something's incorrectly set up on the PCB level.

jsiverskog, 2017-08-15:
When I connect the Total Phase Beagle between the RPI:s internal hub and our Cypress hub, I can see the SPLIT_NAK packet, but when I connect it between our internal hub and the PSoC device, the SPLIT_NAK packet is gone. I don't know USB well enough to analyze this, though. I am attaching the two different runs.
[01-beagle-between-cypress-hub-and-psoc.tdc.zip](https://github.com/raspberrypi/linux/files/1224620/01-beagle-between-cypress-hub-and-psoc.tdc.zip)
[02-beagle-between-rpi-hub-and-cypress-hub.tdc.zip](https://github.com/raspberrypi/linux/files/1224621/02-beagle-between-rpi-hub-and-cypress-hub.tdc.zip)

The PSoC is a full speed device, while our integrated Cypress hub is high speed. If I limit the Raspberry Pi to full speed (dwc_otg.speed=1) everything seems to work. However, that feels like a huge limitation.

jsiverskog, 2017-08-16:
@P33M: We've done some additional investigations. It doesn't seem like there is anything incorrect with the Cypress hub. To clarify things, what I'm doing is to repeatedly send
```
C0 EF 90 00 00 00 40 00 01 02 03 04 05 06 07 08
09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18
19 1A 1B 1C 1D 1E 1F 20 21 22 23 24 25 26 27 28
29 2A 2B 2C 2D 2E 2F 30 31 32 33 34 35 36 37 38
39 3A 3B 3C 3D 3E 3F
```
from the RPI to the PSoC device, which should ack with
```
C0 FE 91 00 00 00 00
```
When looking at the failing transaction, the RPI receives a NAK from the Cypress hub, and re-sends with incorrect data. Please look at this attached image:

[image]

It is taken from the `02-beagle-between-rpi-hub-and-cypress-hub.tdc.zip` run (above).

P33M, 2017-08-16:
Hmm. What happens if you try the test with `dwc_otg.fiq_fsm_mask=0x2` in /boot/cmdline.txt?

Reboot the Pi for the changes to take effect.

I've spotted a potential source of the bug, changing the mask value above will remove the code path from the FIQ that I think is causing the issue.

jsiverskog, 2017-08-16:
I have not been able to run the same exact test setup due to hardware shortage, but I have been able to transfer more data than I have before - so there definitely seems to be a difference!

I believe that you have figured the correct bug source.

P33M, 2017-08-16:
The state flow through the FIQ state machine in this case doesn't touch the hcdma register. I believe that re-enabling the channel after receiving the first NAK results in the host channel trundling along and transmitting from a data buffer offset of +64bytes, which matches the data transmitted.

You probably don't see this with other hub chips handling split transactions between the CDC device and the Pi because many hubs have deep buffers for split transactions to prevent this NAK response situation from occurring. In the NAK response case, 71 bytes of bus bandwidth has been wasted as the hub dropped the 64-byte DATA0 packet on the floor as it didn't have space for it.

I presume your test program is spinning in a tight enough loop that enough USB transfers are occurring resulting in the hub buffers being filled.

jsiverskog, 2017-08-16:
Ok. What side effects may `dwc_otg.fiq_fsm_mask=0x2` have?

P33M, 2017-08-16:
From drivers/usb/host/dwc_otg/dwc_otg_hcd.c:1043

```
			DWC_PRINTF("FIQ FSM acceleration enabled for :\n%s%s%s%s",
				(fiq_fsm_mask & 0x1) ? "Non-periodic Split Transactions\n" : "",
				(fiq_fsm_mask & 0x2) ? "Periodic Split Transactions\n" : "",
				(fiq_fsm_mask & 0x4) ? "High-Speed Isochronous Endpoints\n" : "",
				(fiq_fsm_mask & 0x8) ? "Interrupt/Control Split Transaction hack enabled\n" : "");
```
Non-periodic split transactions have a very high interrupt frequency (2-3 per transfer) if not performed by the FIQ (mask 0x1). Our default throttling when not using the FIQ mask restricts transfers to/from non-periodic split endpoints in the case where a transfer to/from an endpoint is NAK'd. This increases the endpoint servicing interval for non-periodic endpoints to a maximum of 1ms (depending on how the device behaves). This can be reduced by `dwc_otg.nak_holdoff=n` where n results in (125uS * n) maximum latency.

The interrupt/control hack (mask 0x8) treats certain types of periodic split transaction as Control-type transfers, which hubs will happily accept and perform normally. This avoids a lot of periodic transfer scheduling headache that causes transfers to busy full-speed bus segments to get dropped.

The HS Isochronous support (mask 0x4) is mainly of concern to webcams which have high data throughputs and short (1 microframe) service intervals, which cause interrupt storms if the FIQ isn't hand-holding the controller.

jsiverskog, 2017-08-29:
Ok. We will have it set to 2 for the time being.

Thank you.

JamesH65, 2017-09-13:
@P33M Is this still an issue? Are we going to providing a more robust fix?

P33M, 2017-09-13:
It's a bug, on the list of things to fix

popcornmix, 2017-11-24:
The 4.14.2 kernel on the next branch contains @P33M's possible fix. Can you test?
See: https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=197689 for how to update.

jsiverskog, 2017-11-29:
The behavior (on application level) is the same with 4.14.2-v7+. I have not sniffed the traffic to see whether it's exactly the same cause, but the behavior is very similar from the application's point of view.

Setting `dwc_otg.fiq_fsm_mask=0x2` still seems to help.

P33M, 2017-11-29:
Can you provide traces of the issue on the new kernel?

Reading the controller databook, the FIQ now should follow the channel handling flow outlined in pseudocode.

JamesH65, 2018-06-27:
@P33M Where have we got to on this - thread above gets a bit confusing. Seems a fix was merged, but it might not fix things fully? Do we need more information from users?

manio, 2018-10-07:
Just for the reference: Isn't it related with this?
https://github.com/torvalds/linux/commit/df3aa13c7bbb307e172c37f193f9a7aa058d4739
I found a regression in cdc_acm - it is now fixed (reverted).

jsiverskog, 2020-07-23:
Trying on 5.4.51 without anything added to /boot/cmdline.txt now, and it seems to work. I'm happy to close this issue if you are.

pelwell, 2020-07-23:
Thanks - it's nice to hear a positive outcome after the kernel version bump.

closed 2020-07-23 by pelwell

jsiverskog, 2023-08-29:
@P33M : i believe this is still an issue (now with another usb device).

when running through a usb hub i need the `dwc_otg.fiq_fsm_mask=0x2` workaround to not lose data. i've tried with `6.1.21-v8+ #1642 SMP PREEMPT Mon Apr 3 17:24:16 BST 2023` and `6.1.47-v8+ #1674 SMP PREEMPT Fri Aug 25 13:17:22 BST 2023`.

it works if connecting my device directly to the built-in rpi usb hub.