raspberrypi/linux #2142: Sysrq non working over serial line

piso77, 2017-08-02:
While debugging another issue (and trying to use the sysrq over serial) i found it to be not working:

raspberrypi login: 
Raspbian GNU/Linux 8 raspberrypi ttyS0

raspberrypi login: ~#���������

this is the output of a sysrq req command invoked via serial line: BREAK + s for example, and after that the serial console is dead.

I tried different serial terminal adapters (cu, screen and minicom) but the result is always garbage and a dead serial.

On the other hand, invoking sysrq via /sys over ssh works:

pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.9.35-v7+ #1014 SMP Fri Jun 30 14:47:43 BST 2017 armv7l GNU/Linux
pi@raspberrypi:~ $ sudo bash
root@raspberrypi:/home/pi# echo s > /proc/sysrq-trigger

...
[  335.987115] sysrq: SysRq : Emergency Sync
[  336.005982] Emergency Sync complete
...

/boot/config.txt:
...
enable_uart=1

This on a fresh raspbian image just reinstalled and upgraded. Any idea?

P33M, 2017-08-02:
Are you using a Pi 3?

kukabu, 2017-08-02:
I think it's because CONFIG_MAGIC_SYSRQ_SERIAL=n

piso77, 2017-08-02:
Yes, pi3 and there's no such a knob as MAGIC_SYSRQ_SERIAL in the entire Linux kernel

piso77, 2017-08-02:
Oh, and by shutting down the board after invoking it via serial, the entire shut down messages are all garbage:

raspberrypi login: ~#������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������

this led me to think that the serial driver / line is really messed up after the sysrq invokation.

P33M, 2017-08-02:
The mini UART doesn't support reception of BREAK line status (constant 0 on the UART RXD line). It will just spam NUL characters at you, filling up console buffers as it does so.

What happens if you swap to using the PL011 UART? In config.txt:
```
enable_uart=1
dtoverlay=pi3-disable-bt
```

pelwell, 2017-08-02:
Or dtovrr

piso77, 2017-08-02:
Yes, with the pi3-disable-bt it works - thanks

closed 2017-08-02 by piso77

pelwell, 2017-08-02:
You can also use `dtoverlay=pi3-miniuart-bt` if you want to use Bluetooth for keyboards and mice, etc.