raspberrypi/linux #2146: llsec_do_encrypt_unauth encrypts header

tewarid, 2017-08-03:
Function `llsec_do_encrypt_unauth` in [net/mac802154/llsec.c](https://github.com/raspberrypi/linux/blob/rpi-4.9.y/net/mac802154/llsec.c) encrypts the header and the payload. According to 802.15.4-2015 specification (section 9.2.1 Outgoing frame security procedure) just the outgoing payload is encrypted.

popcornmix, 2017-08-04:
This is a file from the upstream linux kernel that has not been edited in raspberry pi repo.
Can you try following this: https://www.kernel.org/pub/linux/docs/lkml/reporting-bugs.html

If the upstream linux kernel agrees there is a bug and provides a fix, it will filter through to our kernel (and we can cherry-pick it earlier to speed up the process).

But the maintainers of that file need to be the ones who approve any change - it is not something we will be doing without their agreement.

tewarid, 2017-08-04:
Understood. Thank you.

closed 2017-08-04 by tewarid