raspberrypi/linux #2151: Raspberry pi3 fails to boot 64bit kernel 4.12.3 with bcm2837 dts

VT-Github-Raspberrypi, 2017-08-07:
Raspberry pi3 fails to boot 64bit kernel 4.12.3 with bcm2837 dts

VT-Github-Raspberrypi, 2017-08-08:
Fix for raspberrypi#2151
https://github.com/VT-Github-Raspberrypi/linux/commit/05d903273846b420c7284af19ca0b4a28e86cb7f

VT-Github-Raspberrypi, 2017-08-08:
Follow these steps for RPI-3 64bit boot up.
https://github.com/raspberrypi/linux/wiki/config.txt

lategoodbye, 2017-08-12:
In short mixing a downstream defconfig and a upstream devicetree will usually cause problems.

Good combinations:
arm64/configs/bcmrpi3_defconfig +bcm2710-rpi-3-b.dts
arm64/configs/defconfig + bcm2837-rpi-3-b.dts

In the end you are trying to backport this [commit](https://github.com/raspberrypi/linux/commit/738f859f520862e51a5662b03fb162524e09307b), which is available since 4.13.

pelwell, 2017-08-13:
We are happy to back-port a commit from a newer kernel if it is proven to solve a problem.

@VT-Github-Raspberrypi On a top-of-tree rpi-4.12.y branch, run:
```
git cherry-pick 738f859f
```
Rebuild the DTBs and confirm that they work as expected.

JamesH65, 2017-09-13:
@VT-Github-Raspberrypi Did you have a chance to test this out?

closed 2017-12-04 by JamesH65