raspberrypi/linux #2161: ath10k (QCA9377) AC WiFi support?

mypiandrew, 2017-08-09:
Hi

We have a compute module based board and we've a customer interested in using a QCA9377-7AC USB WiFi card 

I can see the ath10k files are here :
https://github.com/raspberrypi/linux/tree/rpi-4.9.y/drivers/net/wireless/ath/ath10k

And there are some help files on how to the firmware required here:

https://github.com/erstrom/linux-ath/wiki/Firmware

I'm happy enough to work through the firmware part myself and generate the required files, but is it possible to get the ath10k kernel modules included in the build so I can work on this to prove out the card works?

I'll contribute any firmware files I've got working if that helps?

Andrew.

pelwell, 2017-08-09:
Have you tried building the kernel yourself? It's not that difficult - just follow our guide here: https://www.raspberrypi.org/documentation/linux/kernel/building.md
You will need to define at least CONFIG_ATH10K, so after the `make ... bcmrpi_defconfig` (for CM1 - bcm2709_defconfig for CM3) you should add `scripts/config -m ATH10K` before the main `make` command.

Once you have tried it and verified that it works for you just let us know and we'll add it to the standard defconfigs.

JamesH65, 2017-09-13:
@mypiandrew Any luck/reports with this?

mypiandrew, 2017-09-13:
Hi James

Currently working from the other end with Erik (https://github.com/erstrom)

He has had some success with getting the module working (as well as his other kit does)

I figure once that's all working we'll pick it up from this side

mypiandrew, 2017-10-30:
OK, quick update on this

For the moment we're parking the ATH10K support, Erik has got the module working but I think it's a bit far off in mainline kernel land to be playing with here for the moment. 

We have found an alternative believe we can solve the problem with connecting to 5GHz networks with the Redpine RS9113 module, this supports dual band WiFi networks (2.4GHz and 5GHz) as well as BLE and Zigbee (!) :

http://www.redpinesignals.com/Products/Modules/nLink_Family/RS9113_Wireless_Mini_PCI_Express_Card.php

This is a pretty nifty device as the module (which is also sold separately) can work in both USB and SDIO modes, so it's pretty flexible as far as host interfaces go.

I see the drivers for this redpine module are already in the 4.9 tree :

https://github.com/raspberrypi/linux/tree/204050d0eafb565b68abf512710036c10ef1bd23/drivers/net/wireless/rsi

Once we get our sample through I'll get the icepack out and see if I can decode the installation documentation. The gist I get it it'll either be very easy or really really hard to get it going.

If/when we get it working, does the offer still stand on getting the support modules included with the standard build?

Thanks for the help so far!

Andrew.

pelwell, 2017-10-30:
If it's just a case of adding a few "CONFIG_xxx=m" settings then yes, the offer still stands.

mypiandrew, 2017-10-30:
Great!

mypiandrew, 2017-11-13:
Just to update, turns out the Redpine rsi drivers in the tree are out of date so I'm being issued with new out-of-tree drivers for the moment. Redpine say they are planning to update the tree drivers but I don't have an eta on this

Rather than leave this as a zombie thread do you want me to close this and open a new one as and when any new kernel tree drivers are available for integration?

JamesH65, 2017-12-04:
@mypiandrew Closing to be dealt with as suggested.

closed 2017-12-04 by JamesH65

nckswt, 2018-09-19:
Resurrecting this zombie thread: Any chance the new drivers are available? I'm trying to figure out why I can see the `rsi` [wireless drivers on github](https://github.com/raspberrypi/linux/tree/rpi-4.9.y/drivers/net/wireless/rsi) but when I run `pi@raspberrypi:/lib/modules/4.9.43-v7+/kernel/drivers/net/wireless $ ls` all I see is

```
ath  atmel  broadcom  intersil  mac80211_hwsim.ko  marvell  mediatek  ralink  realtek  rndis_wlan.ko  zydas
```

(note the lack of `rsi`)

popcornmix, 2018-09-19:
We aren't currently enabling ``CONFIG_RSI_91X``. Are you saying you'd like this enabled?

nckswt, 2018-09-19:
Yes please!

mypiandrew, 2018-09-19:
What device are you trying to get going nick?

mypiandrew, 2018-09-19:
> It's an RS9113-NBZ-D1N.

That’s the mpcie usb card, right?

That’s not the same driver, there’s no in-tree solution currently 

Will check later, but I think we have a install script for it as we use a very similar sounding one

nckswt, 2018-09-19:
Nope! It's USB and SDIO, and we're using SDIO.

mypiandrew, 2018-09-19:
Ah
Best tack is to contact redpine support direct for that

nyetwurk, 2020-07-19:
Totally dead. Still zero USB wifi dongles that work with AC in linux and hostapd out of the box.