raspberrypi/linux #2162: USB audio 32bit interface sound sizzle with 4.9.40

arcangeli, 2017-08-10:
Hi,
I've got two USB audio interface. One Asus Xonar U3 (16bit 48000Hz S16_LE) and one Native Instrument Komplete Audio 6 (32bit 96000Hz S32_LE).
No problem so far with the Asus. In /proc/asound/CODEC/stream0, momentary_freq is always 48000Hz.
Sound sizzle with Komplete 6. In /proc/asound/K6/stream0, momentary_freq oscillate between 48000HZ and 48003HZ.
A simple mp3 file recorded directly (from qjackrcd with direct hydrogen output, not from the card physical output):
[http://ovh.to/grn5YBQ](url)
Apparently sound is better (not perfect) with 4.9.35 (from a fresh raspbian install - i'll try to compile it with my main raspbian install soon).

My alsa-info: [http://ovh.to/Nbetj6](url)
First /proc/asound/K6/stream0:
``Native Instruments Komplete Audio 6 at usb-3f980000.usb-1.2, high speed : USB Audio

Playback:
  Status: Running
    Interface = 1
    Altset = 1
    Packet Size = 192
    Momentary freq = 48001 Hz (0x6.0008)
    Feedback Format = 16.16
  Interface 1
    Altset 1
    Format: S32_LE
    Channels: 6
    Endpoint: 1 OUT (ASYNC)
    Rates: 44100, 48000, 88200, 96000
    Data packet interval: 125 us

Capture:
  Status: Stop
  Interface 2
    Altset 1
    Format: S32_LE
    Channels: 6
    Endpoint: 2 IN (ASYNC)
    Rates: 44100, 48000, 88200, 96000
    Data packet interval: 125 us``

Second one, juste one or two second later:
``Native Instruments Komplete Audio 6 at usb-3f980000.usb-1.2, high speed : USB Audio

Playback:
  Status: Running
    Interface = 1
    Altset = 1
    Packet Size = 192
    Momentary freq = 48002 Hz (0x6.0010)
    Feedback Format = 16.16
  Interface 1
    Altset 1
    Format: S32_LE
    Channels: 6
    Endpoint: 1 OUT (ASYNC)
    Rates: 44100, 48000, 88200, 96000
    Data packet interval: 125 us

Capture:
  Status: Stop
  Interface 2
    Altset 1
    Format: S32_LE
    Channels: 6
    Endpoint: 2 IN (ASYNC)
    Rates: 44100, 48000, 88200, 96000
    Data packet interval: 125 us
``

P33M, 2017-08-10:
When you say "sizzle", what does this mean?
Is the sound corrupted or still reasonably audible?
Is the nature of the corruption intermittent or constant?
If it is intermittently corrupted, what is the typical interval between events?

arcangeli, 2017-08-10:
Sorry, i'm  french. I think that "crackle" is better ;)
You can hear a simple hydrogen based kick here: [sound](http://ovh.to/TC4ZyG3).
When there's only one instrument, the crackle appear like if a LFO made it (appear, disappear, appear, ...). But if there are more instrument, the crackle appear always.
More sound = more crackle.
The "momentary freq" frequency from the stream0 file seam to be activated by a LFO (48000, 48001, 48002, 48003, 48002, 48001, 48000, ...). Same with other frequency like 44100 or 96000.
I've test with another 16bit 44100 Hz interface (Hercules Muse Pocket LT3) and it work perfectly, like the Asus Xonar U3.

arcangeli, 2017-08-11:
I can confirm that on 4.9.35 (freshly compiled), there's less crackle. It's usable. Not perfect. Not as good as my 16bit audio interface.
The "momentary freq" from the stream0 file oscillate with 3Hz maximum.

JamesH65, 2018-06-27:
Can people please try the latest kernel (at time of writing 4.14.y) and report back if there are still issues?

This issue will be closed within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested.

closed 2018-08-31 by JamesH65