raspberrypi/linux #2163: block2mtd support?

nwf, 2017-08-12:
Could we get `CONFIG_MTD_BLOCK2MTD` kicked to `m` by default in subsequent builds?  It would be very useful to enable `ubifs` filesystems on sdcards for the crc32 checksums and transparent compression, which would allow smaller sdcards to be useful.

I hope to add support for ubifs root to https://github.com/debian-pi/raspbian-ua-netinst; I recognize that it will require an initramfs to load the `block2mtd` and `ubifs` modules.

popcornmix, 2017-08-13:
This repo is correct for requesting kernel config options.
The option looks reasonable to me - @pelwell any concerns?

pelwell, 2017-08-13:
No concerns. It adds 400KB of modules, but that's not so much.

JamesH65, 2017-09-13:
@popcornmix Was this added?

nwf, 2018-05-19:
Ping?

pelwell, 2018-05-20:
It will be in the next firmware release.

closed 2018-05-20 by pelwell