raspberrypi/linux #2164: RTL8192CU not possible to change tx-power

reset12, 2017-08-14:
Hello,
i have installed Linux raspberrypi4 4.9.35-v7+ #1014 SMP Fri Jun 30 14:47:43 BST 2017 armv7l GNU/Linux
and will use the raspberry as a hotspot.
I am not be able to change tx-power or power-management parameter on wlan1(RTL8192CU).
iwconfig wlan1 txpower 10
Error for wireless request "Set Tx Power" (8B26) :
    SET failed on device wlan1 ; Operation not permitted.
or
iwconfig wlan1 power on
Error for wireless request "Set Power Management" (8B2C) :
    SET failed on device wlan1 ; Operation not permitted.

iwconfig
lo        no wireless extensions.

wlan0     IEEE 802.11  ESSID:"7050"  
          Mode:Managed  Frequency:2.462 GHz  Access Point: C4:6E:1F:3A:87:82   
          Bit Rate=14.4 Mb/s   Tx-Power=31 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:on
          Link Quality=38/70  Signal level=-72 dBm  
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:36  Invalid misc:0   Missed beacon:0

eth0      no wireless extensions.

wlan1     unassociated  Nickname:"<WIFI@REALTEK>"
          Mode:Auto  Frequency=2.412 GHz  Access Point: Not-Associated   
          Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality:0  Signal level:0  Noise level:0
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

Bus 001 Device 007: ID 046d:0991 Logitech, Inc. QuickCam Pro for Notebooks
Bus 001 Device 006: ID 0bda:8178 Realtek Semiconductor Corp. RTL8192CU 802.11n WLAN Adapter
Bus 001 Device 005: ID 046d:082b Logitech, Inc. 
Bus 001 Device 004: ID 04fa:2490 Dallas Semiconductor DS1490F 2-in-1 Fob, 1-Wire adapter
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

with regards
Markus

pelwell, 2017-08-14:
Have you tried `sudo iwconfig wlan1 power on`?

reset12, 2017-08-14:
yes.
I tried all commands from root user.

oot@raspberrypi4:/home/t759f# sudo iwconfig wlan1 power on
Error for wireless request "Set Power Management" (8B2C) :
    SET failed on device wlan1 ; Operation not permitted.

pelwell, 2017-08-14:
From what I can see the rtl8192cu driver does not handle the [SIOCSIWTXPOW ioctl](https://github.com/raspberrypi/linux/blob/rpi-4.9.y/drivers/net/wireless/realtek/rtl8192cu/os_dep/linux/ioctl_linux.c#L11590). The [`dummy`](https://github.com/raspberrypi/linux/blob/rpi-4.9.y/drivers/net/wireless/realtek/rtl8192cu/os_dep/linux/ioctl_linux.c#L3381) function returns -1 (yuck), which translates to -EPERM ("Operation not permitted").

pelwell, 2017-08-14:
To complicate things, this card has multiple drivers - can you check which driver is being loaded (using lsmod)?

pelwell, 2017-08-14:
I think we ship three drivers for that card - `8192cu` (the default, downstream driver), `rtl8192cu` (the upstream driver) and `rtl8xxxu` (the new combo upstream driver). In Raspbian we blacklist `rtl8192cu` and `rtl8xxxu`:
```
$ grep . /etc/modprobe.d/blacklist-rtl8*
/etc/modprobe.d/blacklist-rtl8192cu.conf:blacklist rtl8192cu
/etc/modprobe.d/blacklist-rtl8xxxu.conf:blacklist rtl8xxxu
```
If you want to try another driver you need to blacklist the current driver:
```
sudo echo sh -c "blacklist 8192cu" > /etc/modprobe.d/blacklist-8192cu.conf
```
Then comment out one of the other `blacklist` lines, unplug and replug your dongle.

reset12, 2017-08-14:
here ist my /etc/modprobe.d

ls -al /etc/modprobe.d/
insgesamt 52blacklist-8192cu.conf
drwxr-xr-x   2 root root  4096 Aug 14 13:06 .
drwxr-xr-x 123 root root 12288 Aug 14 08:02 ..
-rw-r--r--   1 root root    17 Aug 14 13:06 blacklist-8192cu.conf
-rw-r--r--   1 root root   336 Aug 14 08:06 blacklist-native-rtl8192.conf
-rw-r--r--   1 root root    20 Mär  2 17:14 blacklist-rtl8192cu.conf
-rw-r--r--   1 root root    19 Mär  2 17:15 blacklist-rtl8xxxu.conf
-rw-r--r--   1 root root   127 Okt  4  2014 dkms.conf
-rw-r--r--   1 root root   390 Apr  2  2016 fbdev-blacklist.conf
-rw-r--r--   1 root root    65 Mai 27  2016 ipv6.conf
-rw-r--r--   1 root root    16 Jul 23  2014 libpisock9.conf
-rw-r--r--   1 root root    50 Nov 14  2014 owfs-common.conf
-rw-r--r--   1 root root     0 Mai 27  2016 raspi-blacklist.conf

After Blacklist the driver lblacklist-8192cu.conf.
Wich driver should work ?

pelwell, 2017-08-14:
Interesting - you have four blacklist files for three drivers. Judging by the timestamps, the original Raspbian-sourced blacklists (for rtl819cu and rtl8xxxu) are still active, so if your new blacklist files include 8192cu then no drivers should be loaded automatically. This actually works - you just have to manually modprobe the driver you want.

I see that you have closed the issue - it would help other users if you could explain what you have discovered - but if you still need assistance then you'll need to show the contents of each of those files.

reset12, 2017-08-14:
Tho close

reset12, 2017-08-14:
The close was an error.
You are right, i have no driver currently active and i dont know wich driver should work.
When i try to activate a driver i get an error:
/etc/modprobe.d# modprobe rtl8192cu
modprobe: ERROR: ../libkmod/libkmod-module.c:960 command_do() Error running install command for rtlwifi
modprobe: ERROR: could not insert 'rtl8192cu': Operation not permitted

pelwell, 2017-08-14:
> if you still need assistance then you'll need to show the contents of each of those files.

pelwell, 2017-08-14:
I don't recognise `blacklist-native-rtl8192.conf` - is that an Ubuntu file? Try deleting/moving it, then commenting out `blacklist rtl8xxxu`, so you get:
```
$ cat blacklist-rtl8xxxu.conf
# blacklist rtl8xxxu
```
That should activate the rtl8xxxu driver - confirm that with an lsmod after reconnecting the dongle. If that doesn't work for you, remove the '#' comment character and put another one in `blacklist-rtl8192cu.conf`:
```
cat blacklist-rtl8192cu.conf
#blacklist rtl8192cu
```
and try again.

reset12, 2017-08-14:
[   75.742174] usbcore: registered new interface driver rtl8xxxu
[  218.786585] usb 1-1.4: USB disconnect, device number 6
[  224.197273] usb 1-1.4: new high-speed USB device number 8 using dwc_otg
[  224.328679] usb 1-1.4: New USB device found, idVendor=0bda, idProduct=8178
[  224.328692] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  224.328699] usb 1-1.4: Product: 802.11n WLAN Adapter
[  224.328707] usb 1-1.4: Manufacturer: Realtek
[  224.328715] usb 1-1.4: SerialNumber: 00e04c000001
root@raspberrypi4:/home/t759f# lsusb
Bus 001 Device 007: ID 046d:0991 Logitech, Inc. QuickCam Pro for Notebooks
Bus 001 Device 008: ID 0bda:8178 Realtek Semiconductor Corp. RTL8192CU 802.11n WLAN Adapter
Bus 001 Device 005: ID 046d:082b Logitech, Inc. 
Bus 001 Device 004: ID 04fa:2490 Dallas Semiconductor DS1490F 2-in-1 Fob, 1-Wire adapter
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
root@raspberrypi4:/home/t759f# ifconfig wlan1 up
wlan1: ERROR while getting interface flags: Kein passendes Gerät gefunden

2. Try:
ifconfig
eth0      Link encap:Ethernet  Hardware Adresse b8:27:eb:f5:59:04  
          inet6-Adresse: fe80::3990:121e:6f07:9f64/64 Gültigkeitsbereich:Verbindung
          UP BROADCAST MULTICAST  MTU:1500  Metrik:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          Kollisionen:0 Sendewarteschlangenlänge:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

lo        Link encap:Lokale Schleife  
          inet Adresse:127.0.0.1  Maske:255.0.0.0
          inet6-Adresse: ::1/128 Gültigkeitsbereich:Maschine
          UP LOOPBACK RUNNING  MTU:65536  Metrik:1
          RX packets:230 errors:0 dropped:0 overruns:0 frame:0
          TX packets:230 errors:0 dropped:0 overruns:0 carrier:0
          Kollisionen:0 Sendewarteschlangenlänge:1 
          RX bytes:18692 (18.2 KiB)  TX bytes:18692 (18.2 KiB)

wlan0     Link encap:Ethernet  Hardware Adresse b8:27:eb:a0:0c:51  
          inet Adresse:192.168.1.5  Bcast:192.168.1.255  Maske:255.255.255.0
          inet6-Adresse: fe80::ba27:ebff:fea0:c51/64 Gültigkeitsbereich:Verbindung
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metrik:1
          RX packets:177 errors:0 dropped:0 overruns:0 frame:0
          TX packets:208 errors:0 dropped:0 overruns:0 carrier:0
          Kollisionen:0 Sendewarteschlangenlänge:1000 
          RX bytes:15568 (15.2 KiB)  TX bytes:71256 (69.5 KiB)

wlan1     Link encap:Ethernet  Hardware Adresse 00:0d:81:ac:8b:f8  
          UP BROADCAST MULTICAST  MTU:1500  Metrik:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          Kollisionen:0 Sendewarteschlangenlänge:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
ifconfig
eth0      Link encap:Ethernet  Hardware Adresse b8:27:eb:f5:59:04  
          inet6-Adresse: fe80::3990:121e:6f07:9f64/64 Gültigkeitsbereich:Verbindung
          UP BROADCAST MULTICAST  MTU:1500  Metrik:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          Kollisionen:0 Sendewarteschlangenlänge:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

lo        Link encap:Lokale Schleife  
          inet Adresse:127.0.0.1  Maske:255.0.0.0
          inet6-Adresse: ::1/128 Gültigkeitsbereich:Maschine
          UP LOOPBACK RUNNING  MTU:65536  Metrik:1
          RX packets:230 errors:0 dropped:0 overruns:0 frame:0
          TX packets:230 errors:0 dropped:0 overruns:0 carrier:0
          Kollisionen:0 Sendewarteschlangenlänge:1 
          RX bytes:18692 (18.2 KiB)  TX bytes:18692 (18.2 KiB)
[... 73 lines cut ...]
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:on
          Link Quality=36/70  Signal level=-74 dBm  
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:20  Invalid misc:0   Missed beacon:0

eth0      no wireless extensions.

wlan1     IEEE 802.11  ESSID:off/any  
          Mode:Managed  Access Point: Not-Associated   Tx-Power=20 dBm   
          Retry short limit:7   RTS thr=2347 B   Fragment thr:off
          Encryption key:off
          Power Management:off
          
root@raspberrypi4:/etc/modprobe.d# iwconfig wlan1 power on
Error for wireless request "Set Power Management" (8B2C) :
    SET failed on device wlan1 ; Operation not supported.

This driver is also not the best solution.

I must now leave, thank you for your support, i will be back iin 2 Days.

with regards
Markus

pelwell, 2017-08-14:
Hmmm, I'd say none of the drivers supports any user control of TX power - they just set it based on channel number and regulatory domain. You either need to try hacking the driver or find another device with the required capability. Either way, it isn't a Pi-specific problem.

reset12, 2017-08-16:
Thank you Pelwell,
i am not be able to hacking the driver, so can you tell me a USB-Wifi Dongle with Long-Range capability.
My Raspberry is located in my chicken House and i need it as Wifi Hotspot when i work in the garden, because the Gsm-Mobile net is not available.

closed 2017-08-16 by reset12

pelwell, 2017-08-16:
I can't recommend a device because I haven't tested external WiFi dongles - sorry - but there is a page listing WiFi dongles supported on Raspberry Pi and some of their features here: http://elinux.org/RPi_USB_Wi-Fi_Adapters

7045kHz, 2020-06-06:
years later I ran into the same issue with the onboard wifi for a pi v 4.  Turned out the issue I had was with iwconfig.  I was able to fix it, and thought I would share.
 

**Power Status**
```
pi@raspberrypi4:~ $ iwconfig
eth0      no wireless extensions.

wlan0     IEEE 802.11  Mode:Master  Tx-Power=31 dBm
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:on
```

**Failure using iwconfig to set power off.**

```
pi@raspberrypi4:~ $ iwconfig wlan0 power off
Error for wireless request "Set Power Management" (8B2C) :
    SET failed on device wlan0 ; Operation not permitted.
```

**Solution - using iw instead of iwconfig**

```
root@raspberrypi4:~# iw dev wlan0 set power_save off
root@raspberrypi4:~# iwconfig
eth0      no wireless extensions.

wlan0     IEEE 802.11  Mode:Master  Tx-Power=31 dBm
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:off

lo        no wireless extensions.

```

Nifra514, 2020-12-26:
try :

 iw wlan0 set txpower fixed 3000