raspberrypi/linux #2166: Kernel crash when enabling ftrace during boot

HiassofT, 2017-08-18:
While trying to debug some early system init issues I noticed that enabling ftrace=function via cmdline.txt leads to a kernel crash and a non-booting system.

I've tested on RPi3, only dtdebug=1 and dtoverlay=pi3-disable-bt in config.txt and used this cmdline.txt
```
dwc_otg.lpm_enable=0 consoleblank=0 console=tty1 console=ttyAMA0,115200 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait earlycon=pl011,mmio32,0x3f201000 ftrace=function
```
kernel is the current rpi-update kernel bf19fe7aa723ef8ddafdf8ca47a7205e35749ddc

This lead to the following crash: https://gist.github.com/HiassofT/65d57fe4c5d0a915f2cda0875a96341d

Originally I noticed that crash with the 4.12.y tree and a quick test showed that RPi1 with 4.9 kernel crashes as well.

Not 100% sure what's going wrong, enabling eg the function ftracer during runtime on RPi3 works fine
```
mount -t tracefs tracefs /sys/kernel/tracing/
echo "function" > /sys/kernel/tracing/current_tracer
```

codekatana, 2017-08-18:
Hi, I am a rather newbie here (I mean on raspberry pi) - I would like to know, how you collect kernel log after the crash? Do you copy it from screen or there is some dump area which stores the kernel log in an event of crash?

pelwell, 2017-08-18:
@HiassofT is using a serial port attached to header pins 6, 8 and 10 (GND, GPIO14, GPIO15). The kernel command line enables it:
```
dwc_otg.lpm_enable=0 [...] console=ttyAMA0,115200 [...] earlycon=pl011,mmio32,0x3f201000
```
For this to work as expected on a Pi 3 you also need:
```
dtoverlay=pi3-disable-bt
```
or
```
dtoverlay=pi3-miniuart-bt
```

pelwell, 2017-08-18:
I don't have a complete answer yet, but I can tell you that the problem is caused by writes to variables that are declared as `__ro_after_init`. The first failure is `vdso_start_data` being written from `vdso_init`, but if you remove the attribute then the fault moves to the next write that fits the pattern (`vdso_text_mapping`, `cntvct_ok`), eventually appearing in a different function altogether (`ptmx_fops`, written from `tty_default_fops`).

What makes this interesting (and is the reason this doesn't fail normally) is that these functions are all called within `kernel_init`, so the `__ro_after_init` attribute shouldn't have caused the variable to be read-only yet.

pelwell, 2017-08-18:
The `mark_readonly` and `mark_rodata_ro` are not being called early (and adding `rodata=n` to the command line doesn't fix it), so it must be that the default protection state is being set to read-only.

Digging deeper I've narrowed it down to the call to `set_kernel_text_ro` from the `__ftrace_modify_code` function in `arch/arm/kernel/ftrace.c`. It uses the same mechanism and section data as `mark_rodata_ro`. My theory is:
1) Data which is marked `__ro_after_init` is initially writeable.
2) The `ro_perms` data covers kernel text, read-only data _and_ `__ro_after_init` data.
3) `set_kernel_text_rw` marks everything in `ro_perms` as writeable.
4) `set_kernel_text_ro` marks  everything in `ro_perms` as read-only, including the `__ro_after_init` data.
5) Using the function tracing code involves code modification, resulting in calls to `__ftrace_modify_code` and `set_kernel_text_ro`.
6) Therefore if function tracing is enabled before kernel_init has completed then the `__ro_after_init` data is made read-only prematurely.

This doesn't look like a Pi-specific bug/feature. A fix would require separating out the section permission data for the `__ro_after_init`, updating the `mark_rodata_ro` to use the new permissions data instead.

HiassofT, 2017-08-18:
@pelwell thanks a lot for looking into it!

I did some more tests and have to revise what I wrote about RPi1, it actually works on RPi1 (tested with B+ with serial console and original B without serial console). Here's the dmesg from the B+ boot: https://gist.github.com/HiassofT/2b3ffc4007eb5b9215335513aa8acb4c

The problem in my previous test probably was that I used the RPi3 pl011 MMIO address on RPi1 in earlycon...

For completeness I also tested RPi2, that is crashing as well, like RPi3.

A quick test on x86_64 (Debian Stretch with current 4.9 Debian kernel) worked fine.

Very puzzling... is the set_kernel_text_ro code on armv6 maybe different  to armv7?

HiassofT, 2017-08-21:
The kernel crash also occurs with upstream kernels 4.9, 4.12 and 4.13-rc. On 4.12/4.13 the crashes are a bit different, seem to be caused by SMP handling in irq-bcm2836.

I've posted on the linux-rpi-kernel list:
http://lists.infradead.org/pipermail/linux-rpi-kernel/2017-August/006875.html

HiassofT, 2017-08-23:
And we've got a solution: https://patchwork.kernel.org/patch/9918149/

@pelwell thanks, your analysis was spot on! Could you pick the patch into 4.9/4.12/4.13 trees?

pelwell, 2017-08-23:
I'll cherry-pick the patch once it's been accepted into linux-next (or a staging area).

HiassofT, 2017-08-23:
Thanks!

I've PRed the patch for LibreELEC and we'll test it in milhouse builds, that should give some additional test coverage https://github.com/LibreELEC/LibreELEC.tv/pull/1917

Local testing went fine and as the patch comes from the ftrace author and brings kernel rodata handling on arm in line with x86 I'm rather confident it'll just work fine.

pelwell, 2017-08-23:
I have no doubts it will work, I'd just rather avoid a merge conflict if there a changes before it is accepted.

HiassofT, 2017-08-23:
Absolutely, and no need to merge/pick before it's accepted. From the current discussion it looks like there could come a v2 with some comments added, crying for merge conflicts if you added it now.

JamesH65, 2017-12-04:
@pelwell @HiassofT Any progress on this one?

HiassofT, 2017-12-05:
The patch doesn't seem to have made it into upstream trees yet, I've just pinged Steven Rostedt.

Quick test with next-20171205 shows that the kernel still oopses without the fix - with the fix it boots fine.

JamesH65, 2018-06-27:
@HiassofT Did this ever make it to upstream?

HiassofT, 2018-06-29:
The proposed patch didn't land in upstream (mainline is still crashing on arm, arm64 is fine), I've pinged upstream again https://lore.kernel.org/lkml/20180629144714.hkvwqzpikweyihmo@camel2.lan/T/#mc31df5db5bdfee4f1054aab198f12e5a4de45e17

pelwell, 2018-06-29:
If you don't get a response I'll apply it downstream anyway,

HiassofT, 2018-06-29:
thanks @pelwell 

Just noticed that Steven sent a newer version of the patch (which applies cleanly on 4.18-rc) a couple of days ago: https://patchwork.kernel.org/patch/10480199/ - for 4.17/18 better use this one

For the 4.14 tree the one from last year should still be fine https://patchwork.kernel.org/patch/9918149/

The patches are more or less identical, the newer one only adds the __read_mostly attribute to the kernel_set_to_readonly flag and is adapted for the debug_checkwx() addition to mark_rodata_ro()

pelwell, 2018-07-03:
Patches applied to rpi-4.14.y, rpi-4.17.y and rpi-4.18.y.

closed 2018-07-03 by HiassofT